home-assistant/operating-system #2684: update from v9.5 to 10.4 don't work

Indianagrenoble, 2023-08-13:
### Describe the issue you are experiencing

I don't know why but It doesn't work..
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.components.websocket_api.http.connection
Source: components/hassio/update.py:214
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 20:04:26 (1 occurrences)
Last logged: 20:04:26

[139646874992960] Error updating Home Assistant Operating System: Unknown error, see supervisor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 212, in async_install
    await async_update_os(self.hass, version)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 52, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1974, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2011, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 421, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 214, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Operating System: Unknown error, see supervisor
```

### Anything in the Host logs that might be useful for us?

```txt
nothing else
```

### System information

## System Information

version | core-2023.8.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4996
Installed Version | 1.32.1
Stage | running
Available Repositories | 1269
Downloaded Repositories | 46

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 62.0 GB
disk_used | 28.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.7.1), Studio Code Server (5.10.0), File editor (5.6.0), Duck DNS (1.15.0), ESPHome (2023.7.1), Mosquitto broker (6.2.1), TasmoAdmin (0.25.0), Node-RED (14.4.4), Nginx Proxy Manager (0.12.3), MariaDB (2.6.1), ZeroTier One (0.16.3), eWeLink Smart Home (1.4.3), Samba share (10.0.2), VLC (0.2.0), AirCast (3.5.3), FTP (4.7.1), Glances (0.19.2), Grafana (9.0.3), Home Panel (2.6.1), motionEye (0.19.1), Home Assistant Google Drive Backup (0.111.1), EMQX (0.2.10), Firefox (1.0.7), Whisper (1.0.0), Piper (1.2.0), Tor (4.2.0), Frigate (0.12.1), InfluxDB (4.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 26
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 27 juin 2023 à 19:04
-- | --
current_recorder_run | 13 août 2023 à 16:23
estimated_db_size | 2573.86 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.2 (2ad1cd7)
-- | --
cloud_online | 16 / 19
local_online | 16 / 16
debug | failed to load: 

</details>

### Additional information

[image]

agners, 2023-08-14:
You shared only the log entry from the main (Core) logs.

Can you share the Supervisor logs and Host logs after an upgrade attempt (under logs, use the top right drop down to choose "Supervisor" and "Host").

Indianagrenoble, 2023-08-15:
@agners Hi Stefan

Please find files requested
[host_2023-08-15T16-31-17.303Z.log](https://github.com/home-assistant/operating-system/files/12346351/host_2023-08-15T16-31-17.303Z.log)

[supervisor_2023-08-15T16-31-11.932Z.log](https://github.com/home-assistant/operating-system/files/12346353/supervisor_2023-08-15T16-31-11.932Z.log)

Indianagrenoble, 2023-08-17:
@agners 
Hi Stefan
Please find errors here :

Error Message: "An exception occurred"
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/meross_iot/manager.py", line 903, in _async_send_and_wait_ack
    return await asyncio.wait_for(future, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/meross_cloud/sensor.py", line 186, in async_update
    await self._device.async_get_instant_metrics(channel=self._channel_id)
  File "/usr/local/lib/python3.11/site-packages/meross_iot/controller/mixins/electricity.py", line 41, in async_get_instant_metrics
    result = await self._execute_command(method="GET",
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/meross_iot/controller/device.py", line 302, in _execute_command
    return await self._manager.async_execute_cmd(destination_device_uuid=self.uuid,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/meross_iot/manager.py", line 847, in async_execute_cmd
    return await self.async_execute_cmd_client(client=client,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/meross_iot/manager.py", line 884, in async_execute_cmd_client
    response = await self._async_send_and_wait_ack(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/meross_iot/manager.py", line 910, in _async_send_and_wait_ack
    raise CommandTimeoutError(message=str(message), target_device_uuid=target_device_uuid, timeout=timeout)
meross_iot.model.exception.CommandTimeoutError
2023-08-17 07:34:52.906 WARNING (MainThread) [aioesphomeapi.connection] esp8266-pir-temp-199 @ 10.10.10.199: Connection error occurred: [Errno 104] Connection reset by peer
2023-08-17 07:35:00.063 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) power W - Main channel reported online event.
2023-08-17 07:35:00.065 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) voltage V - Main channel reported online event.
2023-08-17 07:35:00.066 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) current A - Main channel reported online event.
2023-08-17 07:35:00.067 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) - Main channel reported online event.
2023-08-17 07:35:00.461 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) power W - Main channel reported online event.
2023-08-17 07:35:00.463 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) voltage V - Main channel reported online event.
2023-08-17 07:35:00.464 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) current A - Main channel reported online event.
2023-08-17 07:35:00.466 WARNING (MainThread) [custom_components.meross_cloud] Device Machine Vaisselle (mss310) - Main channel reported online event.
2023-08-17 07:35:02.374 WARNING (MainThread) [aioesphomeapi.connection] esp8266-temp-ballon-duo-196 @ 10.10.10.196: Connection error occurred: esp8266-temp-ballon-duo-196 @ 10.10.10.196: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (Home Assistant 2023.8.2).
2023-08-17 07:35:02.375 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esp8266-temp-ballon-duo-196 @ 10.10.10.196: esp8266-temp-ballon-duo-196 @ 10.10.10.196: Connection closed (APIConnectionError)
2023-08-17 07:39:33.247 WARNING (SyncWorker_0) [custom_components.meross_lan] MerossMQTTConnection(3011057:mqtt-eu.meross.com:443): MQTT DROP device_id:(2211014476143351080748e1e9aedfed) method:(GET) namespace:(Appliance.System.All)
2023-08-17 07:48:09.098 ERROR (MainThread) [custom_components.sonoff.core.ewelink.cloud] Cloud WS exception
Traceback (most recent call last):
  File "/config/custom_components/sonoff/core/ewelink/cloud.py", line 323, in connect
    raise Exception(resp)
Exception: {'error': 406, 'reason': 'Authentication Failed', 'apikey': '104965e6-9430-45b7-ae97-73bed34a4d7c', 'sequence': '1692251288946', 'actionName': 'userOnline'}
2023-08-17 07:49:33.314 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139772701748544] Error updating Home Assistant Operating System: Unknown error, see supervisor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 212, in async_install
    await async_update_os(self.hass, version)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 52, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor
[... cut ...]
  File "/usr/local/lib/python3.11/site-packages/aiohttp/http_websocket.py", line 663, in _write
    raise ConnectionResetError("Cannot write to closing transport")
ConnectionResetError: Cannot write to closing transport
2023-08-17 07:56:38.250 ERROR (MainThread) [frontend.js.latest.202308020] https://lebruyant.duckdns.org:8123/poco-x3-pro-mushroom/0?external_auth=1:0:0 Uncaught 
2023-08-17 07:58:15.443 ERROR (MainThread) [frontend.js.latest.202308020] https://lebruyant.duckdns.org:8123/poco-x3-pro-mushroom/0?external_auth=1:0:0 Uncaught

Indianagrenoble, 2023-08-18:
@agners
Hi Stefan 
Any comment ?

I have clean a lot of beginner's errors...
And perhaps I could send you the result ?

Regards
Alexandre

agners, 2023-08-18:
You've shared the logs of Home Assistant Core. Most errors are from custom_components (see log entries).

The last one is related to the OS update problem, however, for the exact error, I need the Supervisor log:
```
 homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor
```

Please try again to update, wait for a couple of minutes, and then share the *Supervisor* logs.

Indianagrenoble, 2023-08-21:
@agners 
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 1068, in watchdog_container
    await self._restart_after_problem(self, event.state)
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 144, in wrapper
    raise on_condition(
supervisor.exceptions.AddonsJobError: Rate limit exceeded, more then 10 calls in 0:30:00
23-08-21 18:55:07 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-08-21 18:55:07 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-08-21 18:55:08 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-08-21 18:55:08 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/blakeblackshear/frigate-hass-addons repository
23-08-21 18:55:08 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
23-08-21 18:55:08 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/mincka/ha-addons repository
23-08-21 18:55:08 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/CoolKit-Technologies/ha-addon repository
23-08-21 18:55:08 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
23-08-21 18:55:08 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
23-08-21 18:55:08 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
23-08-21 18:55:09 WARNING (SyncWorker_2) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access) Beta (0.12.0)
23-08-21 18:55:10 WARNING (SyncWorker_5) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access) Beta (0.12.0)
23-08-21 18:55:10 INFO (MainThread) [supervisor.store] Loading add-ons from store: 80 all - 0 new - 0 remove
23-08-21 18:55:10 INFO (MainThread) [supervisor.store] Loading add-ons from store: 80 all - 0 new - 0 remove
23-08-21 18:55:19 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'mqtt'
23-08-21 18:55:19 INFO (MainThread) [supervisor.auth] Home Assistant not running, checking cache
23-08-21 18:56:12 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-08-21 18:56:12 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
23-08-21 18:56:12 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-08-21 18:56:12 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-08-21 18:56:27 INFO (MainThread) [supervisor.ingress] Update Ingress as panel for a0d7b954_emqx
23-08-21 18:56:42 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-08-21 18:56:42 INFO (MainThread) [supervisor.api.proxy] WebSocket access from f4f71350_ewelink_smart_home_slug
23-08-21 18:56:42 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-08-21 18:56:46 INFO (SyncWorker_3) [supervisor.docker.manager] Restarting homeassistant
23-08-21 18:56:46 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:1000 is not str
23-08-21 18:56:46 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-08-21 18:56:46 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 8:1000 is not str
23-08-21 18:56:46 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-08-21 18:57:10 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
23-08-21 18:57:12 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-08-21 18:57:12 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-08-21 18:57:13 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-08-21 18:57:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/blakeblackshear/frigate-hass-addons repository
23-08-21 18:57:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
23-08-21 18:57:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/mincka/ha-addons repository
23-08-21 18:57:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/CoolKit-Technologies/ha-addon repository
23-08-21 18:57:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
23-08-21 18:57:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
23-08-21 18:57:13 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
[... cut ...]
23-08-21 19:02:22 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/10.5/haos_generic-x86-64-10.5.raucb
23-08-21 19:08:20 INFO (MainThread) [supervisor.jobs] 'MountManager.reload' blocked from execution, mounting not supported on system
23-08-21 19:17:29 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-10.5.raucb
23-08-21 19:17:30 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found

agners, 2023-08-29:
```
23-08-21 19:17:30 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found
````

It sounds like the system can't access the boot partition any longer :cry: 

Kinda hard to tell why/what happened exactly, it could be data loss or a hardware issue. Probably the easiest way is to take a full backup and download it, then install HAOS from scratch again.

closed 2023-12-05 by github-actions[bot]