home-assistant/operating-system #2690: HassOS keeps breaking after x reboot and sometimes after update

Operations8, 2023-08-16:
### Describe the issue you are experiencing

i switched from Debian 12 + Hass to HassOS, with the idea because i am not so into linux this would be easier to maintain. Big mistake.

I have installed HassOS with the balena Etcher. Every x reboot i it keeps hanging on waiting for CLI… i know now i need to do ctrl+al+F2 and rename the label by using “e2label /dev/sda8 hassos-data” but i cannot reach the machine easily it is in a closet with no monitor, so having to do this often is not really acceptable to me.

Also often after updates (today => 10.4) i get invalid filesystem. There is nothing more i did then a normal HASS update. shouldn’t this proces be rock solid, even more when using HassOS?

I have tried 2x NUC and an Asus NUC wannabee, and switched the SSD.

Reflashed the SSD with 10.4 and will import backup for another new start,
Immediately got the “waiting for CLI to get ready" So i had to rename the label straight after a new start. After that it works. But i af afraid of a reboot :P

Any ideas?

P.S. Is supervisor log relevant? I upload that now, if needed of course i will.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Updating
2. Rebooting

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-08-16 15:35:30 INFO (MainThread) [supervisor.addons.addon] Restoring data for addon cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:30 INFO (MainThread) [supervisor.docker.interface] Attaching to sabeechen/hassio-google-drive-backup-amd64 with version 0.111.1[0m
[32m23-08-16 15:35:31 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on sabeechen/hassio-google-drive-backup-amd64 with version 0.111.1[0m
[32m23-08-16 15:35:31 INFO (MainThread) [supervisor.addons.addon] Finished restore for add-on cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:31 INFO (MainThread) [supervisor.addons] Detect new Add-on after restore cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:31 INFO (MainThread) [supervisor.addons.addon] Restore config for addon a0d7b954_nut[0m
[32m23-08-16 15:35:31 INFO (MainThread) [supervisor.addons.addon] Restore/Install of image for addon a0d7b954_nut[0m
[32m23-08-16 15:35:31 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/hassio-addons/nut/amd64 with tag 0.12.0.[0m
[32m23-08-16 15:35:32 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:32 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:32 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:32 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:32 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:32 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:32 INFO (MainThread) [supervisor.api.middleware.security] /backups/5590e33d/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-08-16 15:35:43 INFO (MainThread) [supervisor.addons.addon] Restoring data for addon a0d7b954_nut[0m
[32m23-08-16 15:35:43 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/nut/amd64 with version 0.12.0[0m
[32m23-08-16 15:35:44 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/nut/amd64 with version 0.12.0[0m
[32m23-08-16 15:35:44 INFO (MainThread) [supervisor.addons.addon] Finished restore for add-on a0d7b954_nut[0m
[32m23-08-16 15:35:44 INFO (MainThread) [supervisor.addons] Detect new Add-on after restore a0d7b954_nut[0m
[32m23-08-16 15:35:44 INFO (MainThread) [supervisor.addons.addon] Restore config for addon a0d7b954_grafana[0m
[32m23-08-16 15:35:44 INFO (MainThread) [supervisor.addons.addon] Restore/Install of image for addon a0d7b954_grafana[0m
[32m23-08-16 15:35:44 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/hassio-addons/grafana/amd64 with tag 9.0.3.[0m
[32m23-08-16 15:36:40 INFO (MainThread) [supervisor.addons.addon] Restoring data for addon a0d7b954_grafana[0m
[32m23-08-16 15:36:40 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/grafana/amd64 with version 9.0.3[0m
[32m23-08-16 15:36:41 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/grafana/amd64 with version 9.0.3[0m
[32m23-08-16 15:36:41 INFO (MainThread) [supervisor.addons.addon] Finished restore for add-on a0d7b954_grafana[0m
[32m23-08-16 15:36:41 INFO (MainThread) [supervisor.addons] Detect new Add-on after restore a0d7b954_grafana[0m
[32m23-08-16 15:36:42 INFO (MainThread) [supervisor.addons.addon] Restore config for addon a0d7b954_vscode[0m
[32m23-08-16 15:36:42 INFO (MainThread) [supervisor.addons.addon] Restore/Install of image for addon a0d7b954_vscode[0m
[32m23-08-16 15:36:42 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/hassio-addons/vscode/amd64 with tag 5.10.0.[0m
[32m23-08-16 15:37:20 INFO (MainThread) [supervisor.addons.addon] Restoring data for addon a0d7b954_vscode[0m
[32m23-08-16 15:37:20 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.10.0[0m
[... 230 lines cut ...]
version_api | ok
installed_addons | Advanced SSH & Web Terminal (15.0.7), Samba share (10.0.2), Silicon Labs Multiprotocol (2.3.1), AdGuard Home (4.8.14), Samba Backup (5.2.0), Home Assistant Google Drive Backup (0.111.1), Network UPS Tools (0.12.0), Grafana (9.0.3), Studio Code Server (5.10.0), Ring-MQTT with Video Streaming (5.5.1), Mosquitto broker (6.2.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 6
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 4 augustus 2023 om 20:44
-- | --
current_recorder_run | 16 augustus 2023 om 15:38
estimated_db_size | 249.02 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

agners, 2023-08-16:
> i switched from Debian 12 + Hass to HassOS, with the idea because i am not so into linux this would be easier to maintain. 

It typically should be.

> Big mistake.

Sorry to hear your regret. You can use the backup option to take a full backup and restore it on a Supervised installation.

> I have installed HassOS with the balena Etcher. Every x reboot i it keeps hanging on waiting for CLI… i know now i need to do ctrl+al+F2 and rename the label by using “e2label /dev/sda8 hassos-data” [...]

Is that on a system with a single disk installed? What is `e2label /dev/sda8` saying (without the new name)?

Did you try to use the data disk feature?

> Immediately got the “waiting for CLI to get ready" So i had to rename the label straight after a new start. After that it works. But i am afraid for a reboot :P

On a very new installation this is expected as things take a moment to boot up. It should not timeout though, unless you have a slow internet connection.

Operations8, 2023-08-17:
> It typically should be.
>
>
> Sorry to hear your regret. You can use the backup option to take a full backup and restore it on a Supervised installation.
>
>
> Is that on a system with a single disk [...]

The NUC has could an internal (i think eMMC) harddisk, but the Asus doesn't. I probably could turn this disc off in the BIOS.

E2label says (from the top of my head) hassos_dis

It doesn't take "a moment" it hangs. I let it run on one of the NUC for hours and no joy. I really have to rename the labe first when this happens. After that it comes up in seconds.

What do you mean by data disk feature?

agners, 2023-08-22:
> What do you mean by data disk feature?

The operating system can use an external disk as data disk. The logic behind this uses file system labels. See also:
https://www.home-assistant.io/common-tasks/os/#using-external-data-disk

> E2label says (from the top of my head) hassos_dis

Hm, this means that the Operating System detects it as being an unrelated data disk (not the one the boot loader loaded the kernel from, this is being done by the [haos-data-disk-detach script at startup](https://github.com/home-assistant/operating-system/blob/rel-10/buildroot-external/rootfs-overlay/usr/libexec/haos-data-disk-detach#L20).

> Turned the internal disc off.

Did this help to fix the problem?

If not, can you run `blkid` when the system is not correctly coming up and share the output?

Operations8, 2023-08-22:
Well i have done 2 updates and about 15 reboots since i turned the internal disc off. Hard to say it this has any effect on the problem. Before turning it off and before this topic there were also x number of reboots and updates without any problems. Also the fact that i normally not so often have to reboot make it difficult to compare.

It is really hard for me to pinpoint what triggers the issue.

Operations8, 2023-08-22:
To be honest i don't see how. I use the elena batcher methode. So i remove the SSD from the NUC and connect it to my pc via SATA to USB converter to flash. It is not even possible to remove the internal MMC disc, so making the mistake of taking out the wrong disc (which is highly unlikely by it self) it not possible.

agners, 2023-08-22:
Yeah I see if you removed the disk to flash then it's pretty much impossible :sweat_smile: 

The other way which the internal disk could have ended up with the data disk label is if you would have used the data disk feature from the UI (https://www.home-assistant.io/common-tasks/os/#using-ui-to-move-the-data-partition).

If both is not the case, then I am really wonder why/how the script ends up detecting your real data partition as an external data disk.

Operations8, 2023-08-22:
> Yeah I see if you removed the disk to flash then it's pretty much impossible 😅
> 
> The other way which the internal disk could have ended up with the data disk label is if you would have used the data disk feature from the UI (https://www.home-assistant.io/common-tasks/os/#using-ui-to-move-the-data-partition).
> 
> If both is not the case, then I am really wonder why/how the script ends up detecting your real data partition as an external data disk.

I didn't even know about the data disk feature / move feature until you brought it up. So again highly unlikely / impossible :p

closed 2023-11-28 by github-actions[bot]