home-assistant/operating-system #2692: http://sources.buildroot.net/os-agent/os-agent-1.5.1.tar.gz 404 not found

hippich, 2023-08-17:
### Describe the issue you are experiencing

I am trying to build generic_x86_64 target using command:

```
sudo scripts/enter.sh make generic_x86_64
```

Build stops due to `404 not found` error:

```
Sending build context to Docker daemon  14.34kB
Step 1/7 : FROM debian:bullseye
---> 07585eb55737
Step 2/7 : SHELL ["/bin/bash", "-o", "pipefail", "-c"]
---> Using cache
---> 718b0a59d27d
Step 3/7 : RUN apt-get update && apt-get install -y --no-install-recommends         apt-transport-https         ca-certificates         curl         gpg-agent         gpg        dirmngr         software-properties-common     && curl -fsSL https://download.docker.com/linux/debian/gpg | apt-key add -     && add-apt-repository "deb https://download.docker.com/linux/debian $(lsb_release -cs) stable"     && apt-get update && apt-get install -y --no-install-recommends         docker-ce     && rm -rf /var/lib/apt/lists/*
---> Using cache
---> d621b2486bb2
Step 4/7 : RUN apt-get update && apt-get install -y --no-install-recommends         bash         bc         binutils         build-essential         bzip2         cpio         file         git         graphviz         jq         make         ncurses-dev         openssh-client         patch         perl         python3         python3-matplotlib         python-is-python3         qemu-utils         rsync         skopeo         sudo         unzip         vim         wget         zip     && rm -rf /var/lib/apt/lists/*
---> Using cache
---> c40f96d7ec49
Step 5/7 : COPY scripts/entry.sh /usr/sbin/
---> Using cache
---> 42eb1cefa9ec
Step 6/7 : ENTRYPOINT ["/usr/sbin/entry.sh"]
---> Using cache
---> be0b93a8e37c
Step 7/7 : WORKDIR /build
---> Using cache
---> 833ae94195ff
Successfully built 833ae94195ff
Successfully tagged hassos:local
config generic_x86_64
/usr/bin/make -C /build/buildroot O=/build/output BR2_EXTERNAL=/build/buildroot-external "generic_x86_64_defconfig"
make[1]: Entering directory '/build/buildroot'
GEN     /build/output/Makefile
#
# configuration written to /build/output/.config
#
make[1]: Leaving directory '/build/buildroot'
build generic_x86_64
/usr/bin/make -C /build/buildroot O=/build/output BR2_EXTERNAL=/build/buildroot-external VERSION_DEV="dev20230817"
make[1]: Entering directory '/build/buildroot'
/usr/bin/make -j1 O=/build/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
GEN     /build/output/Makefile
make[2]: Leaving directory '/build/buildroot'
>>> os-agent 1.5.1 Downloading
wget --passive-ftp -nd -t 3 -O '/build/output/build/.os-agent-1.5.1.tar.gz.a1rJ3z/output' 'https://github.com/home-assistant/os-agent/archive/1.5.1/os-agent-1.5.1.tar.gz'
--2023-08-17 18:44:01--  https://github.com/home-assistant/os-agent/archive/1.5.1/os-agent-1.5.1.tar.gz
Resolving github.com (github.com)... 140.82.113.4
Connecting to github.com (github.com)|140.82.113.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/home-assistant/os-agent/tar.gz/refs/tags/1.5.1 [following]
--2023-08-17 18:44:02--  https://codeload.github.com/home-assistant/os-agent/tar.gz/refs/tags/1.5.1
Resolving codeload.github.com (codeload.github.com)... 140.82.112.9
Connecting to codeload.github.com (codeload.github.com)|140.82.112.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: '/build/output/build/.os-agent-1.5.1.tar.gz.a1rJ3z/output'

/build/output/build/.os-agent-1.5.1.tar.gz.a     [ <=>                                                                                         ]  54.08K  --.-KB/s    in 0.09s

2023-08-17 18:44:02 (570 KB/s) - '/build/output/build/.os-agent-1.5.1.tar.gz.a1rJ3z/output' saved [55383]

go: github.com/getsentry/sentry-go@v0.19.0 requires
github.com/kataras/iris/v12@v12.2.0-beta5: reading github.com/kataras/iris/go.mod at revision v12.2.0-beta5: unknown revision v12.2.0-beta5
go: github.com/getsentry/sentry-go@v0.19.0 requires
github.com/kataras/iris/v12@v12.2.0-beta5: reading github.com/kataras/iris/go.mod at revision v12.2.0-beta5: unknown revision v12.2.0-beta5
wget --passive-ftp -nd -t 3 -O '/build/output/build/.os-agent-1.5.1.tar.gz.iNP69T/output' 'http://sources.buildroot.net/os-agent/os-agent-1.5.1.tar.gz'
--2023-08-17 18:44:03--  http://sources.buildroot.net/os-agent/os-agent-1.5.1.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 172.67.72.56, 104.26.0.37, 104.26.1.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|172.67.72.56|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-08-17 18:44:03 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/build/output/build/.os-agent-1.5.1.tar.gz.pPQWNW/output' 'http://sources.buildroot.net/os-agent-1.5.1.tar.gz'
--2023-08-17 18:44:03--  http://sources.buildroot.net/os-agent-1.5.1.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 104.26.0.37, 172.67.72.56, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-08-17 18:44:03 ERROR 404: Not Found.

make[1]: *** [package/pkg-generic.mk:189: /build/output/build/os-agent-1.5.1/.stamp_downloaded] Error 1
make[1]: Leaving directory '/build/buildroot'
make: *** [Makefile:35: generic_x86_64] Error 2
```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. git clone https://github.com/home-assistant/operating-system/
2. cd operating-system/
3. git submodule update --init
4. sudo scripts/enter.sh make generic_x86_64

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### System information

Ubuntu 22.04

hassinegabsi, 2023-08-18:
i had the same problem and when i tried to acceed to the url "**http://sources.buildroot.net/os-agent-1.5.1.tar.gz**" in the navigator i got "**ERR_TOO_MANY_REDIRECTS**" i think it would be resolved just a question of time.

agners, 2023-08-21:
The buildroot URL is essentially a fallback, and shouldn't really matter.

The primary source for the OS Agent is GitHub, and it does seem it succeeds downloading at first:

```
#
# configuration written to /build/output/.config
#
make[1]: Leaving directory '/build/buildroot'
build generic_x86_64
/usr/bin/make -C /build/buildroot O=/build/output BR2_EXTERNAL=/build/buildroot-external VERSION_DEV="dev20230817"
make[1]: Entering directory '/build/buildroot'
/usr/bin/make -j1 O=/build/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
GEN     /build/output/Makefile
make[2]: Leaving directory '/build/buildroot'
>>> os-agent 1.5.1 Downloading
wget --passive-ftp -nd -t 3 -O '/build/output/build/.os-agent-1.5.1.tar.gz.a1rJ3z/output' 'https://github.com/home-assistant/os-agent/archive/1.5.1/os-agent-1.5.1.tar.gz'
--2023-08-17 18:44:01--  https://github.com/home-assistant/os-agent/archive/1.5.1/os-agent-1.5.1.tar.gz
Resolving github.com (github.com)... 140.82.113.4
Connecting to github.com (github.com)|140.82.113.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/home-assistant/os-agent/tar.gz/refs/tags/1.5.1 [following]
--2023-08-17 18:44:02--  https://codeload.github.com/home-assistant/os-agent/tar.gz/refs/tags/1.5.1
Resolving codeload.github.com (codeload.github.com)... 140.82.112.9
Connecting to codeload.github.com (codeload.github.com)|140.82.112.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: '/build/output/build/.os-agent-1.5.1.tar.gz.a1rJ3z/output'

/build/output/build/.os-agent-1.5.1.tar.gz.a     [ <=>                                                                                         ]  54.08K  --.-KB/s    in 0.09s

2023-08-17 18:44:02 (570 KB/s) - '/build/output/build/.os-agent-1.5.1.tar.gz.a1rJ3z/output' saved [55383]

go: github.com/getsentry/sentry-go@v0.19.0 requires
github.com/kataras/iris/v12@v12.2.0-beta5: reading github.com/kataras/iris/go.mod at revision v12.2.0-beta5: unknown revision v12.2.0-beta5
go: github.com/getsentry/sentry-go@v0.19.0 requires
github.com/kataras/iris/v12@v12.2.0-beta5: reading github.com/kataras/iris/go.mod at revision v12.2.0-beta5: unknown revision v12.2.0-beta5
wget --passive-ftp -nd -t 3 -O '/build/output/build/.os-agent-1.5.1.tar.gz.iNP69T/output' 'http://sources.buildroot.net/os-agent/os-agent-1.5.1.tar.gz'
--2023-08-17 18:44:03--  http://sources.buildroot.net/os-agent/os-agent-1.5.1.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 172.67.72.56, 104.26.0.37, 104.26.1.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|172.67.72.56|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-08-17 18:44:03 ERROR 404: Not Found.
```

Is this issue reproducible?

adeepn, 2023-08-25:
> The buildroot URL is essentially a fallback, and shouldn't really matter.
>
> The primary source for the OS Agent is GitHub, and it does seem it succeeds downloading at first:
>
> ```
> #
> # configuration [...]

Yes. Same problem.
`github.com/kataras/iris/v12@v12.2.0-beta5: reading github.com/kataras/iris/go.mod at revision v12.2.0-beta5: unknown revision v12.2.0-beta5`

citruz, 2023-08-26:
Same problem here. 100% reproducible, even after a clean build.

citruz, 2023-08-26:
The latest sentry version 0.23 depends on a non-beta version of iris. Maybe bump the dependency in os-agent?
https://github.com/getsentry/sentry-go/blob/9ece4d371550648a52f18b4cad76ab95cb9fb1a6/go.sum#L78

citruz, 2023-08-26:
@hippich @hassinegabsi @adeepn To fix the build in the meantime, you can build os-agent from main:
```patch
diff --git a/buildroot-external/package/os-agent/os-agent.mk b/buildroot-external/package/os-agent/os-agent.mk
index 572fce6bf..dfe9060ec 100644
--- a/buildroot-external/package/os-agent/os-agent.mk
+++ b/buildroot-external/package/os-agent/os-agent.mk
@@ -4,7 +4,7 @@
 #
 ################################################################################
 
-OS_AGENT_VERSION = 1.5.1
+OS_AGENT_VERSION = a1bb47d
 OS_AGENT_SITE = $(call github,home-assistant,os-agent,$(OS_AGENT_VERSION))
 OS_AGENT_LICENSE = Apache License 2.0
 OS_AGENT_LICENSE_FILES = LICENSE
```

PR #2734 (2023-09-05): Bump HAOS Agent to 1.6.0
This also fixes #2692.
closed 2023-09-05 by agners