home-assistant/operating-system #2698: Not able to use packages with sub-requirement h5py

basbruss, 2023-08-22:
### Describe the issue you are experiencing

A custom integration that uses `pvlib` to do calculations cannot be installed on HAOS because the installation of the integration constantly fails with a subprocess.

The integration installs fine on docker core versions of HA

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. install https://github.com/basbruss/test as an integration
2. run config flow

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System information

Version | core-2023.8.3
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.11.4
Operating System Family | Linux
Operating System Version | 6.1.45
CPU Architecture | x86_64
Timezone | Europe/Amsterdam
Configuration Directory | /config

### Additional information

Core logs
```
Logger: homeassistant.util.package
Source: util/package.py:107
First occurred: August 21, 2023 at 2:44:31 PM (3 occurrences)
Last logged: August 21, 2023 at 2:45:22 PM

Unable to install package pvlib: error: subprocess-exited-with-error × Building wheel for h5py (pyproject.toml) did not run successfully. │ exit code: 1 ╰─> [127 lines of output] running bdist_wheel running build running build_py creating build creating build/lib.linux-x86_64-cpython-311 creating build/lib.linux-x86_64-cpython-311/h5py copying h5py/version.py -> build/lib.linux-x86_64-cpython-311/h5py copying h5py/ipy_completer.py -> build/lib.linux-x86_64-cpython-311/h5py copying h5py/h5py_warnings.py -> build/lib.linux-x86_64-cpython-311/h5py copying h5py/__init__.py -> build/lib.linux-x86_64-cpython-311/h5py creating build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/vds.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/selections2.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/selections.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/group.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/filters.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/files.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/dims.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/datatype.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/dataset.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/compat.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/base.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/attrs.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl copying h5py/_hl/__init__.py -> build/lib.linux-x86_64-cpython-311/h5py/_hl creating build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_slicing.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_selections.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_ros3.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_objects.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5z.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5t.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5pl.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5p.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5o.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5f.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5d_direct_chunk.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_h5.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_group.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_filters.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_file_image.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_file_alignment.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_file2.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_file.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_errors.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_dtype.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_dims_dimensionproxy.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_dimension_scales.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_datatype.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_dataset_swmr.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_dataset_getitem.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_dataset.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_completions.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_big_endian_file.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_base.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_attrs_data.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_attrs.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/test_attribute_create.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/conftest.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/common.py -> build/lib.linux-x86_64-cpython-311/h5py/tests copying h5py/tests/__init__.py -> build/lib.linux-x86_64-cpython-311/h5py/tests creating build/lib.linux-x86_64-cpython-311/h5py/tests/data_files copying h5py/tests/data_files/__init__.py -> build/lib.linux-x86_64-cpython-311/h5py/tests/data_files creating build/lib.linux-x86_64-cpython-311/h5py/tests/test_vds copying h5py/tests/test_vds/test_virtual_source.py -> build/lib.linux-x86_64-cpython-311/h5py/tests/test_vds copying h5py/tests/test_vds/test_lowlevel_vds.py -> build/lib.linux-x86_64-cpython-311/h5py/tests/test_vds copying h5py/tests/test_vds/test_highlevel_vds.py -> build/lib.linux-x86_64-cpython-311/h5py/tests/test_vds copying h5py/tests/test_vds/__init__.py -> build/lib.linux-x86_64-cpython-311/h5py/tests/test_vds copying h5py/tests/data_files/vlen_string_s390x.h5 -> build/lib.linux-x86_64-cpython-311/h5py/tests/data_files copying h5py/tests/data_files/vlen_string_dset_utc.h5 -> build/lib.linux-x86_64-cpython-311/h5py/tests/data_files copying h5py/tests/data_files/vlen_string_dset.h5 -> build/lib.linux-x86_64-cpython-311/h5py/tests/data_files running build_ext Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module> main() File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main json_out['return_val'] = hook(**hook_input['kwargs']) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel return _build_backend().build_wheel(wheel_directory, config_settings, ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/tmp/pip-build-env-qovfv6be/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 434, in build_wheel return self._build_with_temp_dir( ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/tmp/pip-build-env-qovfv6be/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 419, in _build_with_temp_dir self.run_setup() File "/tmp/pip-build-env-qovfv6be/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 341, in run_setup exec(code, locals()) File "<string>", line 68, in <module> File "/tmp/pip-build-env-qovfv6be/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 107, in setup return distutils.core.setup(**attrs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/tmp/pip-build-env-qovfv6be/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup return run_commands(dist) ^^^^^^^^^^^^^^^^^^ File  [... cut ...] tmp/pip-build-env-qovfv6be/overlay/lib/python3.11/site-packages/Cython/Compiler/Scanning.cpython-311-x86_64-linux-musl.so: Operation not permitted [end of output] note: This error originates from a subprocess, and is likely not a problem with pip. ERROR: Failed building wheel for h5py ERROR: Could not build wheels for h5py, which is required to install pyproject.toml-based projects [notice] A new release of pip is available: 23.1.2 -> 23.2.1 [notice] To update, run: pip install --upgrade pip
```

agners, 2023-08-31:
The Core container and Home Assistant OS use the same base container, with the same Python version etc, so this seems somewhat strange.

It seems there is a problem with Cython?
```
Error loading shared library /tmp/pip-build-env-qovfv6be/overlay/lib/python3.11/site-packages/Cython/Compiler/Scanning.cpython-311-x86_64-linux-musl.so: Operation not permitted [end of output]
```

Which Docker Core version did you use? And what Core version are you using with Home Assistant OS?

basbruss, 2023-09-01:
Both were on `Core-2023.8.3`.

The docker core version is run as a dev-container with `pvlib` added to its `requirements.txt`

studiobts, 2024-03-25:
I'm trying to create a custom integration which needs the same library and I have the same lor error. Do you have resolved the issue?

okkine, 2025-02-08:
I too have this problem. Has anyone found a solution?