home-assistant/operating-system #2699: DNS doesn't work

Bartek200219, 2023-08-22:
### Describe the issue you are experiencing

HAOS never tries to connect to anything besides dns in locals (packet capture; ignores any configuration dns update/network update) that can't be changed 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install homeassistant in bridged network vm
2. DHCP gives nameserver that doesn't work 
3. Change DNS to 1.1.1.1 (dns update/network update) 
4. HAOS ignores changes in configuration
5. Reboot
6. HAOS still ignores changes in configuration

### Anything in the Supervisor logs that might be useful for us?

```txt
[supervisor.docker.interface] Can't install ghcr.io/home-assistant/qemux86-64-homeassistant:2023.8.3: 500 Server Error for http+docker://localhost/v1.42/images/create?tag=2023.8.3&fromImage=ghcr.io%2Fhome-assistant%2Fqemux86-64-homeassistant&platform=linux%2Famd64: Internal Server Error ("Get "https://ghcr.io/v2/": dial tcp: lookup ghcr.io: no such host")
23-08-22 10:52:12 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
```

### Anything in the Host logs that might be useful for us?

```txt
Aug ZZ 10:53:26 homeassistant Using degraded feature set TCP instead of UDP or DNS server 1?2.16.0.10. Aug 22 and ler hcr. io: 10:53. 10:53. 10:53. 10:53. 10:53. 10:53. 10:53. 10:53. prog— 51 10:53. prog— 50 10:53. 10:53. prog— 52 10:53. prog— 53 10:53. prog— 54 10:53. 10:53. 10:53:26 for POST no such •32 •32 •32 •32 •32 •32 •32 •32 •32 •32 •32 •32 •32 •32 •32 homeassistant dockerd[42Zl: level -error zul .42/ imageszcreate returned error: Get N"https ://ghcr. i0/uZ/N": dial tcp: lookup Aug Aug Aug Aug Aug Aug Aug Y. Aug Aug Aug Aug Aug Aug Aug Aug 22 22 22 22 host" homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant homeass i stant systemd[ll: Created sl ice Sl ice zsystenzsystemd—coredump. kernel: audit: type-1334 prog-id=49 op=LOAD kernel: audit: type-1334 prog-id=50 op=LOAD kernel: audit: type-1334 prog-id=51 op=LOAD systemd[ll: Started Process Core Dump (PID 2021/UID 0). systemd—coredump [ZOZZI: Process 201? (cat) of systemd[ll: systemd—coredump@0—ZOZ1—0.service: 1334 1334 1334 1334 1334 1334 user 0 dumped core. Deact ivated successful I kernel : kernel : kernel : kernel : kernel : kernel : audit: audit: audit: audit: audit: audit: type= type= type= type= type= type= prog— 49 op=UNLO op=UNLO op=UNLO op=LOAD op=LOAD op=LOAD systemd[ll: Started Process Core Dump (PID 2027/1.JID O). Resource linits disable core dumping for proce ss 2016 (rlurap). Aug ZZ 10:53:32 homeassistant Aug 22 10:53:32 homeassistant Y. Aug 22 10:53:32 homeassistant Aug 22 10:53:35 homeassistant or DNS server 1?2.16.0.10. Aug ZZ 10:53:35 homeassistant or DNS server 1?2.16.0.10. systemd—coredump [20281: Process 2016 (rlwrap) systemd[ll: systemd—coredump@1—ZOZ7—0.service: kernel : audit: type-1334 prog-id=54 op=UNLO Using degraded feature Using degraded feature of user 0 dumped core. Deact ivated successful I set UDP instead of TCP f set UDP instead of TCP f
```

### System information

[image]

### Additional information

[image]
VM gets ip from DHCP with no problem
Ubuntu in same configured vm works without any problems 
Tried previous system version too, that's why it says 10.4 above

agners, 2023-08-31:
We use the DHCP provided DNS only in OS, or fallback to 1.1.1.1. At least that is what systemd-resolved should do.

> DHCP gives nameserver that doesn't work

In what way does your DNS server not work? Do I udnerstand correctly that Ubuntu works with that DNS server? What Ubuntu version have you tested with? Is your Ubuntu also using systemd-resolved?

Bartek200219, 2023-09-01:
> In what way does your DNS server not work? 

HAOS VM network is bridge and dns server is on the host, so HAOS doesn't see default DNS (adguard home docker)

>Do I understand correctly that Ubuntu works with that DNS server? 

No, it works with 1.1.1.1

>What Ubuntu version have you tested with? 

kubuntu 20.04

>Is your Ubuntu also using systemd-resolved?

I think yes