home-assistant/operating-system #2701: HOAS resolv.conf has wrong search path in homeassistant container

ToxicFrog, 2023-08-24:
### Describe the issue you are experiencing

I've been having DNS problems with a newly set up HOAS install, where it couldn't resolve any local hostnames. Looking at the logs showed it trying to resolve names like `timelapse.local.hass.io` rather than just `timelapse` or using the DNS search path provided by DHCP.

`ha dns info` is fine and `nslookup` works as expected on both the host and in the `hassio_dns` container. However, looking inside the main homeassistant container we see:

```
$ docker exec -it homeassistant cat /etc/resolv.conf
search local.hass.io
nameserver 172.32.30.3
```

That is definitely not the correct DNS search path, and it doesn't match the one in the host system or the DNS container! This looks similar to #454, but that was fixed years ago.

Furthermore, I can't even fix this by editing /etc/resolv.conf in the container, because it gets overwritten every time HA restarts. As a result, HA is basically nonfunctional for me right now.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Install HAOS in an environment where DHCP provides a local DNS search path.
2. Configure HA with a hostname that matches that search path. (I suspect this part isn't necessary and you can name it whatever you like.)
3. Add an integration like MPD using an unqualified hostname.
4. Observe as it fails to talk to the device. Check the DNS logs and see lots of NXDOMAIN for hostname.local.hass.io rather than whatever your local DNS search path is.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope. I was hoping for a nice "overwriting DNS configuration in container" smoking gun or something.
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

### System information

## System Information

version | core-2023.8.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | America/Toronto
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 30.8 GB
disk_used | 4.9 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Whisper (1.0.0), Piper (1.3.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 14, 2023 at 10:03 PM
-- | --
current_recorder_run | August 23, 2023 at 11:51 PM
estimated_db_size | 15.50 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

ToxicFrog, 2023-08-25:
As a workaround, you can automate the fixing of resolv.conf on startup. Replace `example.net` with your real DNS search path.

There's probably a more elegant way to do this that dynamically fetches the correct search path when the container starts up, but I don't know what it is.

`/config/tools/fix-dns`
```shell
#!/usr/bin/env bash

# For some reason sed -i doesn't work inside the container, so we need
# this little dance
sed -E 's,^search .*,search example.net,' /etc/resolv.conf > /tmp/$$
cat /tmp/$$ > /etc/resolv.conf
rm /tmp/$$
```

`/config/configuration.yaml`
```shell
shell_command:
  fix_dns: 'bash /config/tools/fix-dns'
```

And then in Automations, create one with the trigger "HomeAssistant starts" and the action "call service shell_command.fix_dns".

ToxicFrog, 2023-08-28:
Update: the above doesn't work as well as I might hope, because some integrations fire off before the fix script runs -- so e.g. if you have a cmus media sink, it'll try to connect to it before resolv.conf is repaired, and fail. Some of these, like MQTT, will retry and succeed, but cmus doesn't seem to.

agners, 2023-08-29:
The container in Home Assistant use the DNS plug-in which in turn uses CoreDNS to resolve hostnames. I am transferring the issue to that plug-in.