home-assistant/operating-system #2708: No ECC memory support on AMD64 with generic-x86-64 images

francistheodorecatte, 2023-08-28:
### Describe the issue you are experiencing

The relevant kernel module, amd64_edac.ko, which is necessary for enabling ECC memory on AMD processors, is currently not built and included with the generic-x86-64 releases of Home Assistant OS as of version 10.5.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2023.8.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 938.2 GB
disk_used | 5.2 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2023.8.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 28, 2023 at 12:50 AM
-- | --
current_recorder_run | August 27, 2023 at 8:52 PM
estimated_db_size | 0.20 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

See:
https://www.kernelconfig.io/CONFIG_EDAC_AMD64?q=amd64_edac&kernelversion=6.1.38

agners, 2023-08-31:
From what I understand ECC works by default, this driver is mainly about reporting: It is under `CONFIG_EDAC`, which has as description "EDAC (Error Detection And Correction) reporting". The description says:

> EDAC is a subsystem along with hardware-specific drivers designed to report hardware errors.

Do you know where those reports end up at? Does it make sense to enable the driver if we don't really check those reports? :thinking:

francistheodorecatte, 2023-08-31:
afaik, that kernel module is required for overriding system firmware that doesn't enable ECC memory correctly (or at all), like happens on K8+ architecture AMD CPUs. in my case with an HP T530 thinclient, the GX-series APU supports it, but the UEFI does not report ECC support as being available to the kernel; the only workaround is available via the option ecc_enable_override=1.

agners, 2023-09-05:
Enabled with #2736. You can test it with the next development build (available in a couple of hours on https://os-builds.home-assistant.io/).

closed 2023-09-05 by agners