home-assistant/operating-system #2717: After udating to 10.5 no internet on host, but supervisor does

malosaa, 2023-08-31:
### Describe the issue you are experiencing

[image]

I did this update on both my machines running on proxmox.
exactly after i did the first update to 10.5 i cant update any plugin anymore or homeassistant.

So to be sure i did the same update on my other proxmox machine, and happened the same.

What i did now:
- restart the full proxmox server 
- restart the network on the container
- ha network reload 

none of them work.
However the host has normal internet and i can punt, also when i try to use ping in the ha cli it successfully fings to google with no issue.
I can also just connect to home assistant from outside normal.

regards 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Just upgrade to 10.5 

10.4 had no issue.

### Anything in the Supervisor logs that might be useful for us?

```txt
no because supervisor has internet and no issue
```

### Anything in the Host logs that might be useful for us?

```txt
no see screenshot
```

malosaa, 2023-08-31:
the host log full :

```
Aug 31 18:02:34 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.vUYc3z.mount: Deactivated successfully.
Aug 31 18:02:34 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-1fd9020d3ff99f44f2ae4fb4c1fb14b5533ed73a283196b8a9ee2164333b0bc9-runc.tRnLn9.mount: Deactivated successfully.
Aug 31 18:02:35 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50ec2f6d5dc28c3d598ffa954ae5c46e8e9726df5d09c05c7ae648ccef5aeb1e-runc.StnttR.mount: Deactivated successfully.
Aug 31 18:02:39 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.1ysrVc.mount: Deactivated successfully.
Aug 31 18:02:44 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.jGdmk0.mount: Deactivated successfully.
Aug 31 18:02:47 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7d79a09aedbfc7c2cef03334111dc12620a28f4ace83e90a1b664226e68b8dfc-runc.l7fzyR.mount: Deactivated successfully.
Aug 31 18:02:48 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-52f2a0474fd98724ae308ec2548b1e173e193c2970794fcfc196c6312316aa1f-runc.LTXfQz.mount: Deactivated successfully.
Aug 31 18:02:49 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.KLoYlm.mount: Deactivated successfully.
Aug 31 18:02:54 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.1dzRNk.mount: Deactivated successfully.
Aug 31 18:02:59 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-23f645dddf7acfa733b5f00553af16cb20a4cde423e62638a83407018112fef2-runc.Wml90T.mount: Deactivated successfully.
Aug 31 18:03:00 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.gWWdQp.mount: Deactivated successfully.
Aug 31 18:03:05 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.NjLfAL.mount: Deactivated successfully.
Aug 31 18:03:06 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50ec2f6d5dc28c3d598ffa954ae5c46e8e9726df5d09c05c7ae648ccef5aeb1e-runc.fZMUTX.mount: Deactivated successfully.
Aug 31 18:03:10 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.DI3MnU.mount: Deactivated successfully.
Aug 31 18:03:16 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.MgFiDg.mount: Deactivated successfully.
Aug 31 18:03:17 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7d79a09aedbfc7c2cef03334111dc12620a28f4ace83e90a1b664226e68b8dfc-runc.doGtjY.mount: Deactivated successfully.
Aug 31 18:03:21 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.cqNfqV.mount: Deactivated successfully.
Aug 31 18:03:26 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.bqSGHB.mount: Deactivated successfully.
Aug 31 18:03:29 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-23f645dddf7acfa733b5f00553af16cb20a4cde423e62638a83407018112fef2-runc.pKF8Iq.mount: Deactivated successfully.
Aug 31 18:03:31 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.6sKRYz.mount: Deactivated successfully.
Aug 31 18:03:36 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-1fd9020d3ff99f44f2ae4fb4c1fb14b5533ed73a283196b8a9ee2164333b0bc9-runc.vmdTia.mount: Deactivated successfully.
Aug 31 18:03:36 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.73O0rK.mount: Deactivated successfully.
Aug 31 18:03:46 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.P4yjlu.mount: Deactivated successfully.
Aug 31 18:03:52 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-52f2a0474fd98724ae308ec2548b1e173e193c2970794fcfc196c6312316aa1f-runc.5zOGkO.mount: Deactivated successfully.
Aug 31 18:03:56 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.G73t0K.mount: Deactivated successfully.
Aug 31 18:04:00 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-23f645dddf7acfa733b5f00553af16cb20a4cde423e62638a83407018112fef2-runc.Pb4bZR.mount: Deactivated successfully.
Aug 31 18:04:01 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.jHly5n.mount: Deactivated successfully.
Aug 31 18:04:12 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.dYibnY.mount: Deactivated successfully.
Aug 31 18:04:17 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b09ed61f6feb2f0fb27105772b60375f8caf1613619f22223f82f79d33a86bfe-runc.SrXzpw.mount: Deactivated successfully.
[... cut ...]
Aug 31 18:06:19 homeassistant systemd-timesyncd[406]: Network configuration changed, trying to establish connection.
Aug 31 18:06:19 homeassistant systemd[1]: Started Journal Gateway Service.
Aug 31 18:06:20 homeassistant systemd-journal-gatewayd[776263]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
Aug 31 18:06:20 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7d79a09aedbfc7c2cef03334111dc12620a28f4ace83e90a1b664226e68b8dfc-runc.YqnFT5.mount: Deactivated successfully.
```

malosaa, 2023-08-31:
DNS 

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/corefile.sh
cont-init: info: /etc/cont-init.d/corefile.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun coredns (no readiness notification)
s6-rc: info: service legacy-services successfully started
.:53
.:5553
CoreDNS-1.8.4
linux/amd64, go1.15.15, 053c4d5-dirty
[INFO] 172.30.32.1:36561 - 58081 "A IN core-piper.local.hass.io. udp 42 false 512" NXDOMAIN qr,aa,rd 42 0.000790142s
[INFO] 172.30.32.1:33806 - 43611 "A IN core-piper.local.hass.io. udp 42 false 512" NXDOMAIN qr,aa,rd 42 0.000761334s
[INFO] 172.30.32.1:44163 - 17540 "A IN core-piper.local.hass.io. udp 42 false 512" NXDOMAIN qr,aa,rd 42 0.000756517s
[INFO] 172.30.32.1:54203 - 12609 "A IN core-piper.local.hass.io. udp 42 false 512" NXDOMAIN qr,aa,rd 42 0.000325307s
[INFO] 172.30.32.1:45200 - 54793 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000293795s
[INFO] 172.30.32.1:39216 - 6275 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000317681s
[INFO] 172.30.32.1:52022 - 64020 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000375806s
[INFO] 172.30.32.1:60858 - 4678 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000328886s
[INFO] 172.30.32.1:47794 - 58246 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000318411s
[INFO] 172.30.32.1:57086 - 28448 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000262007s
[INFO] 172.30.32.1:53309 - 8202 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000335894s
[INFO] 172.30.32.1:34239 - 10868 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000282469s
[INFO] 172.30.32.1:39098 - 13316 "A IN [::ffff:a64:b1f].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000410238s
[INFO] 172.30.32.1:54708 - 25750 "A IN [::ffff:a64:b23].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000395755s
[INFO] 172.30.32.1:58407 - 23956 "A IN [::ffff:a64:b23].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000377138s
[INFO] 172.30.32.1:54026 - 43409 "A IN [::ffff:a64:b23].local.hass.io. udp 48 false 512" NXDOMAIN qr,aa,rd 48 0.000365996s
```

malosaa, 2023-08-31:
supervisor

```
23-08-31 18:29:11 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 2 add-ons
23-08-31 18:29:11 WARNING (MainThread) [supervisor.docker.addon] Advanced SSH & Web Terminal running with disabled protected mode!
23-08-31 18:29:11 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_a0d7b954_ssh application
23-08-31 18:29:12 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/ssh/amd64 with version 15.0.7
23-08-31 18:29:12 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning addon_a0d7b954_tor application
23-08-31 18:29:14 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/tor/amd64 with version 4.2.0
23-08-31 18:29:25 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.STARTUP
23-08-31 18:29:25 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-08-31 18:29:27 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-08-31 18:29:27 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/blakeblackshear/frigate-hass-addons repository
23-08-31 18:29:27 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository
23-08-31 18:29:27 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/Apipa169/Assistant-Relay-for-Hassio repository
23-08-31 18:29:27 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/AndBobsYourUncle/hassio-addons/ repository
23-08-31 18:29:27 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository
23-08-31 18:29:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
23-08-31 18:29:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/DivanX10/Home-Assistant-Add-on-Deepstack repository
23-08-31 18:29:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
23-08-31 18:29:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
23-08-31 18:29:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/alexbelgium/hassio-addons repository
23-08-31 18:29:32 WARNING (SyncWorker_1) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Google Assistant Webserver
23-08-31 18:29:34 WARNING (SyncWorker_5) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access) Beta (0.12.0)
23-08-31 18:29:36 WARNING (SyncWorker_5) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Google Assistant Webserver
23-08-31 18:29:37 WARNING (SyncWorker_3) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access) Beta (0.12.0)
23-08-31 18:29:38 INFO (MainThread) [supervisor.store] Loading add-ons from store: 169 all - 0 new - 0 remove
23-08-31 18:29:38 INFO (MainThread) [supervisor.store] Loading add-ons from store: 169 all - 0 new - 0 remove
23-08-31 18:30:16 INFO (MainThread) [supervisor.core] Skiping start of Home Assistant
23-08-31 18:30:16 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' starting 3 add-ons
23-08-31 18:30:16 INFO (SyncWorker_5) [supervisor.docker.manager] Cleaning addon_core_configurator application
23-08-31 18:30:18 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-configurator with version 5.6.0
23-08-31 18:30:18 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_core_piper application
23-08-31 18:30:20 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-piper with version 1.3.2
23-08-31 18:30:20 INFO (SyncWorker_4) [supervisor.docker.manager] Cleaning addon_a0d7b954_grocy application
23-08-31 18:30:23 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/grocy/amd64 with version 0.19.1
23-08-31 18:30:52 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
23-08-31 18:30:52 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
23-08-31 18:30:52 INFO (MainThread) [supervisor.core] Supervisor is up and running
23-08-31 18:30:52 INFO (MainThread) [supervisor.host.info] Updating local host information
23-08-31 18:30:52 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json
23-08-31 18:30:52 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-08-31 18:30:52 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-08-31 18:30:52 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-08-31 18:30:52 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DOCKER_CONFIG/ContextType.SYSTEM
[... cut ...]
23-08-31 19:29:09 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-08-31 19:29:54 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-08-31 19:58:42 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 77c436c7a1bb59ca1b676ca7923206ac35d36799e16446d245605f9112c5aafbb0dd2eaf687fbb4a269b17b775255f7611b1a26d0c139ddf3146f8bd
23-08-31 20:00:13 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
```

malosaa, 2023-08-31:
[image]

malosaa, 2023-08-31:
well wtill im in very doubts,

after i checked that https://data-v2.hacs.xyz/integration/data.json was getting a forbidden report.
i start using my hass ip over vpn and all problems solved.. but i dont want that.

i'm thinking my isp blocked me ? or could it be an other problem ?
for some reason on my laptop using my normal settings i get also forbidden (with vpn it works), but when using my ipad with same settings it works.. 

they all using same dns only other ip.... 

how i connected to vpn is via pfsense using wireshark and adding my ip into the vpn group.

agners, 2023-08-31:
Hm, maybe a local routing issue, or CDN issue?

But this seems not really OS related. You can also try downgrading to rule out any OS involvement (`ha os update --version 10.4`).

malosaa, 2023-09-01:
hi agners, 

well the question is why does it only work wit vpn enabled ?
the vpn routing use the same dns as local without vpn.

i don't know what to do as i dont want to use a vpn..
nobody can help me with this issue, im really out of options here.

thanks for the reply :)

malosaa, 2023-09-02:
Well, really nobody knows the issue, so i say its my isp blocking me, (ordered new internet from other ISP)
i will report back but close the issue as its not related to the previous version, as it was a coincidence when i restarted both containers the issue started after that.

closed 2023-09-02 by malosaa