home-assistant/operating-system #2718: firmware file rtl18852bu_fw.bin no found

emilianogetino, 2023-08-31:
### Describe the issue you are experiencing

The problem
I have migrated my Homeassistant from a Raspberri pi 4 to a mini pc:

Bkouen Mini PC Intel I7-10810U(6 cores 12 threads, up to 4.9 GHz), Mini PC Windows 11 16GB DDR4 512GB M.2 SSD, 4K HDMI Triple Screen, WiFi 6, BT5.2

The problem is that the integrated bluetooch adapter does not work, if I connect via ssh I see this error:
firmware file rtl18852bu_fw.bin no found

In homeassistant in the log these errors are recorded:

2023-08-29 19:53:23.501 WARNING (MainThread) [homeassistant.config_entries] Config entry 'E4:5F:01:28:06:7E' for bluetooth integration not ready yet: Bluetooth adapter None with address E4:5F: 01:28:06:7E not found; Retrying in background
2023-08-29 19:53:25.181 INFO (MainThread) [homeassistant.components.bluetooth.base_scanner] Arcon (B4:8A:0A:22:04:70): Bluetooth scanner has gone quiet for 90s, check logs on the scanner device for more information

2023-08-29 19:57:46.391 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not determine the power state of the Bluetooth adapter hci0 [00:00:00:00:00:00] due to timeout after 5 seconds
2023-08-29 19:57:46.898 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not cycle the Bluetooth adapter hci0 [00:00:00:00:00:00]: [Errno 2] No such file or directory
2023-08-29 19:57:46.898 WARNING (MainThread) [bluetooth_auto_recovery.recover] Bluetooth management socket connection lost: [Errno 22] Invalid argument

Although I don't understand much, it seems to me that the problem is the lack of drivers for this adapter. If so, I would appreciate it if you would include them in future updates.
Thank you.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.45

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Boot the system
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2023-08-29 19:53:23.501 WARNING (MainThread) [homeassistant.config_entries] Config entry 'E4:5F:01:28:06:7E' for bluetooth integration not ready yet: Bluetooth adapter None with address E4:5F:01:28:06:7E not found; Retrying in background
2023-08-29 19:53:25.181 INFO (MainThread) [homeassistant.components.bluetooth.base_scanner] Arcon (B4:8A:0A:22:04:70): Bluetooth scanner has gone quiet for 90s, check logs on the scanner device for more information

2023-08-29 19:57:46.391 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not determine the power state of the Bluetooth adapter hci0 [00:00:00:00:00:00] due to timeout after 5 seconds
2023-08-29 19:57:46.898 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not cycle the Bluetooth adapter hci0 [00:00:00:00:00:00]: [Errno 2] No such file or directory
2023-08-29 19:57:46.898 WARNING (MainThread) [bluetooth_auto_recovery.recover] Bluetooth management socket connection lost: [Errno 22] Invalid argument
```

### Anything in the Host logs that might be useful for us?

```txt
firmware file rtl18852bu_fw.bin no found
```

### System information

## System Information

version | core-2023.8.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | Europe/Madrid
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4716
Installed Version | 1.32.1
Stage | running
Available Repositories | 1276
Downloaded Repositories | 45

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 23 de octubre de 2023, 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
remote_server | eu-central-1-3.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | failed to load: unreachable

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 468.7 GB
disk_used | 50.7 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | UniFi Network Application (2.7.0), Studio Code Server (5.10.1), MariaDB (2.6.1), Duck DNS (1.15.0), ESPHome (2023.8.2), Home Assistant Google Drive Backup (0.111.1), Samba share (10.0.2), Speedtest (1.5.0), Mosquitto broker (6.2.1), Zigbee2MQTT (1.32.2-1), Terminal & SSH (9.7.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 30
views | 20
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 29 de agosto de 2023, 15:25
-- | --
current_recorder_run | 31 de agosto de 2023, 12:12
estimated_db_size | 306.89 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Sonoff</summary>

version | 3.5.2 (2ad1cd7)
-- | --
cloud_online | 6 / 8
local_online | 3 / 3

</details>

darthmeatball, 2023-10-17:
I have the same issue, migrated from RPi3 to MiniPC with Intel N95 and getting the same message "firmware file rtl18852bu_fw.bin no found". Could you guys help?

Umbex79, 2023-10-31:
Qualcuno ha trovato una soluzione?

sairon, 2023-12-19:
Bluetooth firmware is available since OS 11.2, upcoming OS release will also include WiFi firmware (#3003) and driver (rtw89 from #2951).

closed 2023-12-19 by sairon