home-assistant/operating-system #2722: Bluetooth on RTW8821ce for miniPC never working after update 10.5

arripeo, 2023-09-01:
### Describe the issue you are experiencing

After updating to 10.5 finally I was able to see my WiFi RTW8821ce as my hassio is installed on mini PC GK3V (evolution of AK1 pay attention that a large miniPC AK, GK, AM are from the same manufacturer Cyxtech and then they are re-branded)
The problem is now on the Bluetooth that is embedded on the same board. Before the update the Bluetooth devices was recognized perfectly with the MAC now it's seem that the OS is not able to see the device and no log are present, only the integration says that is unable to reconnect the device.
If you want I've the drivers for Linux for RTW8821ce that i've received directly from Cyxtech of you want you can contact directly the service at service@cyxtech.com or for EU utaundnico@hotmail.com for UK yalanda.ma@outlook.com
Thanks in advance

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

for reproduce the issue you have to install the OS10.5

### Anything in the Supervisor logs that might be useful for us?

```txt
no logs by supervisor
```

### Anything in the Host logs that might be useful for us?

```txt
Logger: homeassistant.config_entries
Source: config_entries.py:430
First occurred: 11:48:00 (5 occurrences)
Last logged: 11:48:18

Config entry 'D0:A4:6F:9E:FC:5D' for bluetooth integration not ready yet: Bluetooth adapter None with address D0:A4:6F:9E:FC:5D not found; Retrying in background
```

### System information

Home Assistant 2023.8.4
Supervisor 2023.08.1
Operating System 10.5
Frontend 20230802.1 - latest

makgiber1984, 2023-09-02:
+1: Same problem, I have a BMAX B1 PLUS exclusive for home asistsant and when I install 10.5 the bluetooth stops working, I had to downgrade to 10.4.

MatteoD77, 2023-09-05:
+1 same issue.
After upgrade to 10.5 Bluetooth stop working on BMAX (Mini PC B3 8GB DDR4 256GB Jasper Lake N5095)
Please solve the issue

odinosa, 2023-10-07:
+1:Same problem in a BMAX B2S Celeron N4020

sairon, 2023-10-09:
It seems that adding upstream kernel driver for RTW8821ce in #2659 broke the Bluetooth for some reason. Could you please share the `dmesg` output obtained shortly after the system boots?

sairon, 2023-10-18:
I managed to get my hands on this WiFi card, however both WiFi and Bluetooth work without issues. To move any further, I will need more logs from the affected machines. Please let me know if you need more detailed guidance how to get those.

MatteoD77, 2023-10-20:
This is my log:
Logger: bluetooth_auto_recovery.recover
Source: components/bluetooth/util.py:78
First occurred: 16 ottobre 2023 alle ore 22:43:17 (20666 occurrences)
Last logged: 12:49:28

Could not cycle the Bluetooth adapter hci0 [14:F5:F9:59:56:16]: [Errno 71] Protocol error
Bluetooth adapter hci0 [14:F5:F9:59:56:16] could not be reset:
There are no BT controllers present in the system!
hciNone [14:F5:F9:59:56:16] seems not to exist (anymore), check BT interface mac address in your settings; Available adapters: {}

It's happening that after a hard switch off and switch on the BT restart to work. But after a new upgrade that oblidge to restart home assistant BT stop to work.

MatteoD77, 2023-10-20:
I will add also this log:
Logger: homeassistant.components.bluetooth.scanner
Source: components/bluetooth/scanner.py:353
Integration: Bluetooth (documentation, issues)
First occurred: 16 ottobre 2023 alle ore 22:43:26 (10359 occurrences)
Last logged: 13:02:29

hci0 (14:F5:F9:59:56:16): Failed to restart Bluetooth scanner: hci0 (14:F5:F9:59:56:16): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 223, in _async_start
    await self.scanner.start()  # type: ignore[no-untyped-call]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/__init__.py", line 198, in start
    await self._backend.start()
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/scanner.py", line 178, in start
    self._stop = await manager.passive_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/manager.py", line 527, in passive_scan
    assert_reply(reply)
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 353, in _async_restart_scanner
    await self._async_start()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 305, in _async_start
    raise ScannerStartError(
homeassistant.components.bluetooth.scanner.ScannerStartError: hci0 (14:F5:F9:59:56:16): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist

MatteoD77, 2023-10-26:
Please find a solution.  Bluetooth is not working ?
I must buy a dongle?????

makgiber1984, 2023-11-20:
Still the same problem, still not working on HAOS 11.0 and HAOS 11.1, I have reverted to 10.0 which seems to work correctly.  BMAX b1plus.

sairon, 2023-11-22:
@makgiber1984 Please provide us with the information I asked for before - ideally `dmesg` output from the working and non-working system. I tried putting this card into a different mini PC but I am unable to reproduce it, the driver loads just fine on all OS versions.

makgiber1984, 2023-11-23:
After reinstalling HA OS 11.1, it seems that Bluetooth is working (unlike version 10.5, which gave Bluetooth error).

[image]

When booting the system, or restarting it, everything seems to work correctly, but after a while... all bluetooth devices, in my case, the Mi Plant with XIAOMI BLE integration, starts to stop working, going to "PROBLEM" status,

[image]
[image]
[image]

but the Home assistant log does not seem to say anything about it.....

[image]

Attached is a copy of the log from the time they started to fail (20:22) onwards.  

[image]

I should point out that the Mi plants are connected directly to the Blutooth of the BMax (not through ESP Home), which works exclusively with Home assistant OS,

[image]

I don't know what could be going on (but if I go back to OS version 10.4 or earlier, everything works fine).

Sorry if I am missing data, but I am new with Home Assistant and Linux, if you need any more data tell me where I can look at it.

Thank you very much for your help.

makgiber1984, 2023-11-23:
Nuevos datos arrojados por el registro…

‘’’
Logger: homeassistant.components.bluetooth.scanner
Source: components/bluetooth/scanner.py:353
Integration: Bluetooth (documentation, issues)
First occurred: 22:26:59 (90 occurrences)
Last logged: 23:11:17

hci0 (2C:D2:6B:CC:93:9F): Failed to restart Bluetooth scanner: hci0 (2C:D2:6B:CC:93:9F): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 223, in _async_start
    await self.scanner.start()  # type: ignore[no-untyped-call]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/__init__.py", line 198, in start
    await self._backend.start()
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/scanner.py", line 185, in start
    self._stop = await manager.active_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/manager.py", line 405, in active_scan
    assert_reply(reply)
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 353, in _async_restart_scanner
    await self._async_start()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 305, in _async_start
    raise ScannerStartError(
homeassistant.components.bluetooth.scanner.ScannerStartError: hci0 (2C:D2:6B:CC:93:9F): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist

————

Logger: bluetooth_auto_recovery.recover
Source: components/bluetooth/util.py:78
First occurred: 22:26:51 (181 occurrences)
Last logged: 23:11:16

Could not reset the power state of the Bluetooth adapter hci0 [2C:D2:6B:CC:93:9F] due to timeout after 5 seconds
Could not cycle the Bluetooth adapter hci0 [2C:D2:6B:CC:93:9F]: [Errno 71] Protocol error
Bluetooth adapter hci0 [2C:D2:6B:CC:93:9F] could not be reset:
There are no BT controllers present in the system!
hciNone [2C:D2:6B:CC:93:9F] seems not to exist (anymore), check BT interface mac address in your settings; Available adapters: {}

‘’’

dvirs12, 2023-12-20:
+1 Same issue here on also on b1 max mini-pc with RTL8821CE when upgraded to 10.5.
I've rolled back to 10.4 after some trying...
@makgiber1984 - did you had any luck with 11.1 or 11.2 since your last update?

sairon, 2023-12-21:
@dvirs12 Please read my message above, there's no point of adding "+1" posts without providing relevant logs - without them we can't move any further.

dvirs12, 2023-12-21:
@sairon Sorry for that, I've read the entire thread before I've posted, I thought the provided logs the ones above me gave but it wasn't "top priority" therefore there were no development updates.
I can do a short journey between the versions to provide the logs - I'll be glad if you can let me know what else is needed other than dmesg output from working version and non working one and whether you prefer me to compare with 10.5 or only with latest version (currently the OTA offers me 10.5 so if 11.1 or 11.2 is needed i'll have to do couple of upgrades and hopefully to be able to downgrade afterwards all the way).

Thanks for the help!

makgiber1984, 2023-12-21:
@sairon, @dvirs12, Curiously and to my surprise, I installed Home assistant on a virtual machine in "proxmox" on the same computer, directly with version 10.5 of the OS, (which stopped working after a few hours or days in a direct installation on the BMAX b1 PLUS) and it has been working without any bluettooth interruption for a little over two weeks, so I decided to update to version 11.2 of the OS, and everything works correctly, plus it seems that it even works a little faster on proxmox than live.... (I don't quite understand the reason, but that's how it is...)

So for now I will keep this configuration that seems to be working, but I find it curious that being exactly the same hardware, on proxmox there are no problems, and direct it is.

Sorry for my English....

dvirs12, 2023-12-21:
@makgiber1984 When you use Proxmox or any other Hypervisor to run Virtual Machine some of the hardware IDs and drivers are different since the VM is on many cases becomes "Proxy" between your hardware and the VMs, therefore the drivers of the network for example will be totally different and will present to the guest OS (the HAOS on your case) the virtual network adapters created by the hypervisor (proxmox on your case) and not the actual hardware which the proxmox OS will deal with.

MatteoD77, 2024-02-06:
My issue is still here. Bluetooth is not working. And it seems that there will be no solution in the nearest feature

arripeo, 2024-04-13:
I know this type of bug, i got in the past the same problem than solved by an new OS release, so when I've seen that in 12.1 release they changed the driver of Wifi I've deiced to NOT UPDATE waiting for a new suitable release. So if you want, please vote my new feature request that cloud be, on my opinion, a strong solution for all these type of problem that are very common

https://community.home-assistant.io/t/interface-by-ui-or-cli-for-installing-wifi-propitary-driver-or-linux-compatible/716527

closed 2024-07-21 by github-actions[bot]