home-assistant/operating-system #2724: HAOS Busybox is missing two commands that shell_command busybox has

jt6186, 2023-09-02:
### Describe the issue you are experiencing

I was trying to use tree to generate an index.html based on a directories contents.  Because HAOS does not include tree I installed it using apk add tree.  Figured out my command and put it in a shell_command and got errors because the busybox version of tree does not support the same command line options.

Turns out tree is included in the busybox that shell_command uses but not the one that HAOS uses.

the shell command busybox and the HAOS busybox have the same command set except the HAOS one is missing tree and zcip.

Seems to me that the two busyboxes should have the same command set to prevent just this type of confusion.  And if for some reason they were specifically excluded it should be clearly stated in the shell command documentation that those two commands are missing.  I don't know enough about the implementation to know whether the apk version of tree would be usable if the busybox didn't have tree included or if shell command is intentionally sandboxing the user into only what busybox provides.  But by copying the apk installed version of tree to a different directory I was able to use it.  So I don't really understand the purpose of having two different busyboxes.  It seems to me that if somebody has installed something into /usr/bin then it should be used instead of the busybox version.

Hopefully somebody who understands the implementation can somehow make this less confusing.

Also it turns out that things are way easier to test and debug by having the shell_command point to a bash script.  That way you can make changes to your script to test and not have to restart HAOS for every change.  Unfortunate the reload configuration option does not seem to update the shell commands.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.open a shell in HAOS
2. execute tree command  -->  tree: applet not found
3. Add  to configuration.yaml:
shell_command:
  config_tree: tree
4. restart HAOS
5. Under developer tools execute the shell_command.config_tree service ->  generates a tree of the config directory
...
If you similarly just run busybox in both environments it will list the available commands and you will notice that the shell command busybox has the additional tree and zcip commands available

### Anything in the Supervisor logs that might be useful for us?

```txt
I could not find any logs related to this.  The most helpful information came from running the shell command from the developer tools services tab.
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

agners, 2023-09-05:
First of: What are you using `tree` for to build a smart home? :sweat_smile: 

Second: This is entirely unrelated to Home Assistant OS. Home Assistant Core and all its services (including the processes created by `shell_command`) are running inside a container, which uses an Alpine based image.

>  Because HAOS does not include tree I installed it using apk add tree.

Home Assistant OS is not Alpine based. If you can run `apk` commands, your environment is inside the Core container.

By default Alpine uses Busybox and enables the `tree` applet. This is a design decision of Alpine, and not related to Home Assistant.

As for Home Assistant OS: In typical usage you should never use it's shell. it is built to be very minimalistic, so we don't enable/install unnecessary packages.

In any case, your main issue seems to be that you want to use `tree` through `shell_command`. I recommend to just use `apk add tree && tree` to run the full version of it.

closed 2023-09-05 by agners

jt6186, 2023-09-05:
The full version of tree, not the busybox version, is capable of automatically generating HTML index files, which turns out to be quite handy.  Perhaps there is a more elegant way of viewing files within HA but I have not stumbled across it yet.  It is less the building of the smart home that benefits from such things, but rather the maintaining of it.

Second the issue I was having is that the HAOS busybox has a different set of commands available than the busybox that shell_command has access to.  That is not apparent from the documentation and the solution that I took made things worse as I ended up with two completely incompatible versions of tree in my system, despite both claiming to reside at the exact same location in /usr/bin.  

My suggestion for an HAOS fix was that tree and zcip be enabled in busybox so that at least it works the same as the Home Assistant Core out of the box.  You are not saving much space with just two commands of difference.  My situation was not helped by the fact that the Terminal-SSH add on I'm using has a shell prompt "[core-ssh ~]$" which is apparently false and maybe should be "[haos-ssh ~]$"  If I was in core's shell in the first place none of this confusion would have happened.  Does that prompt naming come from the terminal add-on or from HAOS?  My install originated from some raspberry pi image that I started with that I subsequently backed up and restored to an x86 HAOS install I created from this: https://github.com/home-assistant/operating-system/releases/download/10.5/haos_ova-10.5.qcow2.xz  

As for not using the shell in HAOS,  I was only using the shell to test the commands that I wanted to implement in shell_command.  Given that they are not the same environment it turns out it is not very helpful for testing out shell_commands.  Fortunately I now realize that I can make a test script that I can execute from the Developer Tools->Services that will show me output for anything I care to make into a shell_command.  If you know of a way to shell directly into the HA core container environment to bypass that step that would be an improvement.

As for the ultimate solution for my particular problem, I am torn.  I can install tree using apk add into the HA core, that works.  However it is not clear to me whether that would survive a backup and restore cycle.  Obviously the copy of the tree executable that is below my config directory is backed up but it also does not get updates and maybe has a problem in the future.  

In any case, you have clearly documented why I had the problems I was facing and now they are no longer confusing to me.  I guess I had missed the place where HAOS was explained to be running HA core in a container with its own os tools environment.  Thank you.