home-assistant/operating-system #2725: Missing WiFi Support for AR9462 Qualcomm Atheros Wireless Network Adapter

juliovegap, 2023-09-02:
### Describe the issue you are experiencing

Despite installing `apk add linux-firmware` WiFi Atheros module is not loaded by Alpine Linux Kernel

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.45

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. lshw -C network
...

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2023.8.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | Europe/Madrid
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1277
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 116.7 GB
disk_used | 9.2 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Log Viewer (0.15.1), Spotify Connect (0.12.4), ZeroTier One (0.17.0), Plex Media Server (3.2.4), InfluxDB (4.7.0), Grafana (9.0.3), Let's Encrypt (4.12.9), Duck DNS (1.15.0), SQLite Web (3.9.2), Advanced SSH & Web Terminal (15.0.7)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 de agosto de 2023, 16:50
-- | --
current_recorder_run | 30 de agosto de 2023, 18:19
estimated_db_size | 31.41 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

lshw -C netowrk
-network UNCLAIMED
description: Network controller
product: AR9462 Wireless Network Adapter
vendor: Qualcomm Atheros
physical id: 0
bus info: pci@0000:02:00.0
version: 01
width: 64 bits
clock: 33MHz
capabilities: pm msi pciexpress bus_master cap_list
configuration: latency=0
resources: memory:86800000-8687ffff memory:86880000-8688ffff

agners, 2023-09-05:
Home Assistant OS is not related to Alpine Linux. Alpine is only used in the container images where Home Assistant Core and other components are running.

To add support for Qualcomm Atheros WiFi adapters changes to Home Assistant OS's buildroot code is needed.

juliovegap, 2023-09-06:
Thanks a lot for your answer.
I indeed tried to update Home Assistant OS. Following the documentation [https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md](https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md), I created a pendrive with the file ath9k.conf just calling ath9k driver. I created it in the modules folder of the pendrive with the correct UNIX text formating without success.
I don't know how to debug the issue to undertand if it is a problem of not having the module itself or an issue in the way I call it.

sairon, 2023-12-04:
Driver for AR9462 (ath9k) is enabled in latest development builds, scheduled for release in 11.3.rc1.

closed 2023-12-04 by sairon