home-assistant/operating-system #2746: Yellow: NVMe drive sometimes fails to initialize (Device not ready; aborting reset, CSTS=0x1)

jfroy, 2023-09-08:
### Describe the issue you are experiencing

Occasionally, a Samsung SSD 980 500GB drive (my data drive) in my Yellow does not initialize properly, leading to cascade system service failures and a "zombie" device (no services are running and no watchdog daemon attempts to restart the device).

The kernel log shows:

```
nvme nvme0: Device not ready; aborting reset, CSTS=0x1
nvme nvme0: Removing after probe failure status: -19
```

This eventually leads to the data device unit failing:

```
dev-disk-by\x2dlabel-hassos\x2ddata.device: Job dev-disk-by\x2dlabel-hassos\x2ddata.device/start timed out.
Sep 08 15:52:46 homeassistant systemd[1]: Timed out waiting for device /dev/disk/by-label/hassos-data.
```

Which leads to `hassos-expand.service`, `haos-swapfile.service`, `hassos-apparmor.service`, `hassos-supervisor.service` and many more failing to start.

Issuing a `reboot` command from the HA OS shell does not always solve the issue, but a power cycle on the POE switch does work. It's possible `reboot` does not completely power off the board, whereas a POE switch power cycle will.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Unknown as of now. It's happened twice. I have no idea what placed the NVMe device in a bad state in the first place. It's relatively new module.

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

agners, 2023-09-14:
Unfortunately, it seems that Yellow is (currently) incompatible with the Samsung 980 Pro. We had previous reports of this (e.g. see https://github.com/home-assistant/operating-system/discussions/2235#discussioncomment-5271165) and added it to the list of unsupported NVMes (see https://yellow.home-assistant.io/faq/#which-ssds-are-not-supported).

I am not 100% sure if this is hardware or software related. So it could be that a future kernel update resolves the issue. But I wouldn't count on it :cry: 

Unfortunately, the Raspberry Pi (rather the Broadcom BCM2711 SoC's) PCIe part seems to have some caveats which leads to troubles with some NVMe's unfortunately.

jfroy, 2023-09-14:
I see. I have a spare 960 Pro lying around I will try switching to. The 980 Pro has generally been working, other than the above occasional issue. I also wonder if the 960 Pro will also have issues. It does use an older controller (Samsung Polaris instead of Elpis), PCIe3x4 instead of PCIe4x4, NVMe 1.2 instead of 1.3c -- so it is substantially different.

closed 2023-09-14 by jfroy