home-assistant/operating-system #2747: Installation failed: Watchdog found a problem with Home Assistant API!

victorlang, 2023-09-09:
### Describe the issue you are experiencing

try installation on window and mac, use the same vdi file "haos_ova-10.5.vdi",  alway failed in error below, no matter install on mac or window system.  Is there any [solution?](url)
-----

23-09-09 12:05:33 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
23-09-09 12:07:34 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
23-09-09 12:07:34 ERROR (MainThread) [supervisor.misc.tasks] Home Assistant watchdog reanimation failed!
23-09-09 12:09:34 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
23-09-09 12:11:34 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
23-09-09 12:11:34 ERROR (MainThread) [supervisor.misc.tasks] Home Assistant watchdog reanimation failed!
23-09-09 12:13:34 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

haos_ova-10.5.vdi

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. follow the official guide downloading the vdi file and configuration 
2. load the vm and succ to get http://x.x.x.x:8123 to work
3. access the http://x.x.x.x:8123  and wait for ha to prepair but never end
4.  wath the log on web page, and get the error log above
5. try on mac and window with the same vdi and steps, get the same result and error .

is it a known issue for 10.5 release?
...

### Anything in the Supervisor logs that might be useful for us?

```txt
how to get log of host under the situation above? using vm
```

### Anything in the Host logs that might be useful for us?

```txt
how to get log of host under the situation above? using vm
```

pmoston, 2023-09-09:
Exact same issue here. Windows 10 Pro Host OS on Intel i7 with 4GB and 2vCPU allocated to VM in VirtualBox 7.0, Image: haos_ova_10.5.vdi
The Prep-Screen appears and goes through various steps. 
After a while it goes into a loop. (Full log output from Prep-Screen attached).
Setup was performed according to https://www.home-assistant.io/installation/windows

23-09-09 19:16:20 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2023.9.1.
23-09-09 19:16:20 INFO (MainThread) [supervisor.host.info] Updating local host information
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.MULTIPLE_DATA_DISKS/ContextType.SYSTEM
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-09-09 19:16:20 INFO (MainThread) [supervisor.host.services] Updating service information
23-09-09 19:16:20 INFO (MainThread) [supervisor.host.network] Updating local network information
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DOCKER_CONFIG/ContextType.SYSTEM
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.module] Create new suggestion SuggestionType.CREATE_FULL_BACKUP - ContextType.SYSTEM / None
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.module] Create new issue IssueType.NO_CURRENT_BACKUP - ContextType.SYSTEM / None
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-09-09 19:16:20 INFO (MainThread) [supervisor.host.manager] Host information reload completed
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-09-09 19:16:20 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-09-09 19:18:20 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
23-09-09 19:20:20 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
23-09-09 19:20:20 ERROR (MainThread) [supervisor.misc.tasks] Home Assistant watchdog reanimation failed!

[ha_config.log](https://github.com/home-assistant/operating-system/files/12566368/ha_config.log)

pmoston, 2023-09-09:
Installation on 10.4 works, however. Even though it also logs one Watchdog API miss.
https://github.com/home-assistant/operating-system/releases/download/10.4/haos_ova-10.4.vdi.zip

pmoston, 2023-09-09:
Perhaps it was a temporary issue? After successully installing 10.4 I retried with 10.5. an now that worked as well.

moebis, 2023-09-10:
I'm having the same issue. Fresh install on a Pi 3b+, fresh 64GB A2 class sdcard. At the end of the installation I see this:

```
23-09-10 11:39:21 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-09-10 11:41:21 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
23-09-10 11:43:21 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
23-09-10 11:43:21 ERROR (MainThread) [supervisor.misc.tasks] Home Assistant watchdog reanimation failed!
23-09-10 11:45:21 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
```

....and it just keeps going. How can I get this installed?

H-CrazyPan, 2023-09-10:
same question，ESXI 7.0u3，ova-10.5.vmdk （0.0）google so longtime ,should search issue early

agners, 2023-09-10:
At least in @pmoston's case it seems that the Supervisor is still downloading the actual Home Assistant Core (`23-09-09 19:16:20 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2023.9.1.` line never completed). 

Context: The Home Assistant OS comes with a minimal version preinstalled (landing page). On first boot, the latest version of full Home Assistant Core is downloaded. It seems that in these cases the download is still running, or maybe got interrupted.

marcus13371337, 2023-09-11:
I have the exact same problem with a  Pi 3b+

Lassska, 2023-09-12:
Same
...supervisor.misc.tasks] Home Assistant watchdog reanimation failed!
...miss API response from Home Assistant

slothking87, 2023-09-13:
Also having what seems to be the same error. HA has been very weird lately often freezing and needing multiple restarts to come back up. 

[supervisor.misc.tasks] Watchdog miss API response from Home Assistant
[supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
[supervisor.misc.tasks] Home Assistant watchdog reanimation failed!

--RPI Reboots --

[supervisor.misc.tasks] Watchdog miss API response from Home Assistant
[supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
[supervisor.misc.tasks] Home Assistant watchdog reanimation failed!

Lassska, 2023-09-13:
I confirm that only waiting gave results. I looked for an older version 9.5 and left rPi 3 overnight. In the morning it was ok

agners, 2023-09-14:
Yesterday a new Supervisor release 2023.09.2 has been released to the stable channel with quite some bug fixes. On a new installation, the Supervisor is first updated to it's latest version. Can you try again with HAOS 10.5 to see if this latest Supervisor release might have resolved the issues?

Lassska, 2023-09-14:
HiI will try!SZILAGYI Tibi Peter

Trimis din Yahoo Mail pe Android

bschatzow, 2023-09-14:
> Yesterday a new Supervisor release 2023.09.2 has been released to the stable channel with quite some bug fixes. On a new installation, the Supervisor is first updated to it's latest version. Can you try again with HAOS 10.5 to see if this latest Supervisor release might have resolved the issues?

I saw this error after the supervisor update and the core update during the restart. It finally restarted and doesn't seem to be an issue. Not sure what is causing the issue but this is the first time I saw it

JVSDR92, 2023-09-14:
Today i faced the same problem after the 10.5 update.
I fixed it by going to the supervisor tab and  restart core and restart host multiple times. 
Maybe this will also work for people who have the same problem

vmassahi, 2023-09-15:
23-09-15 13:25:36 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
23-09-15 13:27:36 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
23-09-15 13:27:36 ERROR (MainThread) [supervisor.misc.tasks] Home Assistant watchdog reanimation failed!
23-09-15 13:29:36 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant

Having the same issue here. This is a fresh install on a Proxmox VM 
Home assistant OS version: 10.5 Generic x86-64

wilksy, 2023-09-15:
Same issue, First ever install on pi 4. 

I just waited...and waited... and it sorted itself out. 

make a cup of tea and chill.

Lassska, 2023-09-15:
You try to wait 3-4 hour?

Trimis din Yahoo Mail pe Android

wilksy, 2023-09-15:
I just waited. Took about 2 hours and then the create account portal was available.

kemege, 2023-09-17:
I have the same problem, waited for several hours and still nothing changes. The reason seems to be that `https://ghcr.io` cannot be accessed in here, and the docker image `qemux86-64-homeassistant` cannot be pulled.

Is there any workarounds possible, like setting a proxy for `docker pull` or so on?

wilksy, 2023-09-17:
> I have the same problem, waited for several hours and still nothing changes. The reason seems to be that `https://ghcr.io` cannot be accessed in here, and the docker image `qemux86-64-homeassistant` [...]

https://github.com/opensafely-core/proxy

kemege, 2023-09-17:
> https://github.com/opensafely-core/proxy

Thanks for your reply, as it seems to work like a reverse proxy, I assume I should use command like `docker pull docker-proxy.opensafely.org/home-assistant/qemux86-64-homeassistant` to use it?

However, that does not seem possible in HAOS, as `ghcr.io/home-assistant/qemux86-64-homeassistant` seems hard-coded into the system, and I haven't found ways to modify it.

wilksy, 2023-09-17:
Been few years since I've worked at this level...but maybe... Edit your hosts file to redirecte instead, then any script on the machine will redirect to the opensafeley link??

Lassska, 2023-09-17:
ha> network info

Trimis din Yahoo Mail pe Android 

https://github.com/opensafely-core/proxy

Lassska, 2023-09-17:
> Thanks for your reply, as it seems to work like a reverse proxy, I assume I should use command like `docker pull docker-proxy.opensafely.org/home-assistant/qemux86-64-homeassistant` to use it? [...]

Ha> network info

franderlink, 2023-09-17:
`Sep 17 13:32:37 aml-s905x-cc hassio_supervisor[5923]: 23-09-17 09:32:37 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
Sep 17 13:32:39 aml-s905x-cc hassio_supervisor[5923]: 23-09-17 09:32:39 ERROR (MainThread) [supervisor.homeassistant.core] Home Assistant has crashed!
Sep 17 13:32:39 aml-s905x-cc hassio_supervisor[5923]: 23-09-17 09:32:39 ERROR (MainThread) [supervisor.homeassistant.core] Watchdog restart of Home Assistant failed!
Sep 17 13:32:54 aml-s905x-cc hassio_audio[5923]: E: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes inc>
Sep 17 13:37:13 aml-s905x-cc hassio_supervisor[5923]:
Sep 17 13:42:15 aml-s905x-cc hassio_supervisor[5923]:
Sep 17 13:47:16 aml-s905x-cc hassio_supervisor[5923]:
Sep 17 13:52:17 aml-s905x-cc hassio_supervisor[5923]:
Sep 17 13:57:18 aml-s905x-cc hassio_supervisor[5923]:
Sep 17 14:02:19 aml-s905x-cc hassio_supervisor[5923]:
`
Expecting an installation yesterday is absurd this 

Not yet showing the page of user creation by the "IP local":8123

lovelyelfpop, 2023-09-18:
hyper-v, same problem here with 10.5.

Lassska, 2023-09-18:
I notice that at night everything goes faster. The loop lasted less than 1 hour... 0:26 to 1:23!

"Welcome!
Are you ready to awaken your home, reclaim your privacy and join a worldwide community of tinkerers?"

bschatzow, 2023-09-19:
@agners Just did a hacs integration that required a restart.  Error is sill there.  Need to figure out if the os update or supervisor update is causing this?  Restarts are very slow.  What additional information is needed.

Lassska, 2023-09-20:
I think I solved the mystery and that repeating the loop depends on the memory support as model and/or size In my case I had these results: 1. loop ~ 4 hours with USB stick 64Gb 2. loop ~ 1 hour with USB stick 32Gb 3. loop ~ 6 minutes with SD card 16Gb (practically from the start in 13 minutes the login page appeared) It remains for you to test variants...

DPO99, 2023-09-22:
Same Problem with 10.4 and 10.5 in RaspberryPi 3 :/

bschatzow, 2023-09-23:
Simple restart and I am getting
```
23-09-23 14:34:20 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
23-09-23 14:34:20 ERROR (MainThread) [supervisor.misc.tasks] Home Assistant watchdog reanimation failed!
# 23-09-23 14:34:20 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a
problem with Home Assistant API!
-sh: syntax error: unexpected "("
# 23-09-23 14:34:20 ERROR (MainThread) [supervisor.misc.tasks] Home Assistant wa
tchdog reanimation failed!
-sh: syntax error: unexpected "("
```

Lassska, 2023-09-23:
> Same Problem with 10.4 and 10.5 in RaspberryPi 3 :/

Use SD Card or USB Stick in RPi3?

Embosen, 2023-09-26:
ESXI6.7 same problem here with 10.5.

Lassska, 2023-09-26:
> ESXI6.7 same problem here with 10.5.

Use SD Card or USB Stick in RPi3?