home-assistant/operating-system #2752: "no host internet connection" - "can't edit network settings" - "No space left"

DunklerPhoenix, 2023-09-14:
### Describe the issue you are experiencing

Heho
Home Assistant often can't install updates, addons or anything else anymore because it says "no host internet connection"
I tried some solutions like changing the DNS Server, but I can't save the settings.
If I try i get the message:
`Can't update config on eth0: failed to update connection: error writing to file '/etc/NetworkManager/system-connections/Supervisor eth0.nmconnection': failed to create file /etc/NetworkManager/system-connections/Supervisor eth0.nmconnection.ZNJC91: No space left on device`

EDIT: Hardware is HA Blue

```
➜  ~ df -h 
Filesystem                Size      Used Available Use% Mounted on
overlay                 113.9G     21.4G     86.8G  20% /
devtmpfs                  1.7G         0      1.7G   0% /dev
tmpfs                     1.8G         0      1.8G   0% /dev/shm
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /addons
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /data
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /backup
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /config
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /ssl
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /media
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /share
tmpfs                   757.4M      2.0M    755.3M   0% /run/docker.sock
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /etc/asound.conf
tmpfs                   757.4M      2.0M    755.3M   0% /run/dbus
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /run/audio
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /etc/resolv.conf
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /etc/hostname
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /etc/hosts
tmpfs                     1.8G         0      1.8G   0% /dev/shm
tmpfs                   757.4M      2.0M    755.3M   0% /run/log/journal
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /var/log/journal
/dev/mmcblk1p4          113.9G     21.4G     86.8G  20% /etc/pulse/client.conf
tmpfs                     1.8G         0      1.8G   0% /proc/asound
devtmpfs                  1.7G         0      1.7G   0% /proc/keys
devtmpfs                  1.7G         0      1.7G   0% /proc/timer_list
tmpfs                     1.8G         0      1.8G   0% /sys/firmware
```

Do you know what this can be?
After a system restart the host internet connection is for some time available again

### What type of installation are you running?

Home Assistant OS

### Which operating system are you running on?

Home Assistant Operating System

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[33m23-08-12 22:32:11 WARNING (MainThread) [supervisor.addons.options] Option 'require_ssl' does not exist in the schema for Node-RED (a0d7b954_nodered)[0m
[32m23-08-12 22:32:12 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/node-red/aarch64 with version 14.4.4[0m
[33m23-08-12 22:32:13 WARNING (MainThread) [supervisor.addons.options] Option 'dark_mode' does not exist in the schema for Node-RED (a0d7b954_nodered)[0m
[33m23-08-12 22:32:13 WARNING (MainThread) [supervisor.addons.options] Option 'require_ssl' does not exist in the schema for Node-RED (a0d7b954_nodered)[0m
[32m23-08-12 22:32:13 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/appdaemon/aarch64 with version 0.13.4[0m
[32m23-08-12 22:32:15 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/log-viewer/aarch64 with version 0.15.1[0m
[32m23-08-12 22:32:17 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on 2ca71ec5/aarch64-addon-bw-cli-secrets with version 1.4.6[0m
[33m23-08-12 22:32:17 WARNING (MainThread) [supervisor.addons.options] Option 'dark_mode' does not exist in the schema for Node-RED (a0d7b954_nodered)[0m
[33m23-08-12 22:32:17 WARNING (MainThread) [supervisor.addons.options] Option 'require_ssl' does not exist in the schema for Node-RED (a0d7b954_nodered)[0m
[... 172 lines cut ...]
</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2. August 2023 um 22:58
-- | --
current_recorder_run | 9. August 2023 um 15:02
estimated_db_size | 1743.35 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.2 (2ad1cd7)
-- | --
cloud_online | 1 / 3
local_online | 1 / 1

</details>

### Supervisor diagnostics

[config_entry-hassio-1dd8512d8dc2f72a08934be72bd4720b.json.txt](https://github.com/home-assistant/supervisor/files/12328212/config_entry-hassio-1dd8512d8dc2f72a08934be72bd4720b.json.txt)

bdraco, 2023-09-14:
Can someone with the issue send a profiler callgrind file using the profiler.start service?

https://www.home-assistant.io/integrations/profiler/

The file only contains the names of the python code function calls and names of the libraries installed so it should be safe to post here unless you don't want people to know which integrations you have installed

rexbron, 2023-09-14:
@bdraco I have attached the callgrind.out as zip (to allow for uploading to the github comment). Let me know if this meets your requirements.

[callgrind.out.1694691447678308.zip](https://github.com/home-assistant/operating-system/files/12608319/callgrind.out.1694691447678308.zip)

DunklerPhoenix, 2023-09-14:
[callgrind.out.1694691960524360.zip](https://github.com/home-assistant/operating-system/files/12608680/callgrind.out.1694691960524360.zip)

Have fun looking at this mess xD

bdraco, 2023-09-14:
> @bdraco I have attached the callgrind.out as zip (to allow for uploading to the github comment). Let me know if this meets your requirements.
> 
> [callgrind.out.1694691447678308.zip](https://github.com/home-assistant/operating-system/files/12608319/callgrind.out.1694691447678308.zip)

Thanks @rexbron . Looks good. I don't see a massive amount of bluetooth traffic in the profile, but it might be that bluetooth has failed due to the issue.  I suspect https://github.com/home-assistant/operating-system/pull/2751 will fix the issue for you. It would be good to get a new profile to check everything again once you have installed an OS build with 2751 in it.

bdraco, 2023-09-14:
> [callgrind.out.1694691960524360.zip](https://github.com/home-assistant/operating-system/files/12608680/callgrind.out.1694691960524360.zip)
> 
> Have fun looking at this mess xD

@DunklerPhoenix Your profile is much more interesting with 3719 bluetooth advertisements per minute.  This will be faster in 2023.10.x as we recently fixed some performance bottlenecks there.

It also looks like the `mqtt_room` integration is not well optimized and is responsible for the bulk of your run time.

rexbron, 2023-09-15:
@DunklerPhoenix You can run the command below from the physical keyboard to manually clear the cache

`find /var/lib/bluetooth/*/cache -mindepth 1 -type f -atime +7  -exec rm {} \;`

rexbron, 2023-09-15:
@bdraco 
[callgrind.out.1694785421594926.zip](https://github.com/home-assistant/operating-system/files/12619995/callgrind.out.1694785421594926.zip)
Ran the cache clearing command yesterday and re-ran the profiler this morning

DunklerPhoenix, 2023-09-15:
`find /var/lib/bluetooth/*/cache -mindepth 1 -type f -atime +7  -exec rm {} \;`

Hahahah the command took a really long time to finish. Nearly one minute xD

Btw it worked. I can save my network settings again <3

bdraco, 2023-09-15:
> @bdraco [callgrind.out.1694785421594926.zip](https://github.com/home-assistant/operating-system/files/12619995/callgrind.out.1694785421594926.zip) Ran the cache clearing command yesterday and re-ran the profiler this morning

Thanks for that. Your profile looks very clean but there is some I/O in your event loop. I fixed it in https://github.com/home-assistant/core/pull/100455.

rexbron, 2023-12-15:
@DunklerPhoenix This should be marked as closed with the release of HAOS

closed 2023-12-15 by DunklerPhoenix

agners, 2023-12-20:
It is part of HAOS 11.0 and newer (see https://github.com/home-assistant/operating-system/releases/tag/11.0.rc1 changelog)