home-assistant/operating-system #2754: Getting Wake on Lan to work

StSaens, 2023-09-15:
### Describe the issue you are experiencing

I'm running HAOS on a HP t620, which supports wol. It's also enabled in the bios. But when sending a wol-packet it does not wake. I read (https://dietpi.com/forum/t/wake-on-lan-wol-hp-t620-dietpi-how-to-enable-it-permanently-on-interface/16054) that using ethtool it could be that it has to be enabled in the driver(?) as well. I didn't find a way to check what the settings are however on HAOS (even when ssh to 22222).

You might wonder 'but haos has to run permanently so why use wol?!??'
Well, there's a scenario when running with a UPS where 
1. the power goes down, 
2. bat goes low, 
3. HAOS shuts down, 
4. and before the UPS goes down, the power is restored. So there's no powercycle in this scenario. And wol would start HAOS back up.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-09-15 06:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system[0m
[32m23-09-15 06:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server[0m
[32m23-09-15 06:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system[0m
[32m23-09-15 06:31:38 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-09-15 06:31:38 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running[0m
[32m23-09-15 06:31:39 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-09-15 06:31:39 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running[0m
[32m23-09-15 06:31:39 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m23-09-15 06:32:56 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-09-15 06:33:10 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository[0m
[32m23-09-15 06:33:10 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository[0m
[32m23-09-15 06:33:10 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository[0m
[32m23-09-15 06:33:10 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository[0m
[32m23-09-15 06:33:10 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository[0m
[32m23-09-15 06:33:16 INFO (MainThread) [supervisor.store] Loading add-ons from store: 75 all - 0 new - 0 remove[0m
[32m23-09-15 06:33:16 INFO (MainThread) [supervisor.store] Loading add-ons from store: 75 all - 0 new - 0 remove[0m
[32m23-09-15 06:33:55 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-09-15 06:33:55 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m23-09-15 07:02:56 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running[0m
[32m23-09-15 07:31:39 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m23-09-15 07:32:46 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json[0m
[32m23-09-15 07:32:56 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[... 195 lines cut ...]
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.3.1), Terminal & SSH (9.7.1), Zigbee2MQTT (1.33.0-1), Home Assistant Google Drive Backup (0.111.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 9, 2023 at 08:36
-- | --
current_recorder_run | September 14, 2023 at 15:31
estimated_db_size | 276.11 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

ikifar2012, 2023-09-16:
Take a look at the NetworkManager section here, it might help https://wiki.archlinux.org/title/Wake-on-LAN#:~:text=interface%20wol%20g%27-,NetworkManager,-NetworkManager%20provides%20Wake

StSaens, 2023-09-16:
Thanks! The NetworkManager part in that helped me.
It showed that the wol setting was "default". And I changed it to "magic".
Then it worked.

Will this change persist over HAOS updates? If not, can this be set to "magic" in HAOS by default? Or is there a way I can auto change this to "magic" when a HAOS update is installed?

agners, 2023-09-19:
> Will this change persist over HAOS updates? 

Yes, we generally leave other NetworkManager untouched.

> If not, can this be set to "magic" in HAOS by default? Or is there a way I can auto change this to "magic" when a HAOS update is installed?

It typically does lead to higher power off usage, so I rather prefer to keep the default for now. We could also consider add this as an option to the CLI.

closed 2023-09-19 by agners

JonSnow88, 2024-01-22:
> Thanks! The NetworkManager part in that helped me. It showed that the wol setting was "default". And I changed it to "magic". Then it worked.
>
> Will this change persist over HAOS updates? If not, can [...]

I applied the same changes but I still cannot get this to work in HAOS, anything else I canncheck?

I have another identical device which I use for Proxmox and need to use ethtool to get WOL working in proxmox, so I know for sure my BIOS settings are correct.

JonSnow88, 2024-01-22:
> Yes, we generally leave other NetworkManager untouched.
>
>
> It typically does lead to higher power off usage, so I rather prefer to keep the default for now. We could also consider add this as an [...]

If you have WOL configured in BIOS typically you would want this enabled by default and I would think the BIOS settings would keep the system powered up anyways, my x86 system uses 1-2W with WOL enabled in BIOS and HA fully shutting down. 

So please add this as a CLI option, it took me days to find this fix. My HA is on bare metal, I want this system on 24/7/365. It's extremely convenient to be able to use WOL from my phone when you accidently select shutdown instead of reboot when working on the system.

stantond, 2025-01-15:
This would be great to have as a configuration in the Settings UI to make a fallback process much easier to set up.

A further iteration would be to integrate this into the Mobile app so the user can recover from a power outage without any particular special measures.

dhammy925, 2025-02-19:
/necro sorta

It took me some time to realize this, but in order to access the host system from the terminal you have to type "login" and press enter. Leaving this here for anyone else who doesn't know.

> Take a look at the NetworkManager section here, it might help https://wiki.archlinux.org/title/Wake-on-LAN#:~:text=interface%20wol%20g%27-,NetworkManager,-NetworkManager%20provides%20Wake

This worked for me, fingers crossed it wont break anytime soon.

> This would be great to have as a configuration in the Settings UI to make a fallback process much easier to set up.
> 
> A further iteration would be to integrate this into the Mobile app so the user can recover from a power outage without any particular special measures.

I'd like to echo this sentiment -- a UI element in the settings and the mobile app would be a welcome addition