home-assistant/operating-system #2766: HAOS 10.4 to 10.5 shows successful, reboots, stays on 10.4, suggests 10.5 upgrade

GR0DZ1LLA, 2023-09-21:
### Describe the issue you are experiencing

On HAOS 10.4, Settings suggest upgrade to 10.5.  I do it, it reboots itself, still asks for upgrade to 10.5.  Tried many times.  Have gotten some help on Discord getting logs.

[32m23-09-21 10:41:45 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/10.5/haos_generic-x86-64-10.5.raucb[0m
[32m23-09-21 10:41:50 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-10.5.raucb[0m
[32m23-09-21 10:42:07 INFO (MainThread) [supervisor.os.manager] Install of Home Assistant Operating System 10.5 success[0m

Then after reboot:
[32m23-09-21 10:44:12 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 10.4 / BootSlot A[0m

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Go to Settings
2. Select update
3. Select Install
4. Allow system to reboot itself
5. Log in and still on 10.4 with the update to 10.5 being suggested
...

### Anything in the Supervisor logs that might be useful for us?

```txt
https://dpaste.org/AzRLW#L2932

[32m23-09-21 10:44:12 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 10.4 / BootSlot A[0m
[32m23-09-21 10:44:17 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository[0m
[32m23-09-21 10:44:17 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository[0m
[32m23-09-21 10:44:17 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository[0m
[32m23-09-21 10:44:21 INFO (MainThread) [supervisor.store] Loading add-ons from store: 71 all - 71 new - 0 remove[0m
[32m23-09-21 10:44:21 INFO (MainThread) [supervisor.addons] Found 3 installed add-ons[0m
[32m23-09-21 10:44:21 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-zwave_js with version 0.1.90[0m
[32m23-09-21 10:44:21 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.6.0[0m
[32m23-09-21 10:44:21 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/amd64 with version 15.0.8[0m
[32m23-09-21 10:44:21 INFO (MainThread) [supervisor.backups.manager] Found 10 backup files[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.discovery] Loaded 1 messages[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-09-21 10:44:22 INFO (MainThread) [__main__] Running Supervisor[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.addons] Phase 'initialize' starting 0 add-ons[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.addons] Phase 'system' starting 0 add-ons[0m
[32m23-09-21 10:44:22 INFO (MainThread) [supervisor.addons] Phase 'services' starting 2 add-ons[0m
[32m23-09-21 10:44:23 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-zwave_js with version 0.1.90[0m
[33m23-09-21 10:44:23 WARNING (MainThread) [supervisor.docker.addon] Advanced SSH & Web Terminal running with disabled protected mode![0m
[32m23-09-21 10:44:24 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/ssh/amd64 with version 15.0.8[0m
[32m23-09-21 10:44:29 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_zwave_js[0m
[32m23-09-21 10:44:29 INFO (MainThread) [supervisor.core] Start Home Assistant Core[0m
[32m23-09-21 10:44:29 INFO (SyncWorker_3) [supervisor.docker.manager] Starting homeassistant[0m
[32m23-09-21 10:44:30 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready[0m
[32m23-09-21 10:44:49 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state startup[0m
[32m23-09-21 10:44:49 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[... 91 lines cut ...]
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS (0.1.90), File editor (5.6.0), Advanced SSH & Web Terminal (15.0.8)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 12, 2023 at 11:43 PM
-- | --
current_recorder_run | September 21, 2023 at 10:44 AM
estimated_db_size | 16.86 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

agners, 2023-09-26:
After the update, when checking the screen of our system, do see if Home Assistant attempts to boot from "Slot B"? A boot menu should appear briefly with Slot B selected.

GR0DZ1LLA, 2023-09-26:
Ok I see it.  It tries to boot from Slot A.  Should I manually try Slot B?

GR0DZ1LLA, 2023-09-26:
Alright, I took a backup and then attempted the update again.  When it went to reboot automatically, I interrupted the boot and manually selected Slot B and it worked.  I now have 10.5 running.  However, after a reboot, it booted Slot A automatically and I'm back to 10.4.

GR0DZ1LLA, 2023-10-09:
@agners any further help on this considering what i was able to do?

agners, 2023-10-09:
> Ok I see it. It tries to boot from Slot A. Should I manually try Slot B?
> 
> FYI, I run HAOS on an APU2 that is head-less. Over serial, I can see this boot menu. It says `Booting 'Slot A (OK=0 TRY=0)'`
> 

Hm I see, do you have a serial console in Home Assistant? Probably not, afaik the serial console for the generic-x86-64 image is not enabled :cry: 

You can setup an SSH access via port 22222, see [SSH access to the host](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host).

Using the access, can you check this file?
```
# cat /mnt/boot/EFI/BOOT/grubenv
```

GR0DZ1LLA, 2023-10-09:
➜  ~ cat /mnt/boot/EFI/BOOT/grubenv
cat: can't open '/mnt/boot/EFI/BOOT/grubenv': No such file or directory
➜  ~

GR0DZ1LLA, 2023-10-09:
@agners I did this a while ago and am just remembering that I had to do this to get it to work on the APU4:

Connect to the APU
Serial to the box, power on

The APU will not see this on first boot.  Need to enter into the settings, and exit out.  Upon the next reboot the drive will appear.

`mount /dev/sda1 /mnt

grub-install --boot-directory=/mnt/boot --force /dev/sda

nano /mnt/boot/grub/grub.cfg

set root=(hd0,gpt1)
configfile (hd0,gpt1)/efi/boot/grub.cfg

umount /mnt`

Is this why I can't get an upgrade to work effectively?

GR0DZ1LLA, 2023-10-09:
https://community.home-assistant.io/t/add-support-for-apu-devices/449610/7

closed 2024-01-16 by github-actions[bot]