home-assistant/operating-system #2776: Missing driver for ALL mediatek wifis/devices

patryk4815, 2023-09-28:
### Describe the issue you are experiencing

Missing drivers for **ALL** mediatek devices.
For example MT7922 need `mt7921e.ko`
Please enable in kernel as module: `CONFIG_MT7921E=m`

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

n/a

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

patryk4815, 2023-09-28:
Manually recompiled kernel + added firmware:

For some reason i had to copy full director `./output/build/linux-firmware-20230515/mediatek` into image.
❗❗Because there is missing `BT_RAM_CODE_MT7922_1_1_hdr.bin` firmware file.
It is missing in buildroot, i don't see flag for enabling it - https://github.com/buildroot/buildroot/blob/master/package/linux-firmware/linux-firmware.mk

[image]

Diff for build:
```
diff --git a/buildroot-external/board/pc/ova/kernel.config b/buildroot-external/board/pc/ova/kernel.config
index f94e70358..4844be610 100644
--- a/buildroot-external/board/pc/ova/kernel.config
+++ b/buildroot-external/board/pc/ova/kernel.config
@@ -111,3 +111,34 @@ CONFIG_IXGBEVF=m
 CONFIG_I40EVF=m
 CONFIG_MLX5_CORE=m
 CONFIG_MLX5_CORE_EN=y
+
+
+CONFIG_MT7601U=m
+CONFIG_MT76_CORE=m
+CONFIG_MT76_USB=m
+CONFIG_MT76_SDIO=m
+CONFIG_MT76x02_LIB=m
+CONFIG_MT76x02_USB=m
+CONFIG_MT76_CONNAC_LIB=m
+CONFIG_MT76x0_COMMON=m
+CONFIG_MT76x0U=m
+CONFIG_MT76x0E=m
+CONFIG_MT76x2_COMMON=m
+CONFIG_MT76x2E=m
+CONFIG_MT76x2U=m
+CONFIG_MT7603E=m
+CONFIG_MT7615_COMMON=m
+CONFIG_MT7615E=m
+CONFIG_MT7663_USB_SDIO_COMMON=m
+CONFIG_MT7663U=m
+CONFIG_MT7663S=m
+CONFIG_MT7915E=m
+CONFIG_MT7921_COMMON=m
+CONFIG_MT7921E=m
+CONFIG_MT7921S=m
+CONFIG_MT7921U=m
+CONFIG_BT_MTK=m
+CONFIG_BT_MTKSDIO=m
+CONFIG_BT_MTKUART=m
+CONFIG_BT_HCIBTUSB_MTK=y
+
diff --git a/buildroot-external/configs/ova_defconfig b/buildroot-external/configs/ova_defconfig
index 9663324a8..2acabe74e 100644
--- a/buildroot-external/configs/ova_defconfig
+++ b/buildroot-external/configs/ova_defconfig
@@ -54,6 +54,9 @@ BR2_PACKAGE_LINUX_FIRMWARE_IWLWIFI_8000C=y
 BR2_PACKAGE_LINUX_FIRMWARE_IWLWIFI_8265=y
 BR2_PACKAGE_LINUX_FIRMWARE_IWLWIFI_9XXX=y
 BR2_PACKAGE_LINUX_FIRMWARE_MEDIATEK_MT7601U=y
+BR2_PACKAGE_LINUX_FIRMWARE_MEDIATEK_MT7650=y
+BR2_PACKAGE_LINUX_FIRMWARE_MEDIATEK_MT7610E=y
+BR2_PACKAGE_LINUX_FIRMWARE_MEDIATEK_MT76X2E=y
 BR2_PACKAGE_LINUX_FIRMWARE_QUALCOMM_6174A_BT=y
 BR2_PACKAGE_LINUX_FIRMWARE_RALINK_RT73=y
 BR2_PACKAGE_LINUX_FIRMWARE_RALINK_RT2XX=y
 ```

sairon, 2023-12-04:
Enabled in latest development builds, scheduled for release in 11.3.rc1.

patryk4815, 2023-12-04:
@sairon firmware `BT_RAM_CODE_MT7922_1_1_hdr.bin` still missing

sairon, 2023-12-19:
It took a bit longer, because we needed to make room for the extra firmwares first, but it's finally being added in #3003.

patryk4815, 2023-12-19:
@sairon 🎉