home-assistant/operating-system #2777: failed to start at last update core

wds657, 2023-09-28:
### Describe the issue you are experiencing

fail to start haos at last update beta core, the network is a bucle forever
i try slot a or b and nothing effect

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

update last beta core

### Anything in the Supervisor logs that might be useful for us?

```txt
no access
```

### Anything in the Host logs that might be useful for us?

```txt
no access
```

### System information

no access

spiderzero3, 2023-09-28:
I think I'm having the same issue. I'm on HA OS, updated to 2023.10.0b0 from 2023.9.3, updated to OS 10.5, then Home Assistant was no longer accessible.

Tried troubleshooting through the CLI on the machine running HA OS, but couldn't get anything concrete.

Any time I tried to interact with the core at all ("ha core stop", "ha core rebuild", "ha core repair"), I'd always get an error saying "Another job is running for job group home_assistant_core".

Checking the Core logs showed several lines that seem normal before "[Errno 104] Connection reset by peer" would appear. After that error appears, it seems like the Core restarts itself or something and it gets caught in a loop because the error keeps showing up.

Rockman18, 2023-09-28:
Same issue for me. Unable to start.

wds657, 2023-09-28:
i resolv

at boot select 

slot a rescue shell 

appear this
type control D

write console this

ha > core update --version 2023.9.3

agners, 2023-09-29:
This has most likely been caused by https://github.com/home-assistant/core/issues/101027.

The boot slot A and B are only in case a bad OS update got deployed. The rest of the Home Assistant system has it's own fallback behavior.

That being said, there is an issue in Supervisor which doesn't rollback the Core correctly in case of a bad update. This issue is tracked with https://github.com/home-assistant/supervisor/issues/4590.

Closing here as this is not really a OS issue.

closed 2023-09-29 by agners