home-assistant/operating-system #2778: Unable to stop Core to delete corrupt file

spiderzero3, 2023-09-28:
### Describe the issue you are experiencing

Home Assistant is no longer accessible to me after updating to 10.5 and 2023.10.0b

I poked around in the CLI to try and diagnose the issue. All I could find was a reference to a possibly corrupt database file (home-assistant_v2.db).

I want to run `core stop` because deleting the file while the core is running doesn't do anything; the file just reappears as soon as I've deleted it. When I try to run `core stop`, I get this error:
```Error: Another job is running for job group home_assistant_core```

I tried restoring a backup, but after some time processing, it throws the same error:
```Error: Another job is running for job group home_assistant_core```

I need a way to remove this database file but I don't know how.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
Error on call http://172.30.32.1:8123/api/core/state: [Errno 104] Connection reset by peer
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### Additional information

Core logs
```
INFO: Home Assistant Core process received signal 4
... Custom integration warnings ...
WARNING (Recorder) [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //config/home-assistant_v2.db was shutdown cleanly
WARNING (Recorder) [homeassistant.components.recorder.util] Ended unfinished session (id=176 from 2023-09-28 05:07:28.966858)
INFO: Home Assistant Core finish process exit code 256
```
The above snippet repeats multiple times.

niklas389, 2023-09-28:
Exactly the same happend to my instance.
VM on Proxmox using the official QEMU image.

spiderzero3, 2023-09-28:
Was able to roll back to 2023.9.3 by running `su restart` and then spamming `core update --version=2023.9.3` while the supervisor restarted. One of the commands slipped through between the supervisor starting and the supervisor starting the core and performed the update.

Not sure what caused the issue or if there's another solution, so feel free to close if necessary.

niklas389, 2023-09-28:
I got it to update to Version 2023.10.0b1, but that didn't change anything.

agners, 2023-10-03:
@spiderzero3 the issue you are seeing has been caused by a Core bug, specifically https://github.com/home-assistant/core/issues/101027. This has been addressed meanwhile, and should no longer cause issues.

Also, in such cases the Supervisor should rollback automatically. Unfortunately, it didn't work in that particular case. It has been addressed with https://github.com/home-assistant/core/pull/101105.

This error is related to the screen configuration, and should not be problematic:
```
ha > [   88.255420] i915 0000:00:0Z.0: [drm] *ERROR* Failed to read TMDS config: -6
```

I am closing this as this is not related to the Operating System.

closed 2023-10-03 by agners