home-assistant/operating-system #2780: HA Core 9.0 and above causing network issues and devices disappear

Max6993109, 2023-09-28:
### Describe the issue you are experiencing

After updating from 8.4 to 9.0 and above, my internet continuously restarted (around every 10 - 20 minutes).  In a day, I would lose connection 20-30 times.  After much troubleshooting with my ISP and swapping out modems and routers, it was determined that there was no issue.  I realized that it was my HA instance, as the issue started happening right when i updated.  I reverted back to 8.4 and the issue went away.  

Yesterday, I tried updating to 9.3 and the same thing started happening, with the addition of my devices screen not loading or being available.  I had to revert back to 8.4 again.  

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Updating from 8.4 to 9.3
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2023.8.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1294
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | May 30, 2024 at 8:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-11.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.09.2
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 983.6 GB
disk_used | 9.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Terminal & SSH (9.7.1), Mosquitto broker (6.3.1), Studio Code Server (5.10.2), Ring-MQTT with Video Streaming (5.6.3), Grafana (9.0.3), Home Assistant Google Drive Backup (0.111.1), Duck DNS (1.15.0), Z-Wave JS UI (2.0.1), Z-Wave JS (0.1.93), MariaDB (2.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 3
views | 11
mode | storage

</details>

<details><summary>MercedesME 2020</summary>

websocket_connection_state | connected
-- | --
api_endpoint_reachable | ok

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 27, 2023 at 9:02 PM
-- | --
current_recorder_run | September 28, 2023 at 1:40 AM
estimated_db_size | 3.95 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

The issue is not something that can be found in the logs, as I reverted back to 8.4.  However, as soon as I update to 9.0 and above, my internet just shuts down and reboots every 20-30 minutes.  Not sure what is causing it, but it only happens when I update, so I've isolated it to HA.  I've also seen this happen to others (something similar to the below): 

https://community.home-assistant.io/t/updated-entire-network-crashing/579664

agners, 2023-10-03:
I am not aware of anything specifically which could cause such behavior. I can only guess that it must be some weird interaction of Home Assistant and maybe your router and/or ISP, possibly some protection is triggered which blocks internet access?

What I'd recommend to try is if a new/empty installation of Home Assistant causes the same problems (e.g. the latest release 10.5). If this is not the case, I'd try the same version with your installation, but with add-ons and custom components disabled to see if one of these cause the problem.

If the problem is reproducible with a new/vanilla Home Assistant OS 10.5 installation, then you'd have to investigate more what is exactly happening in your network (e.g. tests with pings etc, capture network traffic and/or check with your ISP).

Max6993109, 2023-10-03:
Thanks for the response.  The weird part is that it works perfectly with version 8.4 and below.  I have 0 issues with this version.  Only 9.0 and above are causing my network to become unstable.  -Luis PichardoSent from my iPhone

agners, 2023-10-04:
How about OS 8.5?

According to [the changelog of OS 9.0](https://github.com/home-assistant/operating-system/releases/tag/9.0), there hasn't been a major Buildroot bump (so no new major release of NetworkManager or systemd with OS 9.x). So this is surprising.

Again, it would be really interesting to see how a vanilla installation behaves in your network.

Max6993109, 2023-10-04:
Looks like everything runs smooth on a vanilla version of HA.  Seems like
one of my integration is just not playing nice with 9.0 and above.  I'm
going to upgrade today to 10.0 and see if it fixes the issue.  If not, I'll
troubleshoot by turning them off and turning them on one by one.  Thanks!

Best,

--Luis Pichardo

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Max6993109, 2023-10-04:
Quick update.  Upgraded to 10.0 and I've had 0 issues with network (not
sure if I had updated some integrations prior and that helped or simply
something odd between 9.0 and my system, but so far so good).

Best,

--Luis Pichardo

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Max6993109, 2023-10-05:
Spoke too soon.  It started happening again.  It takes about 2-3 hours and
then it starts increasing in how often it happens.  I will start disabling
integrations.

--Luis Pichardo

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2024-01-11 by github-actions[bot]