home-assistant/operating-system #2783: Support for AW-CM389MA

WerVa, 2023-09-29:
### Describe the issue you are experiencing

Hi is possible to add suport for 

AW-CM389MA

i have wyse 3040 and my SDIO wifi/bt card dont working

dmesg | grep mmc:

mmc0: new ultra high speed SDR104 SDIO card at address 0001

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Checking status of wifi
2. Check card on ubuntu 
3.

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System information

Version | core-2023.9.3
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.11.5
Operating System Family | Linux
Operating System Version | 6.1.45
CPU Architecture | x86_64
Timezone | Europe/Warsaw
Configuration Directory | /config

<div class="card-header" style="justify-content: space-between; display: flex; align-items: center; color: rgb(225, 225, 225); font-family: Roboto, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: 0.4375px; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(28, 28, 28); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><br class="Apple-interchange-newline">

WerVa, 2023-10-01:
I think the mwifiex_pcie driver is missing

WerVa, 2023-10-03:
Maby this can help:
https://www.kernelconfig.io/config_mwifiex_sdio
https://forum.libreelec.tv/thread/21537-dell-wyse-3040-wifi-not-working/?pageNo=1

WerVa, 2023-10-09:
@agners  any idea?

agners, 2023-10-09:
Yeah probably `CONFIG_MWIFIEX_PCIE` is missing. #2761 will address this, but we need to flesh out the details for this PR.