home-assistant/operating-system #2784: Intel(R) Wi-Fi 6 AX200 firmware causing system crash and halt

bkallyat, 2023-10-01:
### Describe the issue you are experiencing

Hi,
The HAOS crashes with error message saying "microcode SW error detected, Restarting 0x0." However, the system does not reboot and is stuck.  The HA is currently running on AMD Mini PC. When the system is unreachable, I would connect this miniPC to a monitor to check the state. I would see a lot of iwlwifi errors post that microcode SW error. Have uploaded a pic of the error.
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.  Start home assistant 
2. Wait 4+ hours
3. connect the system to a monitor to view state
4.  system is stuck at a fault 9doesn't reboot)
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-10-01 23:23:02 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-10-01 23:27:22 INFO (MainThread) [supervisor.api.middleware.security] /os/config/sync access from a0d7b954_ssh
23-10-01 23:27:22 INFO (MainThread) [supervisor.os.manager] Synchronizing configuration from USB with Home Assistant Operating System.
23-10-01 23:27:22 INFO (MainThread) [supervisor.host.services] Restarting local service hassos-config.service
23-10-01 23:29:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/bus/usb/002/002 - None
23-10-01 23:29:35 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sg1 - None
23-10-01 23:29:35 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/bsg/1:0:0:0 - None
23-10-01 23:29:35 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sdb - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0
23-10-01 23:29:35 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sdb1 - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0-part1
23-10-01 23:29:43 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/bsg/1:0:0:0 - None
23-10-01 23:29:43 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/sg1 - None
23-10-01 23:29:43 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/sdb1 - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0-part1
23-10-01 23:29:43 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/sdb - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0
23-10-01 23:29:43 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/bus/usb/002/002 - None
23-10-01 23:30:57 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/bus/usb/002/003 - None
23-10-01 23:30:58 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sg1 - None
23-10-01 23:30:58 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/bsg/1:0:0:0 - None
23-10-01 23:30:58 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sdb - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0
23-10-01 23:30:58 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sdb1 - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0-part1
23-10-01 23:32:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/bsg/1:0:0:0 - None
23-10-01 23:32:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/sg1 - None
23-10-01 23:32:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/sdb1 - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0-part1
23-10-01 23:32:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/sdb - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0
23-10-01 23:32:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/bus/usb/002/003 - None
23-10-01 23:33:19 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/bus/usb/002/004 - None
23-10-01 23:33:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sg1 - None
23-10-01 23:33:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/bsg/1:0:0:0 - None
23-10-01 23:33:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sdb - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0
23-10-01 23:33:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting add hardware /dev/sdb1 - /dev/disk/by-id/usb-USB_SanDisk_3.2Gen1_01015cf3c6f8e83aae7c03803322809b1ef7374b7b317e6c835537337336a351eba60000000000000000000068b755d4ff9e0700ab5581076d2cd29c-0:0-part1
23-10-01 23:34:26 INFO (MainThread) [supervisor.api.middleware.security] /os/config/sync access from a0d7b954_ssh
23-10-01 23:34:26 INFO (MainThread) [supervisor.os.manager] Synchronizing configuration from USB with Home Assistant Operating System.
23-10-01 23:34:26 INFO (MainThread) [supervisor.host.services] Restarting local service hassos-config.service
23-10-01 23:40:20 ERROR (MainThread) [supervisor.api.ingress] Stream error with http://172.30.33.0:1337/stable-2eba7af117ea58d45a6c6449ee4fe63c8d4d53aa/static/out/vs/workbench/services/keybinding/browser/keyboardLayouts/layout.contribution.win.js: Cannot write to closing transport
23-10-01 23:40:21 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-10-01 23:40:21 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_vscode
23-10-01 23:40:21 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-10-01 23:41:18 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 257:None is not WSMsgType.TEXT
23-10-01 23:41:18 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-10-01 23:41:55 INFO (MainThread) [supervisor.api.middleware.security] /os/config/sync access from a0d7b954_ssh
23-10-01 23:41:55 INFO (MainThread) [supervisor.os.manager] Synchronizing configuration from USB with Home Assistant Operating System.
23-10-01 23:41:55 INFO (MainThread) [supervisor.host.services] Restarting local service hassos-config.service
[... 211 lines cut ...]
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 20, 2023 at 5:33 PM
-- | --
current_recorder_run | October 1, 2023 at 10:52 PM
estimated_db_size | 13.94 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

[image]

agners, 2023-10-03:
This looks like some upstream iwlwifi issue. By searching the string `Microcode SW error detected. Restarting 0x0.` this problem seems to happen also on other operating systems at times. What some people suggest is to set iwlwifi options, specifically `11n_disable=1` and `bt_coex_active=0`. You can set such options by editing `cmdline.txt` on the boot partition (`/mnt/boot`). Currently you have to edit this file directly on the OS shell (which you can access by using `login` on the terminal).

bkallyat, 2023-10-04:
Thanks. Will set those options and update if this workaround works.

bkallyat, 2023-10-04:
Didn't work. Crashed in few hours. Pic attached.
[image]

bkallyat, 2023-10-17:
Any other ideas anyone have to resolve this? This crash is basically making my Home Assistant setup useless. All my automation is dead 😖

bkallyat, 2023-10-25:
@agners Can the HAOS be updated to include the new Intel Wifi firmwares? It may possibly resolve this bug. https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git

closed 2024-02-01 by github-actions[bot]