home-assistant/operating-system #2785: HassOS 10.5 - serial driver "ch341-uart" no longer loading

IoTPlay, 2023-10-02:
### Describe the issue you are experiencing

After upgrading to HassOS 10.5 - serial driver "ch341-uart" no longer loading - my serial RS485 WaveShare device which connects many Modbus devices - and have been working for 2 years no longer works. 

On inspection of the USB drives - I found that the driver "ch341-uart" are no longer loaded in the kernel.

Must I load it by hand? any other command to effect it?

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant 10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

In terminal - 
1. ran "lsusb" - the UAT no longer showing
2. ran "lsmod | grep "usbserial" - "ftdi_sio,ch341" no longer showing
3.unplugged the serial device, plugged again, and rebooted
4. ran "ls /lib/modules/$(uname -r)/kernel/drivers/usb/serial | grep "ch34" - uart not showing.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No.
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

### System information

## System Information

version | core-2023.9.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.21-v7
arch | armv7l
timezone | Africa/Johannesburg
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4981
Installed Version | 1.33.0
Stage | running
Available Repositories | 1299
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.09.2
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 13.9 GB
disk_used | 6.5 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.11.0), Samba share (10.0.2), Terminal & SSH (9.7.1), ESPHome (2023.9.2), Paradox Alarm Interface (3.2.1), Mosquitto broker (6.3.1), File editor (5.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 0
views | 14
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 September 2023 at 11:13
-- | --
current_recorder_run | 2 October 2023 at 22:24
estimated_db_size | 141.10 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

agners, 2023-10-03:
The kernel for Raspberry Pi 3 in HAOS 10.5 did not change, so I don't think that this is a HAOS issue.

Can you try downgrading the operating system using `ha os update --version 10.4` to see if that indeed fixes the problem?

If that doesn't help, I'd guess a hardware issue here. Either the USB port or the USB device stopped working :cry:

IoTPlay, 2023-10-03:
Okay, thank you, let me see if I can track down a command to try and load the UART into the kernel again.

agners, 2023-10-04:
You should not need to manually load the driver. The driver will get loaded automatically, if the device is there...

Check the Hardware list in Settings > System > Hardware -> All Hardware (or lsusb on the console) to see if the device is detected.