home-assistant/operating-system #2788: USB ftdi_sio problem with RFXCOM

Jezza34000, 2023-10-03:
### Describe the issue you are experiencing

Hi !
I can't get my RFXCOM to work, I get the following error message:
[image]

Look like this issue : https://github.com/home-assistant/operating-system/issues/1800

I use 4 USB devices : 
1. RFXCOM RFXrtx433
2. Bluetooth SENA UD-100
3. Itead Sonoff Zigbee 3.0
4. Ziblue RFPLAYER

Only the RFXCOM had problem. Tested on my Ubuntu laptop it's working well.

I'm using a VM on a QNAP Nas TS-253D

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Plug RFXCOM into HA's usb port
2. Got error message on HA startup

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing usefull
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing usefull
```

### System information

## System Information

version | core-2023.9.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1306
Downloaded Repositories | 20

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 3 janvier 2024 à 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-2.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.09.2
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 117.4 GB
disk_used | 7.4 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Glances (0.19.3), Home Assistant Google Drive Backup (0.111.1), Mosquitto broker (6.3.1), Node-RED (14.5.0), Terminal & SSH (9.7.1), Zigbee2MQTT (1.33.0-1), Crowdsec (1.5.2-ha1), Crowdsec Firewall Bouncer (v0.0.26)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 5
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 24 septembre 2023 à 14:00
-- | --
current_recorder_run | 3 octobre 2023 à 14:47
estimated_db_size | 359.24 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.11
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 1
total_devices | 1

</details>

### Additional information

Thank you for help

maddinhome, 2023-10-05:
same bug here on synology vm

maddinhome, 2023-10-10:
@agners yes, I use an RFX433XL USB. Only for a few switches... but I use it.

It worked until version 9.2 of Homeassistant. The HAOS version at this time was 10.5. Now I have installed HAOS 11.0rc1.

VM from Synology...

I noticed that when I restart the RFX (plugging it in and out) only the red LED flashes. Otherwise a green one was always added. Maybe there is also a hardware defect. But the RFX is recognized by Homeassistant when it is cold started. Only in the device overview does it say “Error setting up”.

The logs aren't very informative either... hmmm

Jezza34000, 2023-10-10:
> Only in the device overview does it say “Error setting up”.
> The logs aren't very informative either... hmmm

Same for me

maddinhome, 2023-10-12:
Here is an excerpt from my log file:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.11/site-packages/RFXtrx/__init__.py", line 986, in _connect
    self.transport.reset()
  File "/usr/local/lib/python3.11/site-packages/RFXtrx/__init__.py", line 832, in reset
    self.send(b'\x0D\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
  File "/usr/local/lib/python3.11/site-packages/RFXtrx/__init__.py", line 828, in send
    self.serial.write(pkt)
    ^^^^^^^^^^^^^^^^^

later:
2023-10-12 16:22:18.908 ERROR (MainThread) [homeassistant.components.rfxtrx] Connection timeout: failed to receive response from RFXtrx device

hmmm 

Maybe that will help narrow down the error.

maddinhome, 2023-10-18:
Is there something new? Does anyone have a solution?

Jezza34000, 2023-10-18:
Nope

sm0keKings, 2023-10-27:
There was a HA OS update 11.1 released.  It didn't appear to fix it.  Did anyone find another solution?

maddinhome, 2023-10-27:
Apparently there are too few of us and the problem is not being seen.

It's unfortunate... I'm already looking for an alternative

sm0keKings, 2023-10-27:
I think there is a lot of us that are waiting for them to fix the OS issue.  I will start to look for a fix to the OS.

Jezza34000, 2023-10-28:
My workaround was to stop using RFXCOM, i use RFLINK with Arduino instead :-)

WhiteDogBe, 2023-10-29:
Waiting on the same fix... I also notified the vendor about this thread, they might be able to debug this more easily.

WhiteDogBe, 2023-10-30:
I have 2 updates:
- I'm using HAOS via ProxMox and tried the upgrade from 9.5 to 11.1... my RFXTRX433XL is still working. So it's not 'all' VM that are affected. Might depend on how exactly the device is passed through in the VM system you are using.

- The RFXCOM developer responded to me with a few remarks, but as I don't have the issue I can't help solve it. Feel free to reach out to him if you are on a path for troubleshooting.

I go this suggestion though:

> We use a standard FTDI FT232 chip in the RFXtrx. We only changed the Manufacturer and Product description in the USB string descriptors.
>
> You can change this to the FTDI setting if you like using FT_prog https://ftdichip.com/utilities/
> Manufacturer: FTDI
> Product description: FT232R USB UART
>  
> If you change this back to the FTDI settings you need the latest RFXmngr or else the RFXtrx is not detected.

The above is something you can try very easily imho if you are affected by this.
Other suggestion is to use ser2net.

sm0keKings, 2023-10-30:
> I have 2 updates:
>
> * I'm using HAOS via ProxMox and tried the upgrade from 9.5 to 11.1... my RFXTRX433XL is still working. So it's not 'all' VM that are affected. Might depend on how exactly the [...]

I am not using a VM.  I have a native HAOS on a Intel x86 the issue appears to be the OS not specifically the VM config.

TheOriginalMrWolf, 2023-11-07:
Hi @Jezza34000 

I had a similar problem.  Putting an **unpowered** USB2 hub between the RFX & Pi fixed the problem (had to try a few to find the 'golden' one). 

Worked flawlessly for more than a year since.  Don't ask me why....

sm0keKings, 2023-11-07:
Removing the addon and reisntalling chosing the other USB connection method worked for me too!

agners, 2023-11-07:
In the end, this is probably a device which is on the edge (or even violating) some USB specs. Depending on exact hardware configuration (USB host IP/type), and USB bus configuration (with/without hub(s)) things might just work, or just not work.

Home Assistant OS uses a rather vanilla upstream Linux kernel version 6.1 for the OS 10 and 11 series. So on the same hardware, with a operating system using the same kernel version, I'd expect the same issues in the end.

Possible that different kernel version lead to slightly different timing, but I don't think there is a general fix we can come up with from HAOS side :cry:

Jezza34000, 2023-11-07:
OK thank you all for possible "workaround" and @agners for explanation

closed 2023-11-07 by Jezza34000