home-assistant/operating-system #2795: updating to HAOS 11 Rc1 - network sensors become unknown eth0 to end0?

Anto79-ops, 2023-10-05:
### Describe the issue you are experiencing

Hi!

Just updated to HAOS 11RC1 (RPi4-4GB) and I noticed my network sensors from system monitor setup become unknown.

```
#System Health Sensors
- platform: systemmonitor
  resources:
    - type: disk_use_percent
    - type: disk_free
    - type: disk_use
    - type: network_in
      arg: eth0
    - type: network_out
      arg: eth0
    - type: throughput_network_in
      arg: eth0
    - type: throughput_network_out
      arg: eth0
    - type: swap_use_percent
    - type: memory_free
    - type: processor_use
    - type: processor_temperature
    - type: load_1m
    - type: load_5m
    - type: load_15m
    - type: last_boot
```

I see that "eth0" is not shown in my network properties anymore, but rather its now "end0"

Do I simply need to change it to end0 now?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

HAOS 11.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. update to HAOS 11.rc1
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no logs
```

### Anything in the Host logs that might be useful for us?

```txt
no logs
```

agners, 2023-10-05:
Yes, this is a known change with https://github.com/home-assistant/operating-system/pull/2763.

Home Assistant OS relies on systemd, which made a push for stable network interface names. Historically eth0, eth1 etc. have just been assigned based on driver load order. However, newer kernel versions, or just randomness during bootup can change that order.

Previous systemd versions already created stable network interface names such as `enp1s0` based on bus location of a particular network card for PCIe devices.

See also [Predictable Network Interface Names](https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/) article on freedesktop.

I see this can have user impact. I've just added this to the release notes, and will make sure to also mention it on the final release's release notes.

closed 2023-10-05 by agners

Anto79-ops, 2023-10-05:
thanks! Good that it now mentioned as a breaking change....a small one nonetheless. Cheers