home-assistant/operating-system #2796: An error occured while installing Home Assistant, check the logs below for more information.

cfpl02, 2023-10-06:
### Describe the issue you are experiencing

supervisor.docker.interface] Attaching to ghcr.io/home-assistant/generic-x86-64-homeassistant with version landingpage
[supervisor.homeassistant.core] Starting HomeAssistant landingpage
[supervisor.homeassistant.core] Watchdog found Home Assistant failed, restarting...
[supervisor.homeassistant.core] Watchdog restart of Home Assistant failed!
[supervisor.docker.manager] Starting homeassistant
[supervisor.os.manager] Detect Home Assistant Operating System 11.0.rc1 / BootSlot A
[supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
[supervisor.store.git] Loading add-on /data/addons/core repository
[supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
[supervisor.store] Loading add-ons from store: 72 all - 72 new - 0 remove
[supervisor.addons] Found 0 installed add-ons
[supervisor.backups.manager] Found 0 backup files
[supervisor.discovery] Loaded 0 messages
[supervisor.ingress] Loaded 0 ingress sessions
[supervisor.resolution.check] Starting system checks with state setup
[supervisor.resolution.check] System checks complete
[supervisor.resolution.evaluate] Starting system evaluation with state setup
[supervisor.resolution.evaluate] System evaluation complete
[supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup
[supervisor.resolution.evaluate] Starting system evaluation with state setup
[supervisor.resolution.evaluate] System evaluation complete
[__main__] Running Supervisor
[supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
[supervisor.addons] Phase 'initialize' starting 0 add-ons
[supervisor.addons] Phase 'system' starting 0 add-ons
[supervisor.addons] Phase 'services' starting 0 add-ons
[supervisor.core] Skiping start of Home Assistant
[supervisor.addons] Phase 'application' starting 0 add-ons
[supervisor.misc.tasks] All core tasks are scheduled
[supervisor.core] Supervisor is up and running
[supervisor.homeassistant.core] Home Assistant setup
[supervisor.docker.interface] Updating image ghcr.io/home-assistant/generic-x86-64-homeassistant:landingpage to ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.10.0
[supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2023.10.0.
[supervisor.host.info] Updating local host information
[supervisor.resolution.check] Starting system checks with state running
[supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
[supervisor.resolution.checks.base] Run check for security/core
[supervisor.resolution.checks.base] Run check for free_space/system
[supervisor.resolution.checks.base] Run check for no_current_backup/system
[supervisor.resolution.module] Create new suggestion create_full_backup - system / None
[supervisor.resolution.module] Create new issue no_current_backup - system / None
[supervisor.resolution.checks.base] Run check for multiple_data_disks/system
[supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
[supervisor.host.services] Updating service information
[supervisor.host.network] Updating local network information
[supervisor.resolution.checks.base] Run check for pwned/addon
[supervisor.resolution.checks.base] Run check for docker_config/system
[supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
[supervisor.host.manager] Host information reload completed
[supervisor.resolution.checks.base] Run check for trust/supervisor
[supervisor.resolution.check] System checks complete
[supervisor.resolution.evaluate] Starting system evaluation with state running
[supervisor.resolution.evaluate] System evaluation complete
[supervisor.resolution.fixup] Starting system autofix at state running
[supervisor.resolution.fixup] System autofix complete

DOWNLOAD RAW LOGS
[Read our vision](https://www.home-assistant.io/blog/2016/01/19/perfect-home-automation/)
Join our community

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6

[... 172 lines cut ...]
[supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
[supervisor.resolution.checks.base] Run check for security/core
[supervisor.resolution.checks.base] Run check for free_space/system
[supervisor.resolution.checks.base] Run check for no_current_backup/system
[supervisor.resolution.module] Create new suggestion create_full_backup - system / None
[supervisor.resolution.module] Create new issue no_current_backup - system / None
[supervisor.resolution.checks.base] Run check for multiple_data_disks/system
[supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
[supervisor.host.services] Updating service information
[supervisor.host.network] Updating local network information
[supervisor.resolution.checks.base] Run check for pwned/addon
[supervisor.resolution.checks.base] Run check for docker_config/system
[supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
[supervisor.host.manager] Host information reload completed
[supervisor.resolution.checks.base] Run check for trust/supervisor
[supervisor.resolution.check] System checks complete
[supervisor.resolution.evaluate] Starting system evaluation with state running
[supervisor.resolution.evaluate] System evaluation complete
[supervisor.resolution.fixup] Starting system autofix at state running
[supervisor.resolution.fixup] System autofix complete

DOWNLOAD RAW LOGS
Read our vision
Join our community
```

agners, 2023-10-06:
Hm, I don't really see an error. What happens after that? Can you download the raw logs and upload them here?

cfpl02, 2023-10-06:
[logs (1).txt](https://github.com/home-assistant/operating-system/files/12829590/logs.1.txt)

cfpl02, 2023-10-06:
[supervisor.resolution.fixup] Starting system autofix at state running
[supervisor.resolution.fixup] System autofix complete
And then get stuck here for an hour

cfpl02, 2023-10-07:
> 嗯，我真的没有看到错误。之后会发生什么？您可以下载原始日志并上传到这里吗？

Could you please help me take a look

agners, 2023-10-08:
The error is this:

```
23-10-06 14:23:23 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
23-10-06 14:23:23 WARNING (MainThread) [supervisor.homeassistant.core] Watchdog found Home Assistant failed, restarting...
23-10-06 14:23:23 ERROR (MainThread) [supervisor.homeassistant.core] Watchdog restart of Home Assistant failed!
```

However, this error should not be fatal. The system still should download the Core container just fine.

It seems the system starts to download the Core container:

```
23-10-06 14:23:23 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2023.10.0.
```

Do you have access to the container `ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.10.0` from GitHub? Maybe it just takes a while?

cfpl02, 2023-10-09:
> 错误是这样的：
> 
> ```
> 23-10-06 14:23:23 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
> 23-10-06 14:23:23 WARNING (MainThread) [supervisor.homeassistant.core] Watchdog found Home Assistant failed, restarting...
> 23-10-06 14:23:23 ERROR (MainThread) [supervisor.homeassistant.core] Watchdog restart of Home Assistant failed!
> ```
> 
> 然而，这个错误不应该是致命的。系统仍然应该可以正常下载 Core 容器。
> 
> 系统似乎开始下载 Core 容器：
> 
> ```
> 23-10-06 14:23:23 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2023.10.0.
> ```
> 
> `ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.10.0`您可以从 GitHub访问该容器吗？也许只需要一段时间？

> 错误是这样的：
> 
> ```
> 23-10-06 14:23:23 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
> 23-10-06 14:23:23 WARNING (MainThread) [supervisor.homeassistant.core] Watchdog found Home Assistant failed, restarting...
> 23-10-06 14:23:23 ERROR (MainThread) [supervisor.homeassistant.core] Watchdog restart of Home Assistant failed!
> ```
> 
> 然而，这个错误不应该是致命的。系统应该仍然可以正常下载核心容器。
> 
> 系统似乎开始下载核心容器：
> 
> ```
> 23-10-06 14:23:23 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2023.10.0.
> ```
> 
> `ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.10.0`您可以从 GitHub访问该容器吗？也许只需要一段时间？

Continue waiting? After waiting for an hour without any response, I restarted and reported an error after restarting

constancyGmbH, 2023-10-30:
I have the same problem with my Raspberry Pi 3B. What Hardware are you using?

Ruko2010, 2023-10-30:
I have the same problem with my Raspberry Pi 3B. What Hardware are you using @cfpl02 ?

Subito, 2023-12-07:
Having the same issue since a few updates ago. First time it happend was 2023.11.1 -> 2023.11.3, and now the latest update to 2023.12 also triggered this. Most Integrations fail to load, startup takes forever and never finishes. "Some" things in the Interface don't work (eg the restart). Interestingly, what doesn't work seems to change with reboots. Sometimes the ZHA Integration comes up for instance, and sometimes it doesn't.

The last time it fixed "itself" after one night. Worked fine until I started the next update. I use a PI 4 with HA OS.

St8Ev, 2023-12-10:
[logs (1).txt](https://github.com/home-assistant/operating-system/files/13626892/logs.1.txt)
[image]

St8Ev, 2023-12-10:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[06:38:56] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[06:38:56] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
23-12-10 06:38:57 INFO (MainThread) [__main__] Initializing Supervisor setup
23-12-10 06:38:57 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
23-12-10 06:38:57 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: generic-x86-64
23-12-10 06:38:57 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2023.11.6
23-12-10 06:38:57 INFO (MainThread) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
23-12-10 06:38:57 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
23-12-10 06:38:57 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-12-10 06:38:57 INFO (MainThread) [__main__] Setting up Supervisor
23-12-10 06:38:58 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
23-12-10 06:38:58 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
23-12-10 06:38:58 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
23-12-10 06:38:58 INFO (MainThread) [supervisor.host.services] Updating service information
23-12-10 06:38:58 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
23-12-10 06:38:58 WARNING (SyncWorker_1) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
23-12-10 06:38:58 INFO (MainThread) [supervisor.host.network] Updating local network information
23-12-10 06:38:58 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
23-12-10 14:38:58 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
23-12-10 14:38:58 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-cli versions: [<AwesomeVersion CalVer '2023.10.0'>]
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2023.10.0
23-12-10 14:38:59 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2023.10.0 - 172.30.32.5
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-dns versions: [<AwesomeVersion CalVer '2023.06.2'>]
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2
23-12-10 14:38:59 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2023.06.2 - 172.30.32.3
23-12-10 14:38:59 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-audio versions: [<AwesomeVersion CalVer '2023.10.0'>]
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.10.0
23-12-10 14:38:59 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.10.0 - 172.30.32.4
23-12-10 14:38:59 INFO (MainThread) [supervisor.docker.interface] Found ghcr.io/home-assistant/amd64-hassio-observer versions: [<AwesomeVersion CalVer '2023.06.0'>]
[... cut ...]
23-12-10 15:24:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
23-12-10 15:24:13 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-12-10 15:24:13 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
23-12-10 15:24:13 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-12-10 15:24:13 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
23-12-10 15:24:13 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete

St8Ev, 2023-12-10:
Система запущена на Virtual box

Ruko2010, 2023-12-12:
My Problem was solved by using another SD Card on my Pi. Maybe the first one was broken.

agners, 2023-12-12:
@All: "An error occured while installing Home Assistant, check the logs below for more information." is a generic issue when there was a issue during initial installation of Home Assistant. The cause depends on the exact logs. If you suffer from this problem, reboting the system did not help and you can't find the issue even after reviewing the logs, please open a new issue along with your logs instead of posing to this issue, since every case is slightly different.

@cfpl02 have you been able to resolve your problem?

@St8Ev which version of the Home Assistant OS have you been using? 

Your error:
```
23-12-10 14:53:29 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.12.1: 404 Client Error for http+docker://localhost/v1.43/images/ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.12.1/json: Not Found ("No such image: ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.12.1")
````

Typically appears when there was a corruption in the Docker storage (see also https://github.com/home-assistant/supervisor/issues/4738#issuecomment-1834627523). Since this is a new installation probably the best way forward is to start over.

St8Ev, 2023-12-12:
> @ALL: "При установке Home Assistant произошла ошибка, проверьте журналы ниже для получения дополнительной информации". это общая проблема, возникающая при первоначальной установке Home Assistant. Причина зависит от точных журналов. Если вы столкнулись с этой проблемой, перезагрузка системы не помогла, и вы не можете найти проблему даже после просмотра журналов, пожалуйста, откройте новую проблему вместе со своими журналами, вместо того чтобы обращаться к этой проблеме, поскольку каждый случай немного отличается.
> 
> @cfpl02 удалось ли вам решить вашу проблему?
> 
> @St8Ev какую версию ОС Home Assistant вы использовали?
> 
> Ваша ошибка:
> 
> ```
> 23-12-10 14:53:29 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.12.1: 404 Client Error for http+docker://localhost/v1.43/images/ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.12.1/json: Not Found ("No such image: ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.12.1")
> ```
> 
> Обычно появляется при повреждении хранилища Docker (см. Также [home-assistant / supervisor#4738 (комментарий)](https://github.com/home-assistant/supervisor/issues/4738#issuecomment-1834627523)). Поскольку это новая установка, вероятно, лучший способ начать все сначала.

Я использовал установку образа из Ubuntu, посредством записи образа из утилиты "Диски" из инструкции - 
[image]
https://www.home-assistant.io/installation/generic-x86-64 
Проблема самоустранилась после длительного простоя ПК.

Subito, 2024-03-12:
The same symptoms and similar logs. From what I understand an upgrade with HAOS downloads a new core and swaps it in place. I have the same problem that homeassistant does not start with the new core, just that I have a fallback which kicks in and still works. 

You might be right, though. I have no problems with all integrations disabled which certainly is the case on initial installation. So my problem might be indeed a different one, although it shows itself in some of the same ways. I'll try to find some useful info and open a new issue if I find something to report.

closed 2024-06-19 by github-actions[bot]