home-assistant/operating-system #2809: HAOS 11.0rc2 contains vulnerable curl (8.3.0)

daernsinstantfortress, 2023-10-13:
### Describe the issue you are experiencing

HAOS 11.0rc2 contains the 8.3.0 version of curl which is subject to a [high severity](https://www.intruder.io/blog/curl-high-rated-cve-2023-38545) exploit which has [been patched in 8.4.0](https://curl.se/docs/CVE-2023-38545.html).

This, and libcurl, should be updated 8.4.0 as soon as possible.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.0rc2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

```
# curl --version
curl 8.3.0 (x86_64-buildroot-linux-gnu) libcurl/8.3.0 OpenSSL/1.1.1v zlib/1.2.13
Release-Date: 2023-09-13
Protocols: dict file ftp ftps gopher gophers http https imap imaps mqtt pop3 pop3s rtsp smtp smtps telnet tftp
Features: alt-svc AsynchDNS HSTS HTTPS-proxy IPv6 Largefile libz SSL threadsafe TLS-SRP UnixSockets
```

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2023.10.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4914
Installed Version | 1.33.0
Stage | running
Available Repositories | 1308
Downloaded Repositories | 12

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 16 November 2023 at 00:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-10.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0.rc2
-- | --
update_channel | beta
supervisor_version | supervisor-2023.10.0
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 916.2 GB
disk_used | 29.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.2), MariaDB (2.6.1), Advanced SSH & Web Terminal (15.0.8), ESPHome (2023.9.3), File editor (5.6.0), Frigate (Full Access) (0.12.1), Home Assistant Google Drive Backup (0.111.1), Let's Encrypt (4.12.9), Mosquitto broker (6.3.1), Network UPS Tools (0.12.1), Zigbee2MQTT (1.33.1-1), texecom2mqtt (1.2.3), HDD Tools (1.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 4
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 7 October 2023 at 21:49
-- | --
current_recorder_run | 13 October 2023 at 10:00
estimated_db_size | 623.88 MiB
database_engine | mysql
database_version | 10.6.12

</details>

daernsinstantfortress, 2023-10-13:
Ah, looks like there's already been a commit covering this work, which presumably will hit in the next release:
https://github.com/home-assistant/operating-system/commit/ff0a4b44e07de5a9f6129062bb37aee93a478717

Will close, accordingly.

closed 2023-10-13 by daernsinstantfortress

agners, 2023-10-13:
Yes, the fix will be in OS 11.0 which we are about to release today.