home-assistant/operating-system #2814: HAOS 11.0 - Error returned from Supervisor: System is not ready with state: setup

SGXander, 2023-10-13:
### Describe the issue you are experiencing

Updating to 11.0 produces this error after VM restart. HA unavailable and no logs show when running supervisor logs from the console
Home Assistant 2023.10.2 on Hyper-V

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Run HA OS 11.0 update
2. Await restart

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs displayed when running supervisor logs
```

### Anything in the Host logs that might be useful for us?

```txt
VM operating normally from host
```

IvoSvejcar, 2023-10-13:
Exacxtly same story.

agners, 2023-10-13:
So Supervisor stays in setup all the time?

Can you try the following commands in the VM console:

```
login
docker logs hassio_supervisor
```

SGXander, 2023-10-13:
I'd love to but I shut it down and tried a fresh boot but get the cli failed to start: jumping to emergency. even my pre-upgrade snapshot is doing it so in the process of rebuilding :(

MalaZuza, 2023-10-13:
why do they make an update that doesn't work and ruins everything!!! Milion problems for the last month. Oh

SGXander, 2023-10-14:
ok I'm super screwed now. even after restoring the VM from a backup I get this. waited 5 and typed banner which shows the expected info but HA not up on the address shown. Please help!

joshuamallow, 2023-10-14:
Mine dead too 😞

agners, 2023-10-14:
> why do they make an update that doesn't work and ruins everything!!! Milion problems for the last month. Oh

Just to be clear, the update has been successfully tested by 1000+ beta testers. We are not aware of breaking issues.

Without a detailed description of what goes wrong on your end we can't help you :cry:

agners, 2023-10-14:
> ok I'm super screwed now. even after restoring the VM from a backup I get this. waited 5 and typed banner which shows the expected info but HA not up on the address shown. Please help!

Can you type in the command I suggested in https://github.com/home-assistant/operating-system/issues/2814#issuecomment-1762314397 and post the output please?

SGXander, 2023-10-14:
it was a bit late last night but after a another restore from yesterday and shutting down all other services on the host machine I did get it to boot. "login" works from the emergency prompt. fires up HACLI and from there running banner showed the details. I checked the ui page but just gave me the retry splash. Was ready to head to bed when it finally sprang to life about 20 mins later. Spent another 15 repairing the integrations that were doa and then stopped for the night. Is it possible those two errors are appearing due to lack of VM resources? I'm looking into what the cause could be today I suspect windows updates...

SGXander, 2023-10-16:
I've updated HA-Core to 2023.10.3 now after getting a good backup last night and while it's ok and I'll look to doing HAOS 11 later, it seems to be extremely sensitive to any hypervisor latency. not sure if that's a new thing or not but just snapshotting now has a chance of upsetting HA Core and demanding a restart from app controller. I'm convinced Windows updates to the hypervisor and other VMs on it have messed with the performance of them which is impacting the HA VM.

SGXander, 2023-10-23:
Confirmed this does seem to be hypervisor resource issues after windows updates and not HAOS. Upgraded now and working fine. Apologies for the accusation.

closed 2023-10-23 by SGXander

TDuxis, 2024-05-19:
This problem occurred to me "right out of the blue".

My system is running 11.5, and was working. Suddenly I noticed it was not working. Several integrations had stopped and wouldn't restart. So I restarted the RPI4 that HA is running on, and ran into this problem. I found the solution of typing "banner" in the CLI, and then it was working. I am not sure though if typing "banner" really made any difference, or if it just was that some more time passed by while I searched for and tested the "banner" solution, as banner seems to do nothing more than representing the HA-banner and some info in the CLI.

In any case, this problem occurred suddenly for me with version 11.5

willysagefalk, 2024-05-28:
Had the same problem with "error returned from Supervisor....."

Typing banner in cli helped

thogens, 2024-06-13:
Since yesterday I'm facing the same issue after rebooting using HA 2024.6.2 and Supervisor 2024.06.0.
Interestingly after waiting some minutes, HA is running and reachable, but at the same time my virtual machine still shows the same message in the console.
- Virtual machine console shows the error message.
- HA terminal shows correct state...
- Why is that? Any idea?

SGXander, 2024-06-13:
> Since yesterday I'm facing the same issue after rebooting using HA 2024.6.2 and Supervisor 2024.06.0. Interestingly after waiting some minutes, HA is running and reachable, but at the same time my [...]

I've found this is most often caused by a full or dying SSD backing the VM. After replacing a dying one in an array my instance immediately stopped doing this.
That said if it's just being a bit slow then HA throws the error and tries again in a short while so you may just be seeing the console logged error but it's now running ok.

thogens, 2024-06-13:
In my case, HA is running on a Synology NAS with a RAID. Not sure if the HDDs are the issue here.

andrewigali, 2024-12-16:
Same problem, but I use a MiniPC for HA system, on which only it ran.
HAOS 2024.12.3

I tried the following but it didn't help.
banner
supervisor reapir
supervisor restart
ha core rebuild