home-assistant/operating-system #2816: HA operating system update 11.0 --> Error setting up entry Supervisor for hassio

claustro, 2023-10-13:
### Describe the issue you are experiencing

I am having the error "Failed to setup supervisor" after updating to 11.0 . Even with this error all seems working correctly .
 I tried to reboot the machine without luck

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

 6.1.56

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to 00185a40/amd64-addon-simplescheduler with version 2.11[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-check_config with version 3.11.0[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-samba with version 10.0.2[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.6.0[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to sabeechen/hassio-google-drive-backup-amd64 with version 0.111.1[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-ssh with version 9.7.1[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.11.0[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to zigbee2mqtt/zigbee2mqtt-amd64 with version 1.33.1-1[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mosquitto with version 6.3.1[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/tasmoadmin/amd64 with version 0.25.0[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/grafana/amd64 with version 9.0.3[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/amd64 with version 14.6.1[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/influxdb/amd64 with version 4.7.0[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.docker.interface] Attaching to smeagolworms4/mqtt-explorer with version browser-1.0.1[0m
[33m23-10-14 00:48:26 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon TasmoAdmin is unhealthy, restarting...[0m
[32m23-10-14 00:48:26 INFO (SyncWorker_4) [supervisor.docker.manager] Stopping addon_a0d7b954_sonweb application[0m
[32m23-10-14 00:48:26 INFO (MainThread) [supervisor.backups.manager] Found 64 backup files[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.discovery] Loaded 3 messages[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-10-14 00:48:27 INFO (MainThread) [__main__] Running Supervisor[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.addons] Phase 'initialize' starting 0 add-ons[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.core] Supervisor reboot detected[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.core] Supervisor is up and running[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.host.info] Updating local host information[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.host.manager] Host information reload completed[0m
[33m23-10-14 00:48:27 WARNING (MainThread) [supervisor.addons.options] Option 'dark_mode' does not exist in the schema for Node-RED (a0d7b954_nodered)[0m
[33m23-10-14 00:48:27 WARNING (MainThread) [supervisor.addons.options] Option 'require_ssl' does not exist in the schema for Node-RED (a0d7b954_nodered)[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize[0m
[32m23-10-14 00:48:27 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered[0m
[... 237 lines cut ...]

oldest_recorder_run | October 11, 2023 at 20:58
-- | --
current_recorder_run | October 14, 2023 at 00:40
estimated_db_size | 178.11 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.1 (122751c)
-- | --
cloud_online | 0 / 1
local_online | 1 / 1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

cmd85, 2023-10-14:
In same boat, Supervisor have failed for me, not sure how to fix it.

agners, 2023-10-14:
From the logs it seems that Tasmota add-on was holding up the boot process, and got declared as unhealthy.

```
�[32m23-10-14 00:48:31 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/tasmoadmin/amd64 with version 0.25.0�[0m
�[32m23-10-14 00:48:37 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup�[0m
�[32m23-10-14 00:48:37 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup�[0m
�[32m23-10-14 00:49:56 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'mqttuser'�[0m
�[32m23-10-14 00:49:56 INFO (MainThread) [supervisor.auth] Successful login for 'mqttuser'�[0m
�[33m23-10-14 00:50:01 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon TasmoAdmin is unhealthy, restarting...�[0m
```

Is this problem reproducible on every system reboot? If so, can you disable the Tasmota add-on to see if that resolves the issue?

agners, 2023-10-14:
> In same boat, Supervisor have failed for me, not sure how to fix it.

Is your system reachable as well? If so, can you share the Supervisor logs (Settings > System > Logs, choose Supervisor on the top right).

claustro, 2023-10-14:
> From the logs it seems that Tasmota add-on was holding up the boot process, and got declared as unhealthy.

> Is this problem reproducible on every system reboot? If so, can you disable the Tasmota add-on to see if that resolves the issue?

On reboot the Tasmota error is gone. 
Still having error on supervisor , the log show different errors every time _I reboot_

Now this one

23-10-14 11:11:36 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-27073' coro=<Repository.update() done, defined at /usr/src/supervisor/supervisor/store/repository.py:104> exception=StoreGitError()>

agners, 2023-10-14:
There should be an error further above in the logs with details:

```
23-10-14 11:11:36 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-27073' coro=<Repository.update() done, defined at /usr/src/supervisor/supervisor/store/repository.py:104> exception=StoreGitError()>
```

Typically this happens when the add-on git repository is not reachable in one way or another (e.g. Internet connectivity issue). Do you use PiHole or similar services?

claustro, 2023-10-14:
Thank you for spending your time answering me, No no Pi-hole, am using Adguard as an addon.
The fact that is going me crazy is that every time I restart HA I receive different errors in the supervisor log, but my system seems working correctly. I can't understand what this supervisor error should cause 

This is the last log after cleaning and restarting HA

23-10-14 14:36:02 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/BeardedTek/BeardedTek-hassio-addons repository
23-10-14 14:36:02 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/adamoutler/Addons repository
23-10-14 14:36:02 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
23-10-14 14:36:02 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
23-10-14 14:36:02 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-10-14 14:36:32 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
23-10-14 14:38:14 ERROR (MainThread) [supervisor.store.git] Can't update https://github.com/home-assistant/addons repo: Cmd('git') failed due to: exit code(128)
  cmdline: git fetch -v --update-shallow --depth=1 -- origin
  stderr: 'fatal: unable to access 'https://github.com/home-assistant/addons/': Failed to connect to github.com port 443 after 131002 ms: Couldn't connect to server'.
23-10-14 14:38:14 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_reset - store / core
23-10-14 14:38:14 INFO (MainThread) [supervisor.resolution.module] Create new issue corrupt_repository - store / core
23-10-14 14:38:14 ERROR (MainThread) [supervisor.store.git] Can't update https://github.com/marcelveldt/hassio-addons-repo repo: Cmd('git') failed due to: exit code(128)
  cmdline: git fetch -v --update-shallow --depth=1 -- origin
  stderr: 'fatal: unable to access 'https://github.com/marcelveldt/hassio-addons-repo/': Failed to connect to github.com port 443 after 131591 ms: Couldn't connect to server'.
23-10-14 14:38:14 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_reset - store / 62c7908d
23-10-14 14:38:14 INFO (MainThread) [supervisor.resolution.module] Create new issue corrupt_repository - store / 62c7908d
23-10-14 14:38:14 WARNING (SyncWorker_2) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS Server DEVELOPMENT
23-10-14 14:38:14 WARNING (SyncWorker_2) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Google Assistant Webserver
23-10-14 14:38:14 WARNING (SyncWorker_5) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of DeskPi Pro Active Cooling
23-10-14 14:38:14 WARNING (SyncWorker_4) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access) Beta (0.13.0)
23-10-14 14:38:14 WARNING (SyncWorker_5) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS Server DEVELOPMENT
23-10-14 14:38:14 WARNING (SyncWorker_5) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Google Assistant Webserver
23-10-14 14:38:14 WARNING (SyncWorker_0) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of DeskPi Pro Active Cooling
23-10-14 14:38:14 WARNING (SyncWorker_6) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access) Beta (0.13.0)
23-10-14 14:38:14 INFO (MainThread) [supervisor.store] Loading add-ons from store: 106 all - 0 new - 0 remove
23-10-14 14:38:14 INFO (MainThread) [supervisor.store] Loading add-ons from store: 106 all - 0 new - 0 remove
23-10-14 14:38:14 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-55949' coro=<Repository.update() done, defined at /usr/src/supervisor/supervisor/store/repository.py:104> exception=StoreGitError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/git.py", line 136, in pull
    await self.sys_run_in_executor(
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/git/remote.py", line 1014, in fetch
    res = self._get_fetch_info_from_stderr(proc, progress, kill_after_timeout=kill_after_timeout)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/git/remote.py", line 853, in _get_fetch_info_from_stderr
    proc.wait(stderr=stderr_text)
  File "/usr/local/lib/python3.11/site-packages/git/cmd.py", line 602, in wait
[... cut ...]
23-10-14 14:38:25 ERROR (MainThread) [supervisor.services.modules.mqtt] There is already a MQTT service in use from core_mosquitto
23-10-14 14:38:31 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
23-10-14 14:38:31 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-10-14 14:38:31 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-10-14 14:38:31 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running

closed 2024-01-20 by github-actions[bot]