home-assistant/operating-system #2817: Hassio 11.0 - No LAN after install

LinuxfarmerHH, 2023-10-14:
### Describe the issue you are experiencing

RPi4@SSD boot

Yesterday i got version 10.5 with clean reboot and everything work as expected. Today 11.0 followed, it was the last thing i could do with the device. Gone from the network, no ping, no connection.

Plugged onto workbench with HDMI, LAN and Keyboard

- System time set on 30sec delay, got no time.
Timeserver here is my home router with ntpd

-VFS/cifs error on mount, error code 101
This is the backup path on my NAS

ha network info gives
Own IP 172.30.30.0/23
DNS 172.30.30.3

Wrong, my network is 192.168.1.* and the DNS my running home router

There are not enough commands and options in the ha console to do anything about this problem. But for RPI4 this is wa warning, can be the last goodbye with hassio 11.0

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Click on Settings
2. Install hassio 11.0
3. hassio dead an gone
...

### Anything in the Supervisor logs that might be useful for us?

```txt
can't be accessed without network, only ad photo from the monitor
```

### Anything in the Host logs that might be useful for us?

```txt
Wrong IP and subnet, device cannot be pinged.
```

### Additional information

I wish we had something like a sudo login from the ha console, to access real linux commands to inspect and fix such issue.

agners, 2023-10-14:
> ha network info gives
> Own IP 172.30.30.0/23
> DNS 172.30.30.3

This is the IP address range of the internal Docker network. That is expected.

Are there more information further down? Specifically, there should be:

```
- connected: true
  enabled: true
  interface: end0
...
```

You can set a static IP using;

```
ha network update end0 --ipv4-address 192.168.87.59/24 --ipv4-gateway 192.168.87.1 --ipv4-method static
```

Or enable DHCP using:

```
ha network update end0 --ipv4-method auto
```

However, the update really shouldn't break your network. Once you were able to fix your network, can you share the Supervisor logs (from Settings > System > Logs, then choose Supervisor on the top right).

LinuxfarmerHH, 2023-10-14:
Did a full reinstall of 10.5 what for luck is the base for the rpi imager, after install some pieces start to worl again.

But the problem as photographed was

- connected: false
- enabled: false
- interface: end0
method IPV6
- disabled
method IPV6
- disabled

The running 10.5 does offer this and use as expected ETH0
interfaces:                                                                                                                                                           
- connected: true                                                                                                                                                     
  enabled: true
  interface: eth0
  ipv4:
    address:
    - 192.168.1.250/24
    gateway: 192.168.1.1
    method: auto
    nameservers:
    - 192.168.1.1
    ready: true

LinuxfarmerHH, 2023-10-15:
Have access to the corpse ssd now and fetched a system.journal file from the logs folder. But have no clue how to peek into this binary file, there should be the last breaths readable. Who can tell me a tool to read these files?

system@659eeb7e6edd4cf2ab0f8e882c7ad450-00000000009f6eba-000606c2a441a133.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a04fc1-000606dc3b689338.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a19623-000606dc42c0c839.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a2dc67-000606dc4b8f1dc5.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a40afb-000606e2733c5291.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a49e61-00060710d20a8b8a.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a531a9-0006073ffad610f8.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a5c4b2-000607700e3c0419.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a677b0-0005f880831a5671.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a6885e-0005f8808318e1a7.journal
system@659eeb7e6edd4cf2ab0f8e882c7ad450-0000000000a69cff-0005f880831ccb6c.journal
system.journal

a-sukhodolsky, 2023-10-15:
I have the same issue: no LAN available on Home Assistant OS 11.0 (Raspberry Pi 400).
I tried activate the end0 interface via `ha network update end0 --ipv4-method auto` but got the following in logs:
```
Oct 15 12:19:03 ha NetworkManager[454]: <info> [1697372343.1905] audit: op="connection-add-activate" pid=1067 uid =0 result="fail" reason ="failure adding connection: error writing to file '/etc/NetworkManager/system-connections/Supervisor end0.nmconnection': failed to create file /etc/NetworkManager/system-connections/Supervisor end0.nmconnection.QR7QC2: No space left on device"
```
If I do `ha login` and then `df -h /etc/NetworkManager/system-connections/` I get this:
```
Filesystem      1K-blocks   Used  Available  Use%  Mounted on
/dev/mmcblk0p7      84.5M  25.0M      52.8M   32%  /mnt/overlay
```
Any ideas on how to fix the issue without reinstalling OS?

a-sukhodolsky, 2023-10-15:
I resolved the issue via https://github.com/home-assistant/operating-system/issues/2752#issuecomment-1721313940.
The issue is not related to OS update, it is caused by the file system out-of-inodes issue (too many bluetooth devices in cache, more than 24k files, so new network connection file cannot be created).

FloMeyer, 2023-10-16:
I got the same problem after upgrading to 11.0.

I am lucky and have wifi connected, too.

```
➜  ~ ha network info end0
connected: true
enabled: true
interface: end0
ipv4:
  address: []
  gateway: null
  method: static
  nameservers: []
  ready: true
ipv6:
  address:
  - fe80::c523:be19:9c1f:3bfe/64
  gateway: null
  method: disabled
  nameservers: []
  ready: true
mac: E4:5F:01:A5:C8:60
primary: false
type: ethernet
vlan: null
wifi: null
```

```
➜  ~ ha network update end0 --ipv4-address 192.168.4.1/24
Error: Can't update config on end0: ipv4.gateway: gateway is invalid
```

I have no gateway on this interface!

LinuxfarmerHH, 2023-10-17:
Home Assistant 2023.10.3 Supervisor 2023.10.0 Operating System 11.0 Frontend 20231005.0 - latest

Okay checked the bluetooth cache, what possibly was the problem through peoples with mobile phones visiting the house. Was free and 11.0 did install.

Please fix this issue, too tiny folders and file system structures for caching is a bad problem. Think no one but few testers will really build a real world hassio without a SSD. So there is enough capacity, no i'm running on an old 32GB SSD.

moebis, 2023-10-17:
Upgraded to 11 today, and my Pi is basically dead/unresponsive. I can't reach it over the network (static IP address assigned via MAC ID). It just killed my Home Assistant install. How much testing was done on this release?

AliceGrey, 2023-10-19:
I'm having this issue as well on my RPI4. Device is offline after update to 11.0. Clearing the Bluetooth cache did not solve my issue.

agners, 2023-10-19:
> Please fix this issue, too tiny folders and file system structures for caching is a bad problem. Think no one but few testers will really build a real world hassio without a SSD. So there is enough capacity, no i'm running on an old 32GB SSD.

This is resolved in HAOS 11.0, see https://github.com/home-assistant/operating-system/pull/2751. The cache is purged weekly. According too [systemd.time documentation](https://www.freedesktop.org/software/systemd/man/latest/systemd.time.html) weekly is Monday at midnight. Maybe you didn't had Home Assistant OS 11 running for long enough for the first cache maintenance to take place?

inverse, 2023-10-31:
Not sure what other folks are facing but I just upgraded from 10.5 -> 11.1 after putting off doing this from reading reports on here.

It worked perfectly the first time.

I'm running on the PI4 with USB SSD boot.

Static IP  set from my router.

closed 2024-02-06 by github-actions[bot]