home-assistant/operating-system #2818: yalexs_ble cannot connect to lock

Frostis12, 2023-10-14:
### Describe the issue you are experiencing

Yale linus lock and bluetooth integration cannot connect after update to HA OS 11.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update HA OS to 11, before upgrade everything worked perfectly
2. Try to connect yale bluetooth lock
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
BLUETOOTH LOGS:

File "/usr/local/lib/python3.11/site-packages/yalexs_ble/session.py", line 269, in execute
    return await self._write(command, command_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/session.py", line 137, in _write
    return await self._locked_write(command, command_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/session.py", line 195, in _locked_write
    result = await future
             ^^^^^^^^^^^^
asyncio.exceptions.CancelledError: Interrupted by interrupt context manager

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/push.py", line 998, in _execute_deferred_update
    await self._update()
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/push.py", line 108, in _async_wrap_operation_lock
    return await func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/push.py", line 152, in _async_wrap_retry_bluetooth_connection_error
    return await func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/push.py", line 685, in _update
    lock = await self._ensure_connected()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/push.py", line 558, in _ensure_connected
    await self._client.connect()
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/lock.py", line 186, in connect
    await self._setup_session()
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/lock.py", line 244, in _setup_session
    response = await self.secure_session.execute(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/session.py", line 266, in execute
    async with interrupt(
  File "/usr/local/lib/python3.11/site-packages/async_interrupt/__init__.py", line 91, in __aexit__
    raise self._exception(self._message) from exc_val
yalexs_ble.session.DisconnectedError: ETUOVI (L60004K): Disconnected
2023-10-14 15:42:37.348 WARNING (MainThread) [homeassistant.config_entries] Config entry 'ETUOVI' for yalexs_ble integration not ready yet: ETUOVI (L60004K): Disconnected; Try moving the Bluetooth adapter closer to L60004K; Retrying in background
2023-10-14 15:42:38.332 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (00:1A:7D:DA:71:13) [connectable]: E2:82:3A:84:51:9D AdvertisementData(manufacturer_data={1177: b'\x05\x0f\xa4eZ\xb6\x97\xff\xd8\x00\x08\x03\xcc9v\xc9\xf1B\xe2\x82:\x84Q\x9d'}, rssi=-75) match: set()
2023-10-14 15:42:39.616 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (00:1A:7D:DA:71:13) [connectable]: E2:82:3A:84:51:9D AdvertisementData(manufacturer_data={1177: b'\x05\x0f\xa4eZ\xb6\x97\xff\xe0\x00\x08\x03\xc49v\xc9\xf1C\xe2\x82:\x84Q\x9d'}, rssi=-75) match: set()
2023-10-14 15:42:40.902 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (00:1A:7D:DA:71:13) [connectable]: E2:82:3A:84:51:9D AdvertisementData(manufacturer_data={1177: b'\x05\x0f\xa4e_\xb6\x97\xff\xdc\x00\x04\x03\xcc9v\xc9\xf1D\xe2\x82:\x84Q\x9d'}, rssi=-75) match: set()
2023-10-14 15:42:42.184 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (00:1A:7D:DA:71:13) [connectable]: E2:82:3A:84:51:9D AdvertisementData(manufacturer_data={1177: b'\x05\x0f\xa4eZ\xb6\x97\xff\xe4\x00\x10\x03\xc49v\xc9\xf1E\xe2\x82:\x84Q\x9d'}, rssi=-74) match: set()
[... 72 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/push.py", line 558, in _ensure_connected
    await self._client.connect()
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/lock.py", line 186, in connect
    await self._setup_session()
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/lock.py", line 263, in _setup_session
    response = await self.secure_session.execute(cmd, "SEC_INITIALIZATION_COMMAND")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/session.py", line 269, in execute
    return await self._write(command, command_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/session.py", line 137, in _write
    return await self._locked_write(command, command_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/yalexs_ble/session.py", line 190, in _locked_write
    async with util.asyncio_timeout(10):
  File "/usr/local/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
2023-10-14 15:31:43.280 DEBUG (MainThread) [homeassistant.components.bluetooth] Rediscovered adapters: {'hci0': {'address': '00:1A:7D:DA:71:13', 'sw_version': 'homeassistant', 'hw_version': 'usb:v1D6Bp0246d0545', 'passive_scan': True, 'manufacturer': 'cyber-blue(HK)Ltd', 'product': 'CSR8510 A10', 'vendor_id': '0a12', 'product_id': '0001'}}
```

### Additional information

BT adapter is "anne pro" from high performance list.

agners, 2023-10-14:
Can you try to downgrade to HAOS 10.5 to see if that indeed helps to fix this problem (`ha os update --version 10.5`).

bdraco, 2023-10-15:
Traveling at the moment so I can't test but I think its: 

https://github.com/bluez/bluez/issues/594
https://github.com/bluez/bluez/commit/670f0d0f618d3f122e6d95c34de53f59a343a15f

Which is fixed in 5.70
https://github.com/bluez/bluez/compare/5.69...5.70

bdraco, 2023-10-16:
I set up a blue with HAOS 11 and a local adapter with an August lock and confirmed the issue

99% sure the above commit will fix it

Happy to test a dev build when available

bdraco, 2023-10-16:
I didn't see it during the beta because all my locks are serviced via esphome proxies except the ones in Houston where I am not (and have not upgraded since I'm there at the moment)

adamghamel, 2023-10-17:
I'm having the same issue.

andrewraj, 2023-10-17:
I'm facing the same issue, tried changing the adapter. Removed Bluetooth, removed yale integration with no success. Reverting to HA OS 10.5 sorted the issue.

Frostis12, 2023-10-17:
Downgraded to HA OS 10.5 and issue is gone. Waiting for official fix to OS 11

agners, 2023-10-17:
This should be addressed with #2835.

MVB1989, 2023-10-18:
I have the same problem.  Before upgrading everything worked fine via bluetooth.

sairon, 2023-10-19:
The fix is included in `11.1.dev20231018`, if you're feeling adventurous, feel free to try and report back. I had virtually the same issue with [hass_nuki_bt](https://github.com/ronengr/hass_nuki_bt) custom integration and it works again correctly on the latest dev. For the less adventurous ones, please wait for stable 11.1.

Frostis12, 2023-10-19:
How to install dev builds?

bdraco, 2023-10-19:
https://www.home-assistant.io/common-tasks/os/

Be sure to make a backup first

sairon, 2023-10-24:
Home Assistant OS 11.1.rc1 with a fix for this issue has landed in the beta channel today for those less adventurous. Release to the stable channel is expected by the end of the week.

closed 2023-11-16 by sairon