home-assistant/operating-system #2819: HA stuck in "Started containerd container - Stopped containerd container loop" after HAOS 11 update atemped

diegoantilles, 2023-10-14:
### Describe the issue you are experiencing

Home Assistant is stuck in a loop trying to start the "containerd container".

[image]

This happens for several minutes, then the hole VM reboots, and the issue begins form start again

The instance is a VM running on a Proxmox virtualization server.
All the HAOS updates on this VM before worked flawless.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Press install HAOS 11 Option in the settings menu
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

agners, 2023-10-14:
After three startups the GRUB bootloader should select the other boot slot. If that is not the case, you can do that manually using arrow keys. Does choosing the other boot slot helps to resolve the issue?

webshaun, 2023-10-14:
This last update just killed my instance.  I didn't have much configured, no custom out of bound configuration. 

SSD for storage, 1 plugin to control fans on my pi case.  Beyond that, It's a pretty stock instance.

diegoantilles, 2023-10-15:
> After three startups the GRUB bootloader should select the other boot slot. If that is not the case, you can do that manually using arrow keys. Does choosing the other boot slot helps to resolve the issue?

After a few failed attempts to start the containerd contaiers and several forceful restarts via Proxmox, HA starts again and is running, but still on HAOS 10.5 with the notification to upgrade to HAOS 11. From there on, if I try updating, the issue stays the same.

hocomadvies, 2023-10-15:
I experience more or less the same issue. Also on Proxmox VM when push upgrade to 11 the spinning wheels stays for ever and nothing seems to happen. After reboot after long wait it still says there is an udate for HOAS. What would be the Cli command for running the update? Might try that see if that make a differences.

diegoantilles, 2023-12-09:
Unfortunately the problem still persists on my Proxmox instance and I'm stuck on OS 10.5 so far. Does any of you have an idea what could be causing the problem?

diegoantilles, 2024-03-13:
Still persists on my Proxmox instance and I'm still stuck on OS 10.5 so far... don't know how we will get rid of the issue.

closed 2024-06-20 by github-actions[bot]