home-assistant/operating-system #2820: firmware files under /lib/firmware/updates/ are not found anymore

c-MM, 2023-10-14:
### Describe the issue you are experiencing

The bluetooth firmware for my USB bluetooth dongle isn't loaded anymore:
`
[    1.949834] usb 1-1: Product: BCM20702A0
[    4.030557] Bluetooth: hci0: BCM: chip id 63
[    4.037926] Bluetooth: hci0: BCM: features 0x07
[    4.111545] Bluetooth: hci0: homeassistant
[    4.113241] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[    4.120311] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[    4.122368] Bluetooth: hci0: BCM: 'brcm/BCM20702A1-0b05-17cb.hcd'
[    4.124611] Bluetooth: hci0: BCM: 'brcm/BCM-0b05-17cb.hcd'
`
I have the file under:
`
# ls -l /lib/firmware/updates/bcrm/
total 36
-rw-r--r--    1 root     root         35000 Oct  4 09:16 BCM20702A1-0b05-17cb.hcd
`
The firmware was loaded under HAOS 10.5.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. reboot the system, watch dmesg output

### Anything in the Supervisor logs that might be useful for us?

```txt
no problem with supervisor
```

### Anything in the Host logs that might be useful for us?

```txt
see above
```

agners, 2023-10-14:
> The firmware was loaded under HAOS 10.5.

Are you sure? Unfortunately, from what I remember, at bootup that location wasn't available even on HAOS 10.5. 

On OS 11.0, can you plug the device out, reboot and plug it back in after bootup? That should work in either case.

c-MM, 2023-10-28:
Ok, I tried rebooting with and without the dongle attached and the firmware is nnever found.  

you might be right, that 10.5 didn't find the at boot, since I added the dongle and firmware only reccently and probably never rebooted 10.5 since. But at least 10.5 did find the firmware when booted. Neither 11.0 nor 11.1 find the firmware when the dongle is attached after booting the system.

closed 2024-02-04 by github-actions[bot]