home-assistant/operating-system #2821: Bluetooth disconnects after upgrade to 11. Plejd addon to HA.

waroen, 2023-10-15:
### Describe the issue you are experiencing

The Plejd addon to HA uses Bluuetooth to communicate to the Plejd actors/dimmers. This has been rock-solid until my upgrade to 11.

Now it times-out constantly and has to reconnect resulting in unreliable behavior.

Sending 8 byte(s) of data to Plejd. 0c01100098019999
2023-10-15 08:56:39 VRB [plejd-ble] Error pinging Plejd, calling onWriteFailed... In Progress
2023-10-15 08:56:39 DBG [plejd-ble] onWriteFailed #1 in a row. In Progress
DBusError: In Progress
    at _methodReturnHandlers.<computed> (/plejd/node_modules/dbus-next/lib/bus.js:343:27)
    at handleMessage (/plejd/node_modules/dbus-next/lib/bus.js:101:11)
    at EventEmitter.<anonymous> (/plejd/node_modules/dbus-next/lib/bus.js:151:9)
    at EventEmitter.emit (events.js:314:20)
    at /plejd/node_modules/dbus-next/lib/connection.js:116:14
    at Socket.<anonymous> (/plejd/node_modules/dbus-next/lib/message.js:63:9)
    at Socket.emit (events.js:314:20)
    at emitReadable_ (_stream_readable.js:557:12)
    at processTicksAndRejections (internal/process/task_queues.js:83:21)7

[...]

2023-10-15 09:08:20 VRB [plejd-ble] Error message: In Progress
2023-10-15 09:08:20 WRN [plejd-ble] Write error indicates BLE is disconnected. Retry count 5. Reconnecting...
2023-10-15 09:08:20 INF [plejd-ble] Starting reconnect loop...
2023-10-15 09:08:20 VRB [plejd-ble] Starting internal reconnect loop...
2023-10-15 09:08:20 VRB [plejd-ble] Reconnect: Clean up, emit reconnect event, wait 5s and the re-init...
2023-10-15 09:08:20 VRB [plejd-ble] cleanup() - Clearing ping interval and clock update timer
2023-10-15 09:08:20 VRB [plejd-ble] Removing listeners to write events, bus events and objectManager...
2023-10-15 09:08:20 VRB [plejd-ble] Reconnect attempt 1 in a row. Will power cycle every 10th time.
2023-10-15 09:08:20 INF [device-comm] Bluetooth reconnecting...
2023-10-15 09:08:20 VRB [device-comm] Stopping writeQueue loop until connection is established. Write queue length: 0
2023-10-15 09:08:25 INF [plejd-ble] Reconnecting BLE...
2023-10-15 09:08:25 INF [plejd-ble] init()

[...]

023-10-15 09:08:33 DBG [plejd-ble] onWriteFailed #1 in a row. Operation failed with ATT error: 0x0e
DBusError: Operation failed with ATT error: 0x0e
    at _methodReturnHandlers.<computed> (/plejd/node_modules/dbus-next/lib/bus.js:343:27)
    at handleMessage (/plejd/node_modules/dbus-next/lib/bus.js:101:11)
    at EventEmitter.<anonymous> (/plejd/node_modules/dbus-next/lib/bus.js:151:9)
    at EventEmitter.emit (events.js:314:20)
    at /plejd/node_modules/dbus-next/lib/connection.js:116:14
    at Socket.<anonymous> (/plejd/node_modules/dbus-next/lib/message.js:63:9)
    at Socket.emit (events.js:314:20)
    at emitReadable_ (_stream_readable.js:557:12)
    at processTicksAndRejections (internal/process/task_queues.js:83:21)
2023-10-15 09:08:33 VRB [plejd-ble] Error message: Operation failed with ATT error: 0x0e
2023-10-15 09:08:33 ERR [plejd-ble] 'Unlikely error' (0x0e) writing to Plejd. Will retry. Operation failed with ATT error: 0x0e

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Use Plejd-addon
2. Upgrade to 11
3. Watch the addon start failing to communicate.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing relating to Bluetooth there.
```

### Anything in the Host logs that might be useful for us?

```txt
No, nothing during the timeframe the error happened.
```

### Additional information

Also see issue https://github.com/icanos/hassio-plejd/issues/292 for mor discussion on the issue.

bdraco, 2023-10-15:
Possible its the same issue as https://github.com/home-assistant/operating-system/issues/2818#issuecomment-1763329706

sairon, 2023-10-19:
If bdraco is right, then it should be fixed in `11.1.dev20231018` build, see https://github.com/home-assistant/operating-system/issues/2818#issuecomment-1770057546

bdraco, 2023-10-19:
confirmed fixed in `11.1.dev20231018`

```
# ha os info
board: odroid-n2
boot: A
data_disk: G1J39E-0x0a9b25a6
update_available: false
version: 11.1.dev20231018
version_latest: "11.0"
# # bluetoothctl --version
# bluetoothctl: 5.70
```

sairon, 2023-10-24:
Home Assistant OS 11.1.rc1 with a fix for this issue has landed in the beta channel today for those less adventurous. Release to the stable channel is expected by the end of the week.