home-assistant/operating-system #2822: Unable to boot after upgrade to 11.0 on odroid M1

a1j, 2023-10-15:
### Describe the issue you are experiencing

There are multiple reports of this issue, please read [here](https://community.home-assistant.io/t/update-to-haos-11-on-odroid-m1-fails-need-to-restore-backup/626483).

I

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have 10.5 installed, M1 using dedicated SSD for data.
2. Upgrade to 11.0 using HASS UI
3. The system will come up using 11.3 kernel, with root filesystem mounted as read-only and HASS data partition not mounted at all.  If you log in via system prompt (hass-cli is not working because containerd is waiting for filesystem mount) you can mount data directory manually, after that homeassistant starts, but with old OS.

If you reboot the system you would have to repeat mount procedure again.

I looked at the journalctl logs and the only error i foudn that "hass filesystem mount by label" job times out and restarts.  

BUT this may not mean anything since this is boot of old hass version (upgrade failed at some point). 

When ui comes up it asks you to upgdate to 11.0 version again.

It would be nice to know how to fix hass from this half-broken state.

### Anything in the Supervisor logs that might be useful for us?

```txt
I cannot generate logs because upgrade did not complete and it now runs in semi-broken state using old OS image.
```

### Anything in the Host logs that might be useful for us?

```txt
I looked at journalctl and it says it "times out" while executing mout hass by label job, and then it reports that dependency is broken. But this can be an artifact of incomplete upgrade, i cannot find logs of upgrade anywhere.
```

### System information

Odroid M1, SD card, dedicated data disk disk (ssd). upgrade from 10.5 to 11.0 using Hass Web UI.

a1j, 2023-10-15:
I wonder if this is related https://github.com/home-assistant/operating-system/issues/2573

prblase, 2023-10-15:
Same happened at me. I realiezed after a clean install HAOS can not migrate to ssd data disk, because on restart it is not mounting.

matjahs, 2023-10-15:
Same issue here.
The machine boots and will say it's waiting for the HA CLI to start. Upon further inspection it seems like there is expecting a disk partition with the label `hassos-data`. However, the partion that should have that label is labeled `hassos-data-dis`. 
When I proceed to manually re-label the partition, it goes off and does its thing, sort of. After a reboot it still doesn't appear to be happy with my data-disk. I ended up doing a clean install and restoring my most recent backup.

a1j, 2023-10-15:
>  I ended up doing a clean install and restoring my most recent backup.

From this [forum post](https://community.home-assistant.io/t/update-to-haos-11-on-odroid-m1-fails-need-to-restore-backup/626483) it looks like after clean install issues with data disk will still appear after migration to a dedicated SSD.

regan-a, 2023-10-16:
I've hit the same issue. Noticed the same issue as @matjahs with the partition label mismatch. Happy to collect logs from my system if that helps get the root of this, just let me know what to grab.

escoand, 2023-10-16:
I took this manual workaround:

1.  switch to second console Ctrl+Alt+F2
2. wait for the prompt
3. enter root
4. mount the data disk: `mount /dev/nvme0n1p1 /mnt/data`
5. switch back to first console Ctrl+Alt+F2
6. it should enter emergency mode
7. enter `login`

With this i was able to get the system running again, but the update was not installed...

agners, 2023-10-16:
Have you all been using the data disk feature when this happened on upgrade?

The relabel to `hassos-data-dis` should only happen when multiple disks are attached with the same label. It seems that this misfired in you cases (initiated by [this script](https://github.com/home-assistant/operating-system/blob/11.0/buildroot-external/rootfs-overlay/usr/libexec/haos-data-disk-detach)).

The fact that this happened even after restart @matjahs seems to indicate that this is reproducible. I am trying to reproduce it here now.

agners, 2023-10-16:
I've installed HAOS 10.5 from scratch, added an NVMe, and then upgraded to 11.0. In my case things worked out fine.

However, I do have a suspicion what the problem is: The script above is triggered by the `haos-data-disk-detach.service` service. This service should only be run once on very first boot (as mandated by `ConditionFirstBoot=yes`).

Now for some reason, in your cases, the system thought the OS is booting for the first time :thinking: 

The first boot is determined by the U-Boot boot loader on startup, if machine id is not being set. I am currently unclear how this could fail in some situation. Ideally I'd need boot logs capture via a serial console. @regan-a is that something you have the tools for?

Also, @all do yo use an eMMC or SD card device?

regan-a, 2023-10-16:
Hey @agners, thanks for looking into this! I'm running an SD + NVMe for data. Here is a dump of journalctl after boot. Let me know if you need anything else.

[boot.txt](https://github.com/home-assistant/operating-system/files/12918066/boot.txt)

agners, 2023-10-16:
> Hey @agners, thanks for looking into this! I'm running an SD + NVMe for data. Here is a dump of journalctl after boot. Let me know if you need anything else.

From the logs I can confirm my suspicion: Your boot loader triggers first boot mode.

```
... systemd.machine_id= fsck.repair=yes systemd.condition-first-boot=true ..
```

The question is why exactly. Unfortunately the boot loader output isn't available through logs, only through a serial console.

Can you run the following command on the OS shell?

```
fw_printenv
```

Also can you check the SHA256 of the boot script:

```
sha256sum /mnt/boot/boot.scr
```

If anyone has access to the serial console of that board which is suffering from the problem, the capture of the boot loader phase would be helpful.

regan-a, 2023-10-16:
My apologies, I misunderstood the request. I've attached the outputs of the two commands, plus a serial dump of the boot.

[fw_printenv.txt](https://github.com/home-assistant/operating-system/files/12919073/fw_printenv.txt)
[boot.scr.sha256.txt](https://github.com/home-assistant/operating-system/files/12919074/boot.scr.sha256.txt)
[serial boot.txt](https://github.com/home-assistant/operating-system/files/12919079/serial.boot.txt)

agners, 2023-10-17:
> [fw_printenv.txt](https://github.com/home-assistant/operating-system/files/12919073/fw_printenv.txt) 

Hm, it seems the system did write a new machine id, and at least the running OS is able to read the U-Boot environment :thinking: 

> [boot.scr.sha256.txt](https://github.com/home-assistant/operating-system/files/12919074/boot.scr.sha256.txt) 

This is the correct hash of the boot script in HAOS 11.0, so your boot script doesn't seem corrupted or anything.

> [serial boot.txt](https://github.com/home-assistant/operating-system/files/12919079/serial.boot.txt)

This does show the problem really:

```
** Booting bootflow 'mmc@fe2b0000.bootdev.part_2' with script
loading env...
Card did not respond to voltage select! : -110
## Error: bad CRC, import failed
0 bytes read in 1 ms (0 Bytes/s)
0 bytes read in 1 ms (0 Bytes/s)
Loading standard device tree rk3568-odroid-m1.dtb
116634 bytes read in 13 ms (8.6 MiB/s)
Working FDT set to a100000
Trying to boot slot A, 2 attempts remaining. Loading kernel ...
29901312 bytes read in 1564 ms (18.2 MiB/s)
storing env...
Card did not respond to voltage select! : -110
Starting kernel
Moving Image from 0x2080000 to 0x2200000, end=3f30000
## Flattened Device Tree blob at 0a100000
   Booting using the fdt blob at 0xa100000
Working FDT set to a100000
   Loading Device Tree to 00000000ede61000, end 00000000edee5fff ... OK
Working FDT set to ede61000
```

It seems that the U-Boot bootloader is not able to read the environment. :cry:

JirikP, 2023-10-17:
I am having this exact problem. Upgraded to HAOS 11.0 and it never came back. I had a backup on google drive, so i installed the HAOS 11.0 on SD card, restored the backup and from SD card, everything is fine. I just cannot move the Data disk to the SSD (sata). If i try to move it, it will get stuck on waiting for the CLI to get ready.

dbpickles, 2023-10-18:
Exact same problem too.

cryptoluks, 2023-10-19:
For me, the Web UI Upgrade process from 10.5 to 11 did not result in a reboot (tried it several times). However, after resetting the device days later I had the same issues as described here.

On each boot now, the M.2 SSD partition is disabled and booting is only possible when re-labeling it and restarting the docker containers.

However, it seems that I am still on 10.5 according to the web interface.

peterf9, 2023-10-20:
same issue here... cannot move data to NVMe.

gutarin, 2023-10-22:
I ran into the same issue on my Odroid-M1

matjahs, 2023-10-23:
> For me, the Web UI Upgrade process from 10.5 to 11 did not result in a reboot (tried it several times). However, after resetting the device days later I had the same issues as described here.
>
>
>
> On [...]

Same thing here. After each system reboot, I will have a partition labeled `hassos-data-old` on my SD card and one labeled `hassos-data-dis` on the NVMe SSD. If I then relabel the disabled one to `hassos-data`, it continues booting as I would normally expect.

cryptoluks, 2023-10-23:
Could  anybody with a build setup provide a `boot.scr` with a commented-out first-boot check? So we could at least use the m.2 for data without systemd invoking the rename script, right? Correct, @agners? :-)

Thank you very much.

agners, 2023-10-23:
@cryptoluks that would be a possible work around, I should be able to create such a script.

Are you using an SD card?

Ki-csen, 2023-10-23:
Same issue here, I have just not realized any problem after the upgrade until a hard reboot by a power outage. Then it gets stuck on waiting for the CLI to get ready forever. My one boots from SD and data is on an SSD.

cryptoluks, 2023-10-23:
> Are you using an SD card?

Yes.

cryptoluks, 2023-10-23:
@agners I just created a new `boot.scr` using `mkimage` with your change from https://github.com/home-assistant/operating-system/pull/2856.

I can now observe on the hypervisor:

```
# journalctl -b | grep -i first
Oct 23 18:48:41 homeassistant systemd[1]: HAOS data disk detach was skipped because of an unmet condition check (ConditionFirstBoot=yes).
Oct 23 18:48:43 homeassistant systemd[1]: First Boot Complete was skipped because of an unmet condition check (ConditionFirstBoot=yes).
```

```
# dmesg | grep -i "kernel command line:"
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 systemd.machine_id=[snip] fsck.repair=yes  root=PARTUUID=[snip] rootfstype=squashfs ro rootwait rauc.slot=A
```

I did not yet move to the m.2 again for data, but it seems to work now after adding `mmc dev ${devnum}`. The `machine_id` in the kernel command line was populated and therefore the systemd units were skipped. Awesome!

Here are my steps:

```
apt-get -y install u-boot-tools
mkimage -T script -A arm64 -C none -n 'Fixed Boot' -d uboot-boot.ush boot.scr
```

The encoded new [uboot-boot.ush](https://github.com/agners/operating-system/blob/86d4ec2a0c012c28c15d4736c160c8e70536a015/buildroot-external/board/hardkernel/odroid-m1/uboot-boot.ush). has to be placed in `/mnt/boot/boot.scr`.

I also created a base64 encoded version from this [here](https://gist.github.com/cryptoluks/82e2b1c3105c85d91e1c225b8938eca0) for the adventurous. Simply decode it with `curl -s https://gist.githubusercontent.com/cryptoluks/82e2b1c3105c85d91e1c225b8938eca0/raw/2c35340a6486f11ef4d267cd5c5887c2deeedc60/gistfile1.txt | base64 -d > boot.scr`.

```
# md5sum /mnt/boot/boot.scr
bc45f05698d4fefecbba711da0a71052  /mnt/boot/boot.scr
```

cryptoluks, 2023-10-24:
Works now also with data on the m.2 SSD without any issues.

Btw, is the boot partition also updated when updating the operating system?

agners, 2023-10-24:
This will be addressed with HAOS 11.1. You can test it already by updating to 11.1.rc1 on the beta channel.

closed 2023-10-24 by agners

Ki-csen, 2023-10-25:
So, I have managed to replace boot.scr and upgrade to HAOS 11.0 but the sda1 still not auto mounted. Any idea?

cryptoluks, 2023-10-25:
> So, I have managed to replace boot.scr and upgrade to HAOS 11.0 but the sda1 still not auto mounted. Any idea?

Was your disk renamed before to the disabled label? If yes, you probably have to manually rename it back.

Or try the latest Beta with the fixes, maybe this works better for you.

Ki-csen, 2023-10-25:
> Edit: Ah, if you replaced it and then upgraded, it think the boot.scr was simply replaced with the bugged one from 11.0.

No, the md5sum looks right for the new boot.scr.

qwazerty, 2023-10-25:
Tested the HAOS 11.1.rc1, I can confirm that it fixed the issue and upgraded successfully on ODROID-M1 with SSD on nvme.

Before the upgrade to 11.1.rc1, when I tried to upgrade to 11.0. After reboot I had to change label
```
System information
  OS Version:               Home Assistant OS 10.5
  Home Assistant Core:      2023.10.3

~ # ls -l /dev/disk/by-label/
total 0
lrwxrwxrwx    1 root     root            15 Oct 26 01:09 hassos-boot -> ../../mmcblk1p2
lrwxrwxrwx    1 root     root            15 Oct 26 01:12 hassos-data-dis -> ../../nvme0n1p1
lrwxrwxrwx    1 root     root            15 Oct 26 01:09 hassos-data-old -> ../../mmcblk1p9
lrwxrwxrwx    1 root     root            15 Oct 26 01:09 hassos-overlay -> ../../mmcblk1p8
~ # e2label /dev/nvme0n1p1 hassos-data
~ # ls -l /dev/disk/by-label/
total 0
lrwxrwxrwx    1 root     root            15 Oct 26 01:09 hassos-boot -> ../../mmcblk1p2
lrwxrwxrwx    1 root     root            15 Oct 26 01:12 hassos-data -> ../../nvme0n1p1
lrwxrwxrwx    1 root     root            15 Oct 26 01:09 hassos-data-old -> ../../mmcblk1p9
lrwxrwxrwx    1 root     root            15 Oct 26 01:09 hassos-overlay -> ../../mmcblk1p8
```

After the upgrade
```
System information
  OS Version:               Home Assistant OS 11.1.rc1
  Home Assistant Core:      2023.10.3

~ # ls -l /dev/disk/by-label/
total 0
lrwxrwxrwx    1 root     root            15 Oct 26 01:29 hassos-boot -> ../../mmcblk1p2
lrwxrwxrwx    1 root     root            15 Oct 26 01:29 hassos-data -> ../../nvme0n1p1
lrwxrwxrwx    1 root     root            15 Oct 26 01:29 hassos-data-old -> ../../mmcblk1p9
lrwxrwxrwx    1 root     root            15 Oct 26 01:29 hassos-overlay -> ../../mmcblk1p8
```

arunderwood, 2023-10-29:
Incase it helps anyone else, here are the (unoptimized) steps I took to fix my install.  I highly suspect there are some unnecessary steps in here but this fixed mine so I can't easily go back and try to shorted them.  Normally my setup is headless so I had to hook up a keyboard, monitor, and ethernet:

* switch to second console Ctrl+Alt+F2
* wait for the prompt
* enter `root`
* mount the data disk: `mount /dev/nvme0n1p1 /mnt/data`
* switch back to first console Ctrl+Alt+F1
* it should enter emergency mode
* enter `login`
* switch to second console Ctrl+Alt+F2
* enter e2label `/dev/nvme0n1p1 hassos-data` and `systemctl restartstart hassos-supervisor`
* switch back to first console Ctrl+Alt+F1
* enter `os update` to install HASSOS 11.1
* switch to second console Ctrl+Alt+F2
* enter `reboot`

After the reboot everything worked like a charm.

rklasen, 2023-11-02:
Just to make sure, does that already mean HassOS can be booted from nvme directly, without an sd card or an emmc module?

The [official docs] still say it's impossible, but [this repo](https://github.com/inindev/odroid-m1/issues/4) seems to have gotten it working.