home-assistant/operating-system #2824: some addons including nginx proxy is gone after upgrade to v11

Vladx71, 2023-10-15:
### Describe the issue you are experiencing

Can't start several addons including nginx which is my reverse proxy after the upgrade. The error for all the addons are similar to this

`Can't start addon_core_nginx_proxy: 500 Server Error for http+docker://localhost/v1.43/containers/a3b3fd7bb1b40b32b3d61ba3023be8549d8773ae5dd6e19c1b2e90491651e222/start: Internal Server Error ("driver failed programming external connectivity on endpoint addon_core_nginx_proxy (264a54065462bfa7881aab08817314b2823c7daa1a7fdbe815d092a3927fc116): Error starting userland proxy: ")`

Googled, searched the forums but I can't find any recent report on it, so I decided to open an issue

Could you please check this?

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

6.1.56

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.upgrade to v11
2.restart
3.several addons including nginx failed to start with error message
`Can't start addon_core_nginx_proxy: 500 Server Error for http+docker://localhost/v1.43/containers/a3b3fd7bb1b40b32b3d61ba3023be8549d8773ae5dd6e19c1b2e90491651e222/start: Internal Server Error ("driver failed programming external connectivity on endpoint addon_core_nginx_proxy (264a54065462bfa7881aab08817314b2823c7daa1a7fdbe815d092a3927fc116): Error starting userland proxy: ")`
...

### Anything in the Supervisor logs that might be useful for us?

```txt
similar for the newly installed portainer addon

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:138
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 12:17:21 (6 occurrences)
Last logged: 12:39:00

Failed to to call /addons/db21ed7f_portainer/start - Can't start addon_db21ed7f_portainer: 500 Server Error for http+docker://localhost/v1.43/containers/4e960eb3903ce760b9abdffab32d4af6be6940527616f9b9f4e64af5009d1fd4/start: Internal Server Error ("driver failed programming external connectivity on endpoint addon_db21ed7f_portainer (455e50f60a61929983704c44de41081014555926959a25f81fbc28d214ffcbe2): Error starting userland proxy: ")
Failed to to call /addons/core_nginx_proxy/start - Can't start addon_core_nginx_proxy: 500 Server Error for http+docker://localhost/v1.43/containers/5b276c67d8787d4ae73a4bceed9a0f45e3d2857a16988ac539a68a6f632e343f/start: Internal Server Error ("driver failed programming external connectivity on endpoint addon_core_nginx_proxy (f3be89f15882a05c882ba0460fb82b184ad9eb0f05a16ecb6114c51a85d1bf66): Error starting userland proxy: ")
Failed to to call /addons/db21ed7f_portainer/start - Can't start addon_db21ed7f_portainer: 500 Server Error for http+docker://localhost/v1.43/containers/c401940dae93bf50f66d1d59a251b785739b1b6e93c9238adb3c21e2249141c3/start: Internal Server Error ("driver failed programming external connectivity on endpoint addon_db21ed7f_portainer (c011560f6643e1d25564dfd684d4809237f763094824707478ed7cf54afbcd93): Error starting userland proxy: ")
Failed to to call /addons/a0d7b954_sonweb/start - Can't start addon_a0d7b954_sonweb: 500 Server Error for http+docker://localhost/v1.43/containers/c8f423430adbd4fd06b6e9416f2395489b62d42c84c2e289b8c2417643d647f5/start: Internal Server Error ("driver failed programming external connectivity on endpoint addon_a0d7b954_sonweb (0d080708538b313dec8b63f7b9eae104036f2ab809cfaedfbebf0eb2547d9117): Error starting userland proxy: ")
Failed to to call /addons/core_nginx_proxy/start - Can't start addon_core_nginx_proxy: 500 Server Error for http+docker://localhost/v1.43/containers/a3b3fd7bb1b40b32b3d61ba3023be8549d8773ae5dd6e19c1b2e90491651e222/start: Internal Server Error ("driver failed programming external connectivity on endpoint addon_core_nginx_proxy (264a54065462bfa7881aab08817314b2823c7daa1a7fdbe815d092a3927fc116): Error starting userland proxy: ")
```

### Anything in the Host logs that might be useful for us?

```txt
Oct 15 10:44:55 homeassistant systemd[1]: Started Process Core Dump (PID 8703/UID 0).
Oct 15 10:44:55 homeassistant systemd-coredump[8704]: Process 8699 (docker-proxy) of user 0 dumped core.
Oct 15 10:44:55 homeassistant systemd[1]: systemd-coredump@20-8703-0.service: Deactivated successfully.
Oct 15 10:44:55 homeassistant dockerd[429]: time="2023-10-15T10:44:55.600395324Z" level=warning msg="Failed to allocate and map port 4357-4357: Error starting userland proxy: "
Oct 15 10:44:55 homeassistant kernel: hassio: port 9(veth3f721a3) entered disabled state
Oct 15 10:44:55 homeassistant kernel: device veth3f721a3 left promiscuous mode
Oct 15 10:44:55 homeassistant kernel: hassio: port 9(veth3f721a3) entered disabled state
Oct 15 10:44:55 homeassistant dockerd[429]: time="2023-10-15T10:44:55.619068217Z" level=error msg="Handler for POST /v1.43/containers/53b2134b2958b3b704777ee9a8e5c5ea7c7aa86bd2eb12f070264225f53894f4/start returned error: driver failed programming external connectivity on endpoint hassio_observer (3ff56c98f9521fda2615bbc626e70a9bd097d9c7bf435d2a2f5622a508a28c46): Error starting userland proxy: "
```

### System information

## System Information

version | core-2023.10.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | x86_64
timezone | Europe/Budapest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4816
Installed Version | 1.33.0
Stage | running
Available Repositories | 1310
Downloaded Repositories | 23

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 30.8 GB
disk_used | 12.4 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (4.9.0), Whisper (1.0.0), Log Viewer (0.15.1), Studio Code Server (5.11.0), Duck DNS (1.15.0), Advanced SSH & Web Terminal (15.0.8), Nginx Proxy Manager (0.12.3), NGINX Home Assistant SSL proxy (3.5.0), Portainer (2.19.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 15
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2023. október 8. 02:12
-- | --
current_recorder_run | 2023. október 15. 12:12
estimated_db_size | 197.55 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

agners, 2023-10-16:
I am using the NGINX Home Assistant SSL proxy add-on successfully on HAOS 11.0, so there shouldn't be a general issue.

However:

```
Oct 15 10:44:55 homeassistant systemd-coredump[8704]: Process 8699 (docker-proxy) of user 0 dumped core.
```

this is probably the underlying issue. It seems as if Docker can't setup the network configuration correctly, specifically the docker-proxy binary. I haven't seen that before. 

What type of VM Hypervisor are you using?

Can you test a second, new installation from scratch on the same machine to see if that is reproducible?

Vladx71, 2023-10-16:
Hi,

This is Virtualbox (because I was struggling with KVM on the same machine a
lot :)). If this is not a generic issue, I would probably accelerate the
process by moving HA to a dedicated HW (I already planned this, but I was
too lazy to start)

However, it would be great if I can fix this issue, so maybe I'll pull up
another instance and restore the config on it, maybe I have some Filesystem
issue

L:

agners, 2023-10-16:
> However, it would be great if I can fix this issue, so maybe I'll pull up another instance and restore the config on it, maybe I have some Filesystem issue

Yeah, if this is a reproducible issue, I agree, would be nice to pinpoint it!

Vladx71, 2023-10-16:
Hi,

So, I installed a new v11 instance, restored the full backup from the old
one and the errors I see are slightly different. It seems any addon depends
on nginx is failed to start

[13:43:58] INFO: Generating dhparams (this will take some time)...
Generating DSA parameters, 4096 bit long prime
....+...+....+...+.....+..+.+.+.......+....+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*
....+......+.......+............+..+........+.....+...+.+.+.................+.........+...+.+...........+...+..............+.......+.........+.............+..........+........+................+....................+.+....+............+.................+..+.....+...+.....+...............................+.........+.....+.+.+...................+............+...................+...................................+..........+...........+.....+.+.+.+.....................+.........+.+.....+.+.......+......+.........+.+......................+.........+..+................+.......+......+...+....................+.......................+....+..+........+............+..............+...........................+.+........................+....+.....+.................................+....+.+.+....+...........+.........+....+..............+...................+...+............+...+.......+.+.......+.....+.+....+.+......+............+.+......+..................+....+.+....+.......+.......................+...........+...+...............+......+..+...+..+..+.+....+........+.....+....+.........+........+.+...+..............+....+..............+............+.+....................+....+.............+...........+.+.+.+....+........................+.....+....+.........+...........+......+...+.+..+.....+...+.............+...........+.............................+.................................+.....+.+.+.+..+............+......+..+....+..+..+.......+...+...+.......+....+.............+..+......+..+..........+....+......+..........+.......+.+...........+..+.....................+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*
[13:44:05] INFO: Running nginx...

Nothing interesting in other logs except maybe this, but I'm not sure
if this is related, especially because it is just a warning

Logger: homeassistant.components.recorder.util
Source: components/recorder/util.py:323
Integration: Recorder (documentation
<https://www.home-assistant.io/integrations/recorder>, issues
<https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+recorder%22>)
First occurred: 13:07:50 (1 occurrences)
Last logged: 13:07:50
The system could not validate that the sqlite3 database at
//config/home-assistant_v2.db was shutdown cleanly

agners, 2023-10-16:
The sqlite3 database warning is appearing currently as we don't stop the active session at backup time currently.

So with the NGnix starting, things seem to work again then? So it does seem like a corrupted OS image or something similar caused it?

Vladx71, 2023-10-16:
> Not really working as intended. It seems the whole HACS messed up a bit
> because I have no errors when I start an addon, but they are not really
> started... I'm not sure if this is some kind of corruption in any of the
> important files or if this is a bug introduced when I upgraded to v11...
>
> L:

Maybe not HACS but the whole addon part... this is really strange. For
example I added onedrive-backup, configured it, started -> no errors in
log, backup complete, but it is not really running because I cannot open
the frontend of the addon...(always asking "do you want to start addon?)

Any idea what tiny part of the system has gone south?

Thanks
L:

Vladx71, 2023-10-16:
> Ok, so it is becoming more and more interesting. I've reinstalled nginx
> addon, started it and I'm still not seeing any error and UI shows it is not
> started, however it is working as intended and my external connection is
> restored.... it seems this is the same for all addons, showing "red circle"
> -> not started, but "backend" part is working, only  the "frontend" part
> (sidebar) is not...I already tried to clear my browser cache
>
> L:

One more interesting thing... after reinstalling HACS, it's looks quite
different :)

[image: image.png]

agners, 2023-10-17:
Unfortunately the picture didn't made it through to GitHub, so I can't see what you shared.

I expect this remaining issues to be not related to the update of HAOS 11.0. You can downgrade HAOS to 10.5 using `ha os update --version 10.5` to see if the issues go away.

closed 2023-10-18 by agners