home-assistant/operating-system #2825: Bluetooth not working for Intel AX101

felipecrs, 2023-10-15:
### Describe the issue you are experiencing

Note: this issue is not new, my bluetooth card never worked in HAOS. I was just waiting for 11.0 before reporting.

I have a Beelink EQ12 mini computer (super recommended, very affordable and extremely powerful) with a [Intel AX101](https://www.intel.com.br/content/www/br/pt/products/sku/203014/intel-wifi-6-ax101/specifications.html) wi-fi + bluetooth card.

However, bluetooth does not work. Demonstration:

https://github.com/home-assistant/operating-system/assets/29582865/ca4148ad-4186-4a83-831e-f79bf961a28c
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

N/A

### Anything in the Supervisor logs that might be useful for us?

```txt
No.
```

### Anything in the Host logs that might be useful for us?

```txt
Oct 15 17:47:00 homeassistant systemd[1]: Stopped target Bluetooth Support.
Oct 15 17:47:00 homeassistant kernel: usb 1-10: reset full-speed USB device number 2 using xhci_hcd
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: Device revision is 2
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: Secure boot is enabled
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: OTP lock is enabled
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: API lock is enabled
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: Debug lock is disabled
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: Minimum firmware build 1 week 10 2014
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
Oct 15 17:47:00 homeassistant kernel: Bluetooth: hci0: Failed to load Intel firmware file intel/ibt-0040-1050.sfi (-2)
Oct 15 17:47:00 homeassistant systemd[1]: Reached target Bluetooth Support.
Oct 15 17:47:19 homeassistant kernel: kauditd_printk_skb: 34 callbacks suppressed
Oct 15 17:47:19 homeassistant kernel: audit: type=1334 audit(1697392039.635:852): prog-id=210 op=LOAD
Oct 15 17:47:19 homeassistant systemd-timesyncd[481]: Network configuration changed, trying to establish connection.
Oct 15 17:47:19 homeassistant systemd[1]: Started Journal Gateway Service.
Oct 15 17:47:19 homeassistant systemd-timesyncd[481]: Contacted time server 162.159.200.1:123 (time.cloudflare.com).
Oct 15 17:47:19 homeassistant systemd-journal-gatewayd[573228]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: Device revision is 2
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: Secure boot is enabled
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: OTP lock is enabled
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: API lock is enabled
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: Debug lock is disabled
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: Minimum firmware build 1 week 10 2014
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
Oct 15 17:47:36 homeassistant kernel: Bluetooth: hci0: Failed to load Intel firmware file intel/ibt-0040-1050.sfi (-2)
Oct 15 17:47:41 homeassistant systemd[1]: Stopped target Bluetooth Support.
```

### System information

## System Information

version | core-2023.10.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | x86_64
timezone | America/Sao_Paulo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1320
Downloaded Repositories | 11
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 468.7 GB
disk_used | 377.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.15.0), Advanced SSH & Web Terminal (15.1.0), NGINX Home Assistant SSL proxy (3.5.0), Studio Code Server (5.12.0), ESPHome (2023.9.3), Mosquitto broker (6.3.1), Double Take (1.13.11.4), WireGuard (0.9.0), Frigate Beta (0.13.0) (0.13.0-beta2), Exadel CompreFace (1.2.0), OneDrive Backup (2.2.2), Tailscale (0.11.1), Netdata (1.42.4-2), Glances (0.19.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 3
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 7, 2023 at 4:09 AM
-- | --
current_recorder_run | October 14, 2023 at 12:19 PM
estimated_db_size | 707.93 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.2 (2ad1cd7)
-- | --
cloud_online | 4 / 6
local_online | 4 / 4

</details>

### Additional information

Home Assistant core logs:

```
Logger: bluetooth_auto_recovery.recover
Source: components/bluetooth/util.py:78
First occurred: October 14, 2023 at 12:52:09 PM (7983 occurrences)
Last logged: 2:43:55 PM

Could not determine the power state of the Bluetooth adapter hci0 [00:00:00:00:00:00] due to timeout after 5 seconds
Could not cycle the Bluetooth adapter hci0 [00:00:00:00:00:00]: [Errno 2] No such file or directory
Bluetooth adapter hci0 [00:00:00:00:00:00] could not be reset:
Closing Bluetooth adapter hci0 [00:00:00:00:00:00] failed: [Errno 9] Bad file descriptor
```

```
Logger: bluetooth_auto_recovery.recover
Source: runner.py:186
First occurred: October 14, 2023 at 12:52:09 PM (1996 occurrences)
Last logged: 2:43:50 PM

Bluetooth management socket connection lost: [Errno 22] Invalid argument
```

tonizombie, 2023-10-16:
+1 Even same computer (Beelink EQ12)

I read on home assistant forums that AX101 isnt supported by current linux kernel that HAOS currently uses
https://community.home-assistant.io/t/beelink-n305-eq12-pro-bluetooth-hci0-failed-to-load-intel-firmware-file-intel-ibt-0040-1050-sfi-2/616207

Only recommendations i see is wait or get cheap bluetooth dongle

felipecrs, 2023-10-16:
Well, if HAOS team want, they can include the driver themselves:

- https://aur.archlinux.org/packages/iwlwifi-ax101-dkms

I would understand however if they decide not to do it.

GaryOkie, 2023-11-21:
This issue is a duplicate of https://github.com/home-assistant/core/issues/98523 opened 4 months ago.   There are reports that a symlink to BT drivers that already exist in pre-6.5 kernels works properly.   On a HAOS system, I'm not sure how that can be accomplished for testing.    I was hoping that a release candidate could include it for general testing.

e1ace, 2024-01-05:
Is there an indication as to when HOAS will move to Linux 6.6 LTS to resolve this issue?

martinw72, 2024-01-09:
Is there an option to include the firmware files earlier?

sunshad0w, 2024-01-16:
the same issue (Beelin mini S12)

closed 2024-01-17 by sairon