home-assistant/operating-system #2840: Network storage cant connect to Samba share , error : Excess Arguments

vks007, 2023-10-18:
### Describe the issue you are experiencing

I have my HA running on machine 1 (Debian) and Frigate running on another machine 2 (Debian). As Frigate stores all recordings on the local disk (SSD) of machine 2 , I want to add that to the media storage in HA. 
to enable this I am running Samba share as a docker container on machine 2 to expose the frigate folder. I am able to browse that share well from machine 1 via the smbclient but when I try to configure it via “Add Network Storage” in HA , it fails with an error `Excess arguments.`

Connection from the HA machine to the Frigate machine running samba:

```
~$ smbclient //192.168.1.104/frigate -U ha_samba
Enter WORKGROUP\ha_samba's password:
Try "help" to get a list of possible commands.
smb: \> ls
  .                                   D        0  Sun Oct 15 16:07:50 2023
  ..                                  D        0  Wed Oct 18 10:02:25 2023
  frigate.db-shm                      N    32768  Tue Oct 17 21:21:04 2023
  recordings                          D        0  Tue Oct 17 06:16:53 2023
  clips                               D        0  Tue Oct 17 21:15:51 2023
  frigate.db-wal                      N  4165352  Tue Oct 17 21:21:04 2023
  frigate.db                          N  1093632  Tue Oct 17 12:12:39 2023
```
I have verified that the samba share is running the latest version of samba, extract from smb.conf
```
client max protocol = SMB3
   client min protocol = SMB2_10
   server max protocol = SMB3
   server min protocol = SMB2_10
```

I have tried 2.0 / 1.0 modes too, it results in the same error

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

5.10.0-25-amd64

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Click on Add Network Storage under HA->Settings->System->Storage
Fill up all values as shown in the image below and click connect

[image]
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
23-10-18 18:19:59 INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: frigate
23-10-18 18:19:59 ERROR (MainThread) [supervisor.mounts.mount] Mounting frigate did not succeed. Check host logs for errors from mount or systemd unit usr-share-hassio-mounts-frigate.mount for details.
```

### Anything in the Host logs that might be useful for us?

```txt
$  systemd unit usr-share-hassio-mounts-frigate.mount
Excess arguments.
```

### System information

## System Information

version | core-2023.7.3
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 5.10.0-25-amd64
arch | x86_64
timezone | Asia/Calcutta
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1312
Downloaded Repositories | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.5.1
docker_version | 24.0.5
disk_total | 158.4 GB
disk_used | 61.8 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2023.7.0), Home Assistant Google Drive Backup (0.105.2), MariaDB (2.6.1), Mosquitto broker (6.2.1), Node-RED (14.4.2), Samba share (10.0.2), Uptime Kuma (0.8.0), Cloudflared (4.2.1), File editor (5.6.0), Studio Code Server (5.9.0), Glances (0.19.4), SQLite Web (3.10.0), phpMyAdmin (0.8.9)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 18
views | 28
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 26 July 2023 at 3:54 pm
-- | --
current_recorder_run | 15 October 2023 at 4:26 pm
estimated_db_size | 4044.45 MiB
database_engine | mysql
database_version | 10.6.12

</details>

agners, 2023-10-18:
> Anything in the Host logs that might be useful for us

The host logs are at the same location under Settings > System > Logs and then choose Host on the top right.

You can also get the logs specifically for that unit. From the console, you can use the following command:

```
ha host logs --identifier usr-share-hassio-mounts-frigate.mount
```

closed 2024-01-24 by github-actions[bot]

sairon, 2024-01-24:
@vks007 If so, please provide us with the detailed logs as instructed above, without that we can't move any further.

vks007, 2024-01-24:
Hi @sairon  , sorry there are no logs relating to this in the host file. I put up the logs from the supervisor and systemd logs, that's all there is.