home-assistant/operating-system #2841: No wifi driver for RTL8821CE

bartlebi, 2023-10-18:
### Describe the issue you are experiencing

Home Assistant OS appears to be missing wifi support for on-board wireless card RTL8821CE.
I am using Hassos in proxmox VA.
I have thrown a PCI device into the virtual machine, but HA does not see any wifi device. 
I think there is a driver missing for this chip. What can I check or do on my end to solve the problem or may you just include driver in image?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

	Home Assistant OS 11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Just boot

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2023.10.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | x86_64
timezone | Europe/Moscow
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4873
Installed Version | 1.33.0
Stage | running
Available Repositories | 1313
Downloaded Repositories | 18

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 30.8 GB
disk_used | 8.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.2), Terminal & SSH (9.7.1), Mosquitto broker (6.3.1), go2rtc (1.8.1), RPC Shutdown (2.4), SQLite Web (3.2.1), Zigbee2MQTT (1.33.1-1), Node-RED (14.6.2), MariaDB (2.6.1), File editor (5.6.0), go2rtc master hardware (master-hardware)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 8
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 16 октября 2023 г. в 21:20
-- | --
current_recorder_run | 18 октября 2023 г. в 17:58
estimated_db_size | 142.72 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Xiaomi Gateway 3</summary>

version | 3.3.2 (4efeaee)
-- | --

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.13
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 0
total_devices | 0

</details>

### Additional information

rtl8821ce 802.11ac pcie wireless network adapter

sairon, 2023-10-18:
Coincidentally I just confirmed this card working today on a bare-bones install - there are some issues discussed in https://github.com/home-assistant/operating-system/issues/2722 but I haven't been able to reproduce them either. Maybe there's some problem with the passthrough, can you check output of `lspci` and `dmesg` in the guest OS?

bartlebi, 2023-10-18:
As far as I could understand in the output lspci shows that it finds the device with codes 10ec:c821 at the same time in dmesg I see only information about connecting bluetooth module via USB with a similar chip but different VID/PID: 0bda:c821.

bartlebi, 2023-10-18:
Read carefully and I see that it does find the pci device, but doesn't seem to be trying to use it.
I'll attach screen (cannot copy output).
[image]

bartlebi, 2023-10-19:
I tried removing the pci device and rescan and here is all I see in dmesg afterwards
[image]

sairon, 2023-11-16:
I went back to look into this issue and sorry for the confusion earlier - the driver is actually enabled only for the `generic_x86_64` target, not for `ova`. I will try enabling it there too and test if it works then.

sairon, 2023-11-20:
Quick update here - I tried enabling the driver and tested it in Proxmox on my mini PC with RTL8821CE card, yet the WiFi driver fails to probe the device with a null pointer dereference error :( I might look into it a bit more later but currently there's no easy fix for you.

Out of curiosity, why do you prefer passthrough of the device to the guest OS, rather than setting up the networking on host (and potentially reusing it for other guests)? For Bluetooth it makes perfect sense, but I am curious what is the use case of having dedicated card for a single guest.