home-assistant/operating-system #2843: Update 10.5 to 11.0 - can't boot, no internet, no HDMI, nothing

maricn, 2023-10-19:
### Describe the issue you are experiencing

Just ran the system update and bricked the device. Tried attaching HDMI to both outputs and turning on the hdmi safe mode in `config.txt`, but i don't get login console not even after 20 minutes of waiting.

I can mount some partitions of the sdcard to provide additional information upon request.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Running 10.5 HAOS
2. Update to 11 HAOS
3. Fails to boot
...

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

aserno, 2023-10-19:
Same probleem on my rpi 4.
Running my data on a external HDD.

i had to restore my backup but then the database corrupted, so i lost a year of (energy) data

a1j, 2023-10-21:
I wonder if [this issue](https://github.com/home-assistant/operating-system/issues/2822) related not to only odroid but to any setup with dedicated data device.

agners, 2023-10-24:
Hm, this is very curious. Between HAOS 10.5 and 11.0 there is really no update on the Raspberry Pi side (we use the same Linux kernel, the same Raspberry Pi firmware in both releases).

Needless to say we tested the update on Raspberry Pi 4 with and without external data disk. Also there are thousands of updated instances already reported (according to [analytics.home-assistant.io](https://analytics.home-assistant.io/)).

This sounds like some corruption on disk or similar. Can you check the files on the first boot partition? Since you have no HDMI output, I'd expect corrupted firmware files (e.g. start4.elf). My HAOS 11.0 rpi4-64 installations checksums are:

```
$ sha256sum /mnt/boot/*
a77b7c138cf2e0abcac3122010b9b75a5993ff83a0c27cb2c09b01209112c87f  /mnt/boot/bcm2711-rpi-4-b.dtb
6a64391325b3b1ea937db4c9967bb22d40633d5b1613320064da7ded51f02e2a  /mnt/boot/bcm2711-rpi-400.dtb
4144625bad01c11a5930295c3130f1c5038b02d26f99ea530deee712988050d3  /mnt/boot/bcm2711-rpi-cm4.dtb
9d336be87ed8566bbe34128a3ad1bb48e857f6638280f19fde2f63413489de2d  /mnt/boot/boot.scr
fc114d04ae0ac68cc228ab773b5d4823cd3a3790deb371eaa04e54d6c1e16e62  /mnt/boot/cmdline.txt
5c36df3771dd5484001b52e631e699877a60394962981abc0fd4757d4f8c86bc  /mnt/boot/config.txt
5df7a849d3d27a9cdc33bb7334c4537d6a6e657f18643f0f11a998ab10cc531f  /mnt/boot/fixup4.dat
6c2b3553a16a336d2a8de4a9caddd8ef2d7925e92b5712facdd157dfa5d6a88e  /mnt/boot/fixup4cd.dat
575fad4362f3e4ebc2977ee74d8721f43cdcde3a1df0e8c953b682207d95708f  /mnt/boot/fixup4x.dat
sha256sum: can't read '/mnt/boot/overlays': Is a directory
1b5c031dd5b4daea65949b3cee55bf94a699d449b42a560c29968df5b45d6313  /mnt/boot/start4.elf
8d556607423f3da30f3e63d1b6b246df682d10882a62f5299c924d4df128524c  /mnt/boot/start4cd.elf
b6275e517ff74a1b05979efc400976acfaa5b7434e50df1807a442af6d1c89c1  /mnt/boot/start4x.elf
dfb53cd9956f8ef4d9ea3fb1cd27b610af39824e3b53f4b1721771bbff2028a5  /mnt/boot/u-boot.bin
```

agners, 2023-10-24:
> Same probleem on my rpi 4. Running my data on a external HDD.
>
> i had to restore my backup but then the database corrupted, so i lost a year of (energy) data

@aserno what database have you been using? This should really not be the case when using SQLite and the built-in backup functionality!

aserno, 2023-10-25:
> > Same probleem on my rpi 4. Running my data on a external HDD.
> > i had to restore my backup but then the database corrupted, so i lost a year of (energy) data
> 
> @aserno what database have you been using? This should really not be the case when using SQLite and the built-in backup functionality!

I don't think it is related to HAOS 11.

After the upgrade not working I reinstalled HA and restored the fullback I made with the HA backup tools.
But then i noticed DB "home-assistant_v2" was new and the one from the backup was renamed and contains "corrupted"

aserno, 2023-10-26:
> I don't think it is related to HAOS 11.
>
> After the upgrade not working I reinstalled HA and restored the fullback I made with the HA backup tools. But then i noticed DB "home-assistant_v2" was new [...]

Well, I tried it again, now monitoring through HDMI en the Observer (in case to see where is get stuck in the process) but the upgrade went fast and smooth this time. no idea why i didn't work the first time. 

Maybe the fresh install of HA? maybe sometimes it's like windows ;)

maricn, 2023-11-27:
Tried @agners suggestion, only different shas are:
> ```
> $ sha256sum /mnt/boot/*
> fc114d04ae0ac68cc228ab773b5d4823cd3a3790deb371eaa04e54d6c1e16e62  /mnt/boot/cmdline.txt
> 5c36df3771dd5484001b52e631e699877a60394962981abc0fd4757d4f8c86bc  /mnt/boot/config.txt;
> ```

Which is expected, afaik.

I reformatted the card with fresh installation and it failed again. Then I realized I had a GPIO pin connected preventing the rpi from booting 🤦 . Not 100% sure if it was there all the time, but very likely that was the cause of my issue 😭

closed 2023-11-27 by maricn