home-assistant/operating-system #2850: HAOS 10.5 to 11.0 update failing on HA Yellow

hwobu, 2023-10-21:
### Describe the issue you are experiencing

While performing the HAOS upgrade from 10.5 to 11.0, my home assistant yellow acts like normal, but after the booting process has completed, it acts like the upgrade never took place. I've attempted this process 5 or 6 times. I also read the release announcement' breaking changes and the wired Ethernet did change from 'eth0' to 'end0' as expected. The logs, however, were blank, except for the most recent attempt, which still wasn't much.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Select Home Assistant Operating System Update (11.0)
2. Read Release Announcement
3. Return to Home Assistant
4. Click Install
5. Wait for Connection Lost
6. Exit completely out of Home Assistant App
7. Wait about 5-7 minutes (not timed for accuracy)
8. Open the Home Assistant App again and note the remaining update notification for HAOS 11.0.
9. Check Logs
10. Retried steps 1-9 with IPv6 disabled completely in both home assistant and on the connected network, with a reboot after making those changes. Retried again with everything re-enabled.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-10-20 23:39:37 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/yellow-homeassistant with version 2023.10.3
23-10-20 23:39:37 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 10.5 / BootSlot A
23-10-20 23:39:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
23-10-20 23:39:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
23-10-20 23:39:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
23-10-20 23:39:39 INFO (MainThread) [supervisor.store] Loading add-ons from store: 72 all - 72 new - 0 remove
23-10-20 23:39:39 INFO (MainThread) [supervisor.addons] Found 19 installed add-ons
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/zwave-js-ui/aarch64 with version 2.2.3
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-matter-server with version 4.9.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/grafana/aarch64 with version 9.1.1
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/influxdb/aarch64 with version 4.8.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-zwave_js with version 0.2.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2023.10.1
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-whisper with version 1.0.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.3.1
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/aarch64 with version 14.6.3
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-silabs-multiprotocol with version 2.3.2
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-piper with version 1.4.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-letsencrypt with version 4.12.9
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/home-panel/aarch64 with version 2.7.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-almond with version 3.0.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.7.1
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-otbr with version 2.3.2
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-samba with version 10.0.2
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/aarch64 with version 15.1.0
23-10-20 23:39:39 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/bitwarden/aarch64 with version 0.20.1
23-10-20 23:39:39 INFO (MainThread) [supervisor.backups.manager] Found 91 backup files
23-10-20 23:39:40 INFO (MainThread) [supervisor.discovery] Loaded 7 messages
23-10-20 23:39:40 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
23-10-20 23:39:40 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
23-10-20 23:39:40 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-10-20 23:39:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
23-10-20 23:39:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-10-20 23:39:40 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup
23-10-20 23:39:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
23-10-20 23:39:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-10-20 23:39:40 INFO (MainThread) [__main__] Running Supervisor
[... 228 lines cut ...]
estimated_db_size | 7.43 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

The only unusual logs are for Home Assistant Core; though still not very helpful. And only appeared on the most recent attempt.

Logger: homeassistant.setup
Source: runner.py:186
First occurred: October 20, 2023 at 23:41:16 (1 occurrences)
Last logged: October 20, 2023 at 23:41:16

Setup of zone is taking over 10 seconds.

Just noticed, there also seems to be a bit of a discrepancy for the upgrade itself. Here, HA indicates that the installed version is 10.5, but the System Information indicated version is core-2023.10.3.

Home Assistant Operating System
Installed version
10.5
Latest version
11.0
[Read release announcement](https://github.com/home-assistant/operating-system/releases/tag/11.0)

hwobu, 2023-10-23:
It seems that once the core-2023.10.5 update was available and then installed, then the HAOS 11.0 update would install.

Would recommend confirmation from the developers that this was causation; not simply correlation.

If this was the cause, perhaps an update to the upgrade process would be beneficial. Such a new feature might include a dependency check on updates, visible prior to attempting the update, where unmet dependencies are listed on the update details and if that dependency needs to be installed first and won't/can't be installed as part of the specific update, then an explicit override would be needed to attempt anyway. Otherwise, point to the package needing installed or upgraded as a prerequisite.

guandalf, 2023-10-24:
But maybe I am affected by https://github.com/home-assistant/operating-system/commit/aed1bb807c421c5c37e6ac57a536bc6c77d76574

agners, 2023-10-25:
@guandalf the ODROID-M1 issue is handled in #2822

@hwobu @aqus555 that is an issue we've seen previously, unfortunately only some devices seem affected.

Do you use an NVMe on your device?

After an update attempt, can you share the logs of the previous boot? you can get the logs of the previous boot using the following command in the Web terminal:

```
ha host logs -b -1 -n 10000 > /config/last-boot.log
```

Then share the last-boot.log on this thread.

There will be a 11.1 release soon,  but I don't expect this to change.

> Just noticed, there also seems to be a bit of a discrepancy for the upgrade itself. Here, HA indicates that the installed version is 10.5, but the System Information indicated version is core-2023.10.3.

Note that different components of the Home Assistant stack have different version numbers. Home Assistant Core uses calendar based versioning, the latest release is 2023.10.5. The Operating System uses classic version numbers, the latest release is 11.0 (before it was 10.5).

hwobu, 2023-10-27:
It took me a bit before I could get enough time to sit down and answer the questions posed.

First, yes, I'm running on an NVME SSD, though the CM4 itself is the CM4108032 model (8Gig with WiFi/BT and 32G emmc).

`ha host logs -b -1 -n 10000 > /config/last-boot.log`

`{"result": "error", "message": "Logs only contain 0 boots"}#`

GregEdge, 2023-10-27:
Similar issues. Upgraded core 9.3 -> core 10 -> core 10.1 -> 10.5. Tried 11 update but that resulted in being at 9.5 and apparmour errors.
Upgraded to core 10.3 but it reports I'm on 9.5 and subsequent attempts to upgrade do nothing.  Running on Raspberry Pi, I'll wait for resolution

GregEdge, 2023-11-10:
So tried the OS upgrade again and sure enough it failed and reverted to 9.5.
From reading feedback to others, it seems the only way this could happen is if the upgrade installs into the current slot (my terminology correct ?) and when it fails 3x it reverts to the previous slot which in my case has 9.5 installed.
Certainly fits the symptoms others have reported.

agners, 2023-12-05:
@GregEdge @JaCoLJcL after a failed upgrade, can you try to run the commands mentioned in https://github.com/home-assistant/operating-system/issues/2850#issuecomment-1778910888?

The most likely case here is that the boot partition somehow corrupted, and the system can't update the new boot state properly. That causes the system to revert to the previous version after three boot attempts. Probably the easiest way to resolve this is to take a full backup and start with a new installation.

GregEdge, 2024-03-05:
Resolved my issue but believe it is an unforeseen condition. 
I run on an external usb. I had a corruption due to a faulty power lead. I put an sd card in my Pi 4 to recover the ssd.
If the ssd and the sd card are both connected,  although it boots to the ssd, if you try to upgrade the OS, it appears to try to upgrade the sd os and then when the ssd is loaded it shows the previous OS level.
Remove the sd and the ssd gets updated correctly. 
Unlikely others will hit this but it is reproducible.

closed 2024-06-12 by github-actions[bot]