home-assistant/operating-system #2851: ssl folder not restored

plu29um, 2023-10-21:
### Describe the issue you are experiencing

Full restore from local backup leaves the ssl folder empty

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.  Setup SSL using NGINX Home Assistant SSL proxy. Port: 443
2.  Have other instance of ACME copy wildcard certificates to ssl folder
3.  Create backup
4.  Restore backup

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2023.10.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | x86_64
timezone | Europe/Copenhagen
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1310
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 468.7 GB
disk_used | 11.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.3.1), Zigbee2MQTT (1.33.1-1), Home Assistant Google Drive Backup (0.111.1), Studio Code Server (5.13.0), RPC Shutdown (2.4), Samba share (10.0.2), Terminal & SSH (9.7.1), NGINX Home Assistant SSL proxy (3.5.0), Node-RED (14.6.3), Network UPS Tools (0.12.2), File editor (5.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 3
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 12, 2023 at 16:34
-- | --
current_recorder_run | October 21, 2023 at 21:16
estimated_db_size | 142.68 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

Before copying the certicates to ssl folder, I was only able to access the GUI by original URL: http://homeassistant.mydomain.tld:8123. The URL set in NGINX and local DNS resolver did not work.

Config of NGINX:
domain: ha.MYDOMAIN.TLD
hsts: max-age=31536000; includeSubDomains
certfile: fullchain.pem
keyfile: privkey.pem
cloudflare: true
customize:
  active: true
  default: nginx_proxy_default*.conf
  servers: nginx_proxy/*.conf

plu29um, 2023-10-21:
Just noticed that the backup that I restored from is 3 MB smaller than the previous and subsequent backup. I always do full backups.

closed 2023-10-24 by github-actions[bot]

sairon, 2023-10-24:
This should be filed as a Supervisor issue if there's indeed some problem. You can first also check the backup was created correctly - just download and open it (it's a standard `tar` archive) and check it contains file `ssl.tar.gz`, which should eventually contain the certificates to be restored.