home-assistant/operating-system #2852: VLAN interface lost after upgrading to 11.0

tomasbedrich, 2023-10-22:
### Describe the issue you are experiencing

In my pre-11.0 setup there was a secondary VLAN-enabled adapter (based on `eth0`) to isolate the network traffic. After upgrading to 11.0 the adapter disappeared. After re-adding it manually using `nmcli con add type vlan con-name end0@vlan2 dev end0 id 2 ip4 192.168.2.20/23 gw4 192.168.2.1`, everything works again.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. have second network adapter with VLAN enabled
2. upgrade to HassOS 11.0
3. the secondary adapter is lost
4. fix the

### Anything in the Supervisor logs that might be useful for us?

```txt
likely nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.2597] manager: (end0.2): new VLAN device (/org/freedesktop/NetworkManager/Devices/38)
Oct 22 20:13:56 homeassistant kernel: 8021q: 802.1Q VLAN Support v1.8
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.2801] device (end0.2): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.2941] audit: op="connection-add" uuid="b3e062e9-ddfe-48d4-8b56-3e595bdc8fa8" name="end0@vlan2" pid=23539 uid=0 result="success"
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.2965] device (end0.2): carrier: link connected
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.2990] device (end0.2): state change: unavailable -> disconnected (reason 'carrier-changed', sys-iface-state: 'managed')
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.3014] policy: auto-activating connection 'end0@vlan2' (b3e062e9-ddfe-48d4-8b56-3e595bdc8fa8)
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.3031] device (end0.2): Activation: starting connection 'end0@vlan2' (b3e062e9-ddfe-48d4-8b56-3e595bdc8fa8)
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.3035] device (end0.2): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.3051] device (end0.2): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.3063] device (end0.2): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Oct 22 20:13:56 homeassistant systemd-resolved[434]: end0.2: Bus client set default route setting: no
Oct 22 20:13:56 homeassistant systemd-resolved[434]: end0.2: Bus client set MulticastDNS setting: yes
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.3308] device (end0.2): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
Oct 22 20:13:56 homeassistant systemd[1]: Starting Network Manager Script Dispatcher Service...
Oct 22 20:13:56 homeassistant systemd[1]: Started Network Manager Script Dispatcher Service.
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.4162] device (end0.2): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.4172] device (end0.2): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
Oct 22 20:13:56 homeassistant NetworkManager[443]: <info>  [1698005636.4192] device (end0.2): Activation: successful, device activated.
Oct 22 20:13:56 homeassistant systemd-resolved[434]: Using degraded feature set UDP instead of UDP+EDNS0 for DNS server 192.168.0.1.
Oct 22 20:14:06 homeassistant systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Oct 22 20:14:23 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-4fdee32d3114cf9e590d4bb416bd7ad3f24373d034191134820bc5547151eaa9-runc.mR0iAq.mount: Deactivated successfully.
Oct 22 20:14:40 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b429a79c0b977ae23bcc22c5ed48fdcb3255f05cd1707d1a15a76221895a5d88-runc.HPLFgF.mount: Deactivated successfully.
Oct 22 20:15:25 homeassistant NetworkManager[443]: <info>  [1698005725.7192] audit: op="connection-update" uuid="b3e062e9-ddfe-48d4-8b56-3e595bdc8fa8" name="end0@vlan2" args="connection.timestamp" pid=23758 uid=0 result="success"
```

agners, 2023-10-24:
We realized that the NetworkManager profiles for VLAN did not get applied correctly after the interface rename. There is a Supervisor fix, which just made it to the beta version of Supervisor (see https://github.com/home-assistant/supervisor/pull/4633).

However, this fix will only work if (the fixed version of) Supervisor runs with correctly setup VLAN. So essentially, for now you'd have to downgrade to HAOS 10.5 (using `ha os update --version 10.5`), then wait until the next Supervisor is deployed (newer then 2023.10.0). Once your Supervisor has that new version, it should be safe to update to 11.0 again.

Manually creating the VLAN profile works as a work around, however I am not exactly sure how Supervisor will handle that profile. Do you see the configuration using `ha network info`?

tomasbedrich, 2023-10-24:
> you'd have to downgrade to HAOS 10.5

Thank you, I think I can stay on 11.0 as there is currently no real issue on my side. I created this ticket only for a record - to report the misbehaviour and possibly let you decide if it's worth fixing / noting in the changelog.

>  I am not exactly sure how Supervisor will handle that profile

So far it works flawlessly:

```
➜  ~ ha network info
docker:
  address: 172.30.32.0/23
  dns: 172.30.32.3
  gateway: 172.30.32.1
  interface: hassio
host_internet: true
interfaces:
- connected: false
  enabled: false
  interface: wlan0
  ipv4:
    address: []
    gateway: null
    method: disabled
    nameservers: []
    ready: false
  ipv6:
    address: []
    gateway: null
    method: disabled
    nameservers: []
    ready: false
  mac: DC:A6:32:B5:F8:9E
  primary: false
  type: wireless
  vlan: null
  wifi: null
- connected: true
  enabled: true
  interface: end0
  ipv4:
    address:
    - 192.168.0.20/23
    gateway: 192.168.0.1
    method: static
    nameservers:
    - 192.168.0.1
    ready: true
  ipv6:
    address:
    - fe80::a4c8:1e76:3c41:8a1/64
    gateway: null
    method: disabled
    nameservers: []
    ready: true
  mac: DC:A6:32:B5:F8:9D
  primary: true
  type: ethernet
  vlan: null
  wifi: null
- connected: true
  enabled: true
  interface: end0.2
  ipv4:
    address:
    - 192.168.2.20/23
    gateway: 192.168.2.1
    method: static
    nameservers: []
    ready: true
  ipv6:
    address:
    - fe80::75fa:73ba:27aa:f4df/64
    gateway: null
    method: auto
    nameservers: []
    ready: false
  mac: DC:A6:32:B5:F8:9D
  primary: false
  type: vlan
  vlan:
    id: 2
    parent: end0
  wifi: null
supervisor_internet: true
```

[image]

closed 2023-10-24 by agners