home-assistant/operating-system #2857: Disk space mysteriously being consumed

UncleLeoTheDad, 2023-10-23:
### Describe the issue you are experiencing

I've detailed my experience here as well:
https://community.home-assistant.io/t/where-is-my-disk-space-going/622795/4

I know there are lots of posts about this and I apologize for adding another one, but I’m stumped. I have 64gb of space allocated to my VM:
[image]

When I look at what I am using I see that more than half is gone:
[image]

So then I use ncdu (fancy du) from / to drill into where the space is going and I can only find about 6gigs being used. So where are the other 33 gigs? Is there some secret space between the VM and the HA OS that I am unable to see?

I tried SSHing in and did both a "ha supervisor repair" and rebooted, but didn't notice a significant difference.  I considered doing some docker pruning based on community suggestions, but figured you'd rather have this fixed via an HA wrapped command or process.

How did I discover this?  I recently had a problem where I ran out of space due to a log file going nuts and that prompted me to investigate file space.  I then noticed this missing space problem and saw that it's reported quite a bit on the community without a clear solution.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.56

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

I know there are lots of posts about this and I apologize for adding another one, but I’m stumped. I have 64gb of space allocated to my VM:
[image]

When I look at what I am using I see that more than half is gone:
[image]

So then I use ncdu (fancy du) from / to drill into where the space is going and I can only find about 6gigs being used. So where are the other 33 gigs? Is there some secret space between the VM and the HA OS that I am unable to see?

I tried SSHing in and did both a "ha supervisor repair" and rebooted, but didn't notice a significant difference.  I considered doing some docker pruning based on community suggestions, but figured you'd rather have this fixed via an HA wrapped command or process.

### Anything in the Supervisor logs that might be useful for us?

```txt
No errors or warnings.  I did try deleting all logs in /config folder
```

### Anything in the Host logs that might be useful for us?

```txt
No errors or warnings.  I did try deleting all logs in /config folder
```

### System information

## System Information

version | core-2023.10.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1325
Downloaded Repositories | 51

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | December 29, 2023 at 7:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-2.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 62.3 GB
disk_used | 50.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.2), MariaDB (2.6.1), Home Assistant Google Drive Backup (0.111.1), Z-Wave JS UI (2.2.3), Node-RED (14.6.3), Glances (0.19.4), Grafana (9.1.1), AirCast (3.6.1), Studio Code Server (5.13.0), File editor (5.6.0), Zigbee2MQTT (1.33.1-1), Mosquitto broker (6.3.1), AirSonos (3.6.1), Advanced SSH & Web Terminal (15.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 12
-- | --
resources | 27
views | 21
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 1, 2023 at 5:13 PM
-- | --
current_recorder_run | October 20, 2023 at 3:51 PM
estimated_db_size | 17382.89 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Victor Smart-Kill</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2023-10-26:
Could it be that the disk is indeed full? Maybe this operation needs to copy the disk image, which would mean it needs as much free space as the size of the disk image itself?

I am not really familiar with VMware, so I don't know really.

UncleLeoTheDad, 2023-10-26:
I freed up space as you suggested and that got me past the error, so thank you!  Unfortunately, after I ran this and rebooted everything, I didn't see any disk space freed up within the VM.  :(

UncleLeoTheDad, 2023-10-29:
Okay, figured something out that might be obvious to others, but wasn’t to me. I have continued to poke around posts and articles to troubleshoot this problem and came across this bit:
https://community.home-assistant.io/t/where-has-all-the-disk-space-gone/357971/8?u=uncleleothedad

So on the host machine, in the VMWare window, I typed “login” to the command prompt and then used my HA credentials to log in. From there, when I run du, etc, I see the space. It turns out mine was in various places that I couldn’t see without taking this additional login step. It would be nice if the documentation was updated somehow to make this obvious, but I see light at the end of the tunnel now:

[image]

[image]

[image]

[image]

agners, 2023-10-31:
Yeah finding used disk space is not straight forward. It would be a nice addition to Home Assitant to have some more insight in the frontend. Something for the [Forum's feature request section](https://community.home-assistant.io/c/feature-requests/13).

Thanks for the update, I am closing the issue here then.

closed 2023-10-31 by agners