home-assistant/operating-system #2869: HAOS >=10.3 randomly loses connection to Sonoff Zigbee 3.0 USB Dongle Plus

robtpaton, 2023-10-26:
### Describe the issue you are experiencing

After upgrading to Home Assistant OS 10.3 or above, I randomly lose connection to my Sonoff Zigbee 3.0 USB Dongle Plus and only a hardware reboot fixes the issue (until it happens again). The dongle has been upgraded to the latest firmware (CC1352P2_CC2652P_launchpad_coordinator_20230507).

When it happens, the dongle is not shown under Hardware.

Downgrading back to OS 10.2 resolves the issue.

I originally added to the issue raised here https://github.com/home-assistant/core/issues/94992#issuecomment-1781888345 but this seems like an OS issue rather than a Core issue.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start with Home Assistant OS 10.2
2. Observe no issues with connection to Sonoff Zigbee 3.0 USB Dongle Plus
3. Upgrade to Home Assistant OS 10.3 or above (have tried with every version since!)
4. Wait anywhere from 1 hour to 24 hours (seems to be no pattern as to exactly how long)
5. Obverse that Home Assistant has lost connection to Sonoff Zigbee 3.0 USB Dongle Plus so no Zigbee commands can be sent or received
7. Perform a hardware reboot to get the dongle working again
8. Go to step 4

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2023.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.29
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1320
Downloaded Repositories | 21

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.2
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 219.4 GB
disk_used | 58.3 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Autossh (0.4.0), Mosquitto broker (6.3.1), Studio Code Server (5.13.0), Dnsmasq (1.6.0), Advanced SSH & Web Terminal (15.1.0), Apache2 Minimal (2.7.6), ESPHome (2023.10.3), go2rtc (1.8.1), Let's Encrypt (4.12.9), Whisper (1.0.0), Piper (1.4.0), Stock Check (1.0.0), openWakeWord (1.8.1), Assist Microphone (2.2.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 9
views | 24
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 16 October 2023 at 15:03
-- | --
current_recorder_run | 26 October 2023 at 21:01
estimated_db_size | 240.15 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

rickgardner077, 2023-10-30:
Similar problem here with HAOS on RPi4B and SONOFF Zigbee 3.0 USB Dongle Plus ZigBee 3.x.0 TI CC2652P + CP2102N Coordinator

Issue started since updating HAOS to 11.1 (Home Assistant 2023.10.5; Supervisor 2023.10.1; Operating System 11.1). Home Assistant loses connections to all devices connected via the SONOFF dongle and Z2M add-on. System then unresponsive, reboot and shutdown from GUI and CLI not possible. Only way to resolve is hard power off after which Z2M and all Zigbee devices show up again until the problem arises again

agners, 2023-10-31:
> When it happens, the dongle is not shown under Hardware.

That sounds as if the device disappears from the USB bus completely.

Can you check the kernel logs after it happens next time using:

```
ha host logs -b 0 -n 10000 -t kernel > /config/last-boot.log
```

There should be at least a USB disconnect event there when it happened.

@rickgardner077 since your issue started with HAOS 11.1, is on a Raspberry Pi and leads to a complete unresponsive system this seems a different issue. Can you please open a new issue along with all information? Thank you.

robtpaton, 2023-11-02:
Hi @agners, the disconnect has happened again. I found this log in Home Assistant Core logs:

```
Logger: zigpy_znp.zigbee.application
Source: /usr/local/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py:778
First occurred: 14:23:35 (962 occurrences)
Last logged: 15:43:51

Failed to reconnect
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_fa23a7c87d29ec11ae146f7840c9ce8d-if00-port0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py", line 772, in _reconnect
    await self.connect()
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py", line 115, in connect
    await znp.connect()
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 720, in connect
    self._uart = await uart.connect(self._config[conf.CONF_DEVICE], self)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/uart.py", line 170, in connect
    _, protocol = await zigpy.serial.create_serial_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/serial.py", line 48, in create_serial_connection
    transport, protocol = await pyserial_asyncio.create_serial_connection(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/serial_asyncio_fast/__init__.py", line 508, in create_serial_connection
    serial_instance = await loop.run_in_executor(None, callback)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_fa23a7c87d29ec11ae146f7840c9ce8d-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_fa23a7c87d29ec11ae146f7840c9ce8d-if00-port0'

```

and after running the command you provided, there are definitely disconnects at 14:23:

```
Nov 02 14:20:18 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
Nov 02 14:20:18 homeassistant kernel: Bluetooth: hci0: command 0x200c tx timeout
Nov 02 14:20:20 homeassistant kernel: Bluetooth: hci0: command 0x200c tx timeout
Nov 02 14:20:20 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
Nov 02 14:20:20 homeassistant kernel: Bluetooth: hci0: failed to restart LE scan: status -110
Nov 02 14:20:23 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
Nov 02 14:20:23 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -110
Nov 02 14:20:23 homeassistant kernel: Bluetooth: hci0: command 0x200c tx timeout
Nov 02 14:20:25 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
Nov 02 14:20:25 homeassistant kernel: Bluetooth: hci0: command 0x200c tx timeout
Nov 02 14:20:25 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -110
Nov 02 14:20:25 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -110
Nov 02 14:20:27 homeassistant kernel: Bluetooth: hci0: command 0x200c tx timeout
Nov 02 14:20:27 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
Nov 02 14:20:27 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -110
Nov 02 14:20:27 homeassistant kernel: Bluetooth: hci0: No way to reset. Ignoring and continuing
Nov 02 14:20:27 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -110
Nov 02 14:20:37 homeassistant kernel: Bluetooth: hci0: command 0x200c tx timeout
Nov 02 14:20:37 homeassistant kernel: Bluetooth: hci0: No way to reset. Ignoring and continuing
Nov 02 14:20:37 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
Nov 02 14:20:37 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -110
Nov 02 14:20:37 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -110
[... 87 lines cut ...]
Nov 02 14:23:34 homeassistant kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Nov 02 14:23:34 homeassistant kernel: usb 2-1: USB disconnect, device number 2
Nov 02 14:23:35 homeassistant kernel: usb 2-3: USB disconnect, device number 3
Nov 02 14:23:35 homeassistant kernel: cp210x ttyUSB0: failed set request 0x7 status: -19
Nov 02 14:23:35 homeassistant kernel: cp210x ttyUSB0: failed set request 0x12 status: -19
Nov 02 14:23:35 homeassistant kernel: cp210x ttyUSB0: failed set request 0x0 status: -19
Nov 02 14:23:35 homeassistant kernel: cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[... cut ...] 6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F37343835396639666631633931653837363432383534333465
Nov 02 15:48:21 homeassistant kernel: audit: type=1334 audit(1698940101.726:747): prog-id=255 op=UNLOAD
Nov 02 15:48:21 homeassistant kernel: audit: type=1334 audit(1698940101.726:748): prog-id=254 op=UNLOAD
Nov 02 15:48:21 homeassistant kernel: audit: type=1334 audit(1698940101.726:749): prog-id=256 op=LOAD

```

Does this point to a particular issue or anything else I should try? TIA!

robtpaton, 2023-12-17:
I've updated the bug report as originally I said the issue only happened with 10.3 but it actually occurs with other versions as well.

I've also just found this issue which could be related. 

https://github.com/home-assistant/supervisor/issues/3740

Can anyone help?

Thanks

robtpaton, 2023-12-17:
@mike10101011 I see you raised a similar issue, did you ever solve it? TIA

Sebazzz, 2024-01-08:
Is this the same issue as #2756?

robtpaton, 2024-01-08:
Since changing USB power settings in the BIOS of my NUC (suggestion from Discord), I have not had a single occurrence of this (so it probably wasn't an issue with Home Assistant itself). Closing this issue for now!

agners, 2024-01-08:
@robtpaton you used the native HAOS installation on your NUC correct?

Can you be more specific what the power option was?

robtpaton, 2024-01-15:
@agners that's correct HAOS installed on the NUC using the x86_64 image.

I haven't had a chance to connect it back up to a screen to review the settings but I tweaked anything I could find related to powering USB devices to favour less power-saving, and generally disabled any advanced USB features (I think I even set some sort of USB Handling setting to Legacy or Basic, something like that).

So far so good, everything rock solid since then. I will reply with more specifics when I can

Thanks

closed 2024-01-15 by robtpaton

rt18de, 2024-01-30:
I think I have the exactly same issue with Sonoff P dongle (HA on Raspberry Pi).  When the problem occurs, restarting HA will not resolve the issue. All devices are not available. Restarting Zigbee2MQTT add-on can resolve the problem and all devices will immediately work again. That's why I suspect it's Sonoff P drop.  I don't know however, where I can change the USB settings. 
Looking forward to hearing your experienes again

robtpaton, 2024-02-09:
Not sure if yours is the same issue or a different one @rt18de. The settings that have definitely helped for me were in my BIOS, so probably specific to the host hardware. But here are the screens I believe I changed:

[image]
[image]

I have to say I can't remember exactly what the settings were before I changed them, but I think I enabled USB Legacy and anything that sounded related to USB power.

Good luck!