home-assistant/operating-system #2870: Update from 11.0 to 11.1 does nothing

BebeMischa, 2023-10-26:
### Describe the issue you are experiencing

Got a message, new version is available. Click install, everything looks like it is installing. After the automatic reboot HA comes back alive with still OS 11.0 installed and a message about new version.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Try installing 11.1 by clicking on install
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
All green, no errors to be seen
```

### Anything in the Host logs that might be useful for us?

```txt
Nope
```

### System information

```
## System Information

version | core-2023.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1331
Downloaded Repositories | 32

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 20 november 2023 om 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-10.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.1
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 219.4 GB
disk_used | 43.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS (0.2.1), Mosquitto broker (6.3.1), Terminal & SSH (9.7.1), File editor (5.6.0), VLC (0.2.0), MariaDB (2.6.1), Studio Code Server (5.13.0), Samba Backup (5.2.0), TimescaleDB (3.0.2), DSMR Reader (1.10.2), Music Assistant BETA (2.0.0b73), Samba share (10.0.2), motionEye (0.19.1), Dutch gas prices (2023.10.15.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 21
views | 56
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 26 augustus 2023 om 21:33
-- | --
current_recorder_run | 26 oktober 2023 om 23:04
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Sonoff</summary>

version | 3.5.3 (a8c6d45)
-- | --
cloud_online | 27 / 27
local_online | 27 / 27

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>
```

aflecha87, 2023-10-26:
I've got the same exact issue. It eventually went through, was just very slow.

francescopeloi, 2023-10-27:
I've waited 20 mins and the app hasn't come up yet... should I be worried or this update takes long? Never took so long, and all other tens of previous updates never had a minimal issue.

francescopeloi, 2023-10-27:
After ~1h of waiting I decided to force a restart, and now it came up, and with the new version installed. First time I had this issue, not sure what went wrong as I can't find anything useful in the logs.

agners, 2023-10-27:
@BebeMischa After an update attempt, can you share the logs of the previous boot? you can get the logs of the previous boot using the following command in the Web terminal (the `-b -1` instructs to return the logs of the previous boot):

```
ha host logs -b -1 -n 10000 > /config/last-boot.log
```

@francescopeloi can you share the same log in your current situation (where the previous boot was the one which got stuck)?

francescopeloi, 2023-10-27:
Here it is. I ran the updated at ~8 CET, I guess the log timestamp are UTC, so the update starts at ~6am in the logs

```
Oct 27 05:31:53 homeassistant kernel: audit: type=1334 audit(1698384713.656:637): prog-id=262 op=UNLOAD
Oct 27 05:31:53 homeassistant kernel: audit: type=1334 audit(1698384713.656:638): prog-id=261 op=UNLOAD
Oct 27 05:31:53 homeassistant kernel: audit: type=1334 audit(1698384713.656:639): prog-id=260 op=UNLOAD
Oct 27 06:04:27 homeassistant rauc[455]: input bundle: /mnt/data/supervisor/tmp/hassos-11.1.raucb
Oct 27 06:04:30 homeassistant kernel: loop0: detected capacity change from 0 to 248570
Oct 27 06:04:30 homeassistant kernel: loop0: detected capacity change from 248570 to 248568
Oct 27 06:04:30 homeassistant kernel: loop1: detected capacity change from 0 to 65536
Oct 27 06:05:19 homeassistant systemd[1]: docker-0a62a12670a6263f87b27bf310bd0af82e90c5e8e99bd942760c18fe0390243d.scope: Deactivated successfully.
Oct 27 06:05:19 homeassistant systemd[1]: docker-0a62a12670a6263f87b27bf310bd0af82e90c5e8e99bd942760c18fe0390243d.scope: Consumed 4min 4.909s CPU time.
Oct 27 06:05:19 homeassistant dockerd[556]: time="2023-10-27T06:05:19.649963954Z" level=info msg="ignoring event" container=0a62a12670a6263f87b27bf310bd0af82e90c5e8e99bd942760c18fe0390243d module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Oct 27 06:05:19 homeassistant kernel: docker0: port 1(veth4497a78) entered disabled state
Oct 27 06:05:19 homeassistant kernel: vetha5a98ab: renamed from eth0
Oct 27 06:05:19 homeassistant NetworkManager[450]: <info>  [1698386719.7948] manager: (vetha5a98ab): new Veth device (/org/freedesktop/NetworkManager/Devices/27)
Oct 27 06:05:19 homeassistant kernel: docker0: port 1(veth4497a78) entered disabled state
Oct 27 06:05:19 homeassistant kernel: device veth4497a78 left promiscuous mode
Oct 27 06:05:19 homeassistant kernel: docker0: port 1(veth4497a78) entered disabled state
Oct 27 06:05:19 homeassistant kernel: audit: type=1700 audit(1698386719.821:640): dev=veth4497a78 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
Oct 27 06:05:19 homeassistant kernel: audit: type=1300 audit(1698386719.821:640): arch=c00000b7 syscall=206 success=yes exit=32 a0=c a1=4001591f20 a2=20 a3=0 items=0 ppid=1 pid=556 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj=unconfined key=(null)
Oct 27 06:05:19 homeassistant kernel: audit: type=1327 audit(1698386719.821:640): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D636F6E7461696E6572643D2F72756E2F636F6E7461696E6572642F636F6E7461696E6572642E736F636B
Oct 27 06:05:19 homeassistant kernel: hassio: port 2(vethf84b4fe) entered disabled state
Oct 27 06:05:19 homeassistant kernel: veth1a96227: renamed from eth1
Oct 27 06:05:19 homeassistant NetworkManager[450]: <info>  [1698386719.9559] manager: (veth1a96227): new Veth device (/org/freedesktop/NetworkManager/Devices/28)
Oct 27 06:05:19 homeassistant kernel: hassio: port 2(vethf84b4fe) entered disabled state
Oct 27 06:05:19 homeassistant kernel: device vethf84b4fe left promiscuous mode
Oct 27 06:05:19 homeassistant kernel: hassio: port 2(vethf84b4fe) entered disabled state
Oct 27 06:05:19 homeassistant kernel: audit: type=1700 audit(1698386719.969:641): dev=vethf84b4fe prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
Oct 27 06:05:20 homeassistant kernel: audit: type=1300 audit(1698386719.969:641): arch=c00000b7 syscall=206 success=yes exit=32 a0=c a1=400138c2c0 a2=20 a3=0 items=0 ppid=1 pid=556 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj=unconfined key=(null)
Oct 27 06:05:20 homeassistant kernel: audit: type=1327 audit(1698386719.969:641): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D636F6E7461696E6572643D2F72756E2F636F6E7461696E6572642F636F6E7461696E6572642E736F636B
Oct 27 06:05:20 homeassistant systemd[1]: run-docker-netns-eb06fb53b0f4.mount: Deactivated successfully.
Oct 27 06:05:20 homeassistant systemd[1]: var-lib-docker-overlay2-9be9ce732c56f6a63d39ec3587aa813d246211e5925c827275dad958b1701739-merged.mount: Deactivated successfully.
Oct 27 06:05:20 homeassistant systemd[1]: mnt-data-docker-overlay2-9be9ce732c56f6a63d39ec3587aa813d246211e5925c827275dad958b1701739-merged.mount: Deactivated successfully.
Oct 27 06:05:20 homeassistant hassos-supervisor[22328]: 0
Oct 27 06:05:20 homeassistant systemd[1]: hassos-supervisor.service: Deactivated successfully.
Oct 27 06:05:20 homeassistant systemd[1]: hassos-supervisor.service: Consumed 2.691s CPU time.
Oct 27 06:05:25 homeassistant systemd[1]: hassos-supervisor.service: Scheduled restart job, restart counter is at 2.
Oct 27 06:05:25 homeassistant systemd[1]: Stopped HassOS supervisor.
Oct 27 06:05:25 homeassistant systemd[1]: hassos-supervisor.service: Consumed 2.691s CPU time.
Oct 27 06:05:25 homeassistant systemd[1]: Starting HassOS supervisor...
Oct 27 06:05:25 homeassistant systemd[1]: Started HassOS supervisor.
Oct 27 06:05:25 homeassistant hassos-supervisor[25419]: [INFO] Starting the Supervisor...
Oct 27 06:05:25 homeassistant systemd[1]: docker-738c6cac3b3600721d9169ff95a87ae5caa61cf2fbb8eaafee722b13d4a6651d.scope: Deactivated successfully.
[... cut ...]
Oct 27 06:06:02 homeassistant kernel: audit: type=1334 audit(1698386762.804:675): prog-id=266 op=UNLOAD
Oct 27 06:17:42 homeassistant systemd-journald[136]: Data hash table of /var/log/journal/c891ea21a1624b92b2f47959a86805cc/system.journal has a fill level at 75.0 (85335 of 113777 items, 25165824 file size, 294 bytes per hash table item), suggesting rotation.
Oct 27 06:17:42 homeassistant systemd-journald[136]: /var/log/journal/c891ea21a1624b92b2f47959a86805cc/system.journal: Journal header limits reached or header out-of-date, rotating.
```

agners, 2023-10-27:
Hm, it looks like shutdown somehow failed. First the update is run, then the restart is triggered by the Supervisor, but somehow this did not lead to a full reboot:

```
Oct 27 06:04:27 homeassistant rauc[455]: input bundle: /mnt/data/supervisor/tmp/hassos-11.1.raucb
...
Oct 27 06:05:20 homeassistant systemd[1]: hassos-supervisor.service: Deactivated successfully.
Oct 27 06:05:20 homeassistant systemd[1]: hassos-supervisor.service: Consumed 2.691s CPU time.
Oct 27 06:05:25 homeassistant systemd[1]: hassos-supervisor.service: Scheduled restart job, restart counter is at 2.
Oct 27 06:05:25 homeassistant systemd[1]: Stopped HassOS supervisor.
Oct 27 06:05:25 homeassistant systemd[1]: hassos-supervisor.service: Consumed 2.691s CPU time.
Oct 27 06:05:25 homeassistant systemd[1]: Starting HassOS supervisor...
Oct 27 06:05:25 homeassistant systemd[1]: Started HassOS supervisor.
Oct 27 06:05:25 homeassistant hassos-supervisor[25419]: [INFO] Starting the Supervisor...
...
```

Since you seem to be on a virtual machine, can you login to the OS shell and get the complete logs using:
```
ha > login # < to get to the OS shell
```
then
```
journalctl -b -1 > /mnt/data/supervisor/homeassistant/last-boot-full.log
```

With that you should be able to find the full log in the Home Assistant config folder as `last-boot-full.log`.

francescopeloi, 2023-10-27:
what is the exact command to get the OS shell sorry? It seems that is not doing anything for me.
fyi to run commands on the machine, I am using the [Terminal & SSH addon](https://github.com/home-assistant/addons/tree/master/ssh).

agners, 2023-10-27:
@francescopeloi you can access the OS shell using keyboard/mouse connected to your system. What type of hardware are you using?

The Terminal & SSH add-on doesn't give access to the underlying OS shell, so you can't use that.

francescopeloi, 2023-10-27:
I am on a raspberry pi 4, unfortunately I don't have a mini hdmi cable with me to do this. There's no other way I guess?

agners, 2023-10-27:
There is: You can access the operating system console using SSH on port 22222. But you have to deploy a public key to the device first via a USB flash drive. The process is documented [here](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host).

francescopeloi, 2023-10-27:
Magic! It worked! Thanks.

Here's the output of 
```
journalctl -b -1 > /mnt/data/supervisor/homeassistant/last-boot-full.log
```
[last-boot-full.log](https://github.com/home-assistant/operating-system/files/13188772/last-boot-full.log)

francescopeloi, 2023-10-27:
Isn't the bootup log starting from this line?

```
Oct 27 06:05:25 homeassistant systemd[1]: hassos-supervisor.service: Scheduled restart job, restart counter is at 2.
```

Reboots never had an issue.

Anyway everything is working for me and I am taking back ups daily, so all good for me. It was more for you guys if you think you could have a bug somewhere.

agners, 2023-10-27:
> Isn't the bootup log starting from this line on the above attached file?

Normally there is much more before that, the hole kernel boot sequence (`homeassistant kernel ...`).

> Anyway everything is working for me and I am taking back ups daily, so all good for me. It was more for you guys if you think you could have a bug somewhere.

I do have Raspberry Pi 4 running here, and the restart went through fine after the update for me. I also guess that for most people it went through fine. It would be interesting what type of hickup your system had on restart, but the logs just seem incomplete.

Thanks again!

francescopeloi, 2023-10-27:
thank you for your instance on this investigation! I can now also ssh into the host machine, which I did not know I could do :)

BebeMischa, 2023-10-27:
So, I did leave it for night and working day alone. Just arrived from work now, and tried to run the update for maybe the 10th time now.

[last-boot.log](https://github.com/home-assistant/operating-system/files/13191548/last-boot.log)

regularguy01, 2023-10-27:
I had some wierdness as well.  I am on Proxmox and HAOS.

I had an OS 11.1 and a Supervisor update notice.

I clicked on the install 11.1.  Gave an error in the browser.  Nothing in the GUI happened.  So I refreshed cleared the browser cache.  clicked again Nothing.

So I thought maybe the Supervisor needed to go first?  So clicked install on that it went through and reported installed.

Then I tried the OS again seemed to go through.  After this the system seemed really slow all over.  So I rebooted the entire VM on proxmox.  It came up and reported updated to 11.1.  Seems to be fine now.

BebeMischa, 2023-10-29:
@agners Can you, please, have a look at my log above here? Looks, like my case is a bit hidden between all the more or less not related posts :-)

agners, 2023-10-29:
@BebeMischa hm, there seems to be very little information around that time when the update happens :thinking: 

```
Oct 27 16:54:35 homeassistant rauc[365]: input bundle: /mnt/data/supervisor/tmp/hassos-11.1.raucb
Oct 27 16:54:37 homeassistant kernel: loop0: detected capacity change from 0 to 413818
Oct 27 16:54:37 homeassistant kernel: loop0: detected capacity change from 413818 to 413816
Oct 27 16:54:37 homeassistant kernel: loop1: detected capacity change from 0 to 65536
```

Could you maybe try to get the logs via SSH on port 22222 as suggested in https://github.com/home-assistant/operating-system/issues/2870#issuecomment-1782715374 as well?

BebeMischa, 2023-10-29:
OMG, I just after 3 years of Supervised switched over to HAOS about a month a go, as widely advised on the HA forums. Just because it should give me a hassle free experience, no more self handling the OS level maintenance.
Well, that starts good...
With Supervised, I was able to SSH into it like a childplay.
Now I need to walk around with USB drives to carry over SSH keys?
What a smooth change i did myself... From a self managed system without troubles to a locked system failing on me...

Sorry for the rant, but this is, how I feel about it now.

I'm packing stuff for holidays, I have to work tomorrow and Tuesday and finally on Wednesday I have to leave for a month.
I'm not sure, if I will manage this in the last two evenings...

Damn....

BebeMischa, 2023-10-29:
Well, shoot me...

Out of frustration I did hit the update button for the 11th time and guess, what happened?
It updated.

I even don't try to understand... :-)

pedrofnm, 2023-10-30:
Yeah I have the same problem, I click to update, it download the update then it goes offline, reboots a few times then it comes online again after 10 mins with the same version (10.5) and with the update (to 11.1) notification. I'm using raspberry pi 4 4g with an old hard drive. Maybe it's because of the hard drive? Anyway I'm upgrading to an SSD soon so I hope it gets fixed at that time.

BebeMischa, 2023-10-30:
This gets even crazier now... So I was already on 11.1 . Today I rebooted the server (no any changes made) and it came back on with 11.0 and a update message... What the hell is going on here?

agners, 2023-10-30:
@BebeMischa sorry for the experience. It is the goal to make it hassle free, unfortunately bugs are a reality in any software :cry:  Ultimately I'd like to make it easier to get SSH access still, for exactly this reason. 

It seems that the detection failed that the system successfully booted. In that case, after three (presumably) failed attempts, the system falls back to the previous OS. 

This is done by the Supervisor, once it reaches the startup state. You should see a log entry similar to this in the Supervisor log:

```
Oct 30 19:43:51 ha-fujitsu hassio_supervisor[407]: 23-10-30 20:43:51 INFO (MainThread) [supervisor.os.manager] Rauc: B - marked slot kernel.1 as good
```

BebeMischa, 2023-10-30:
But how does a system, which did perform the update successfully, stating that in the system info, get downgraded back just by the next restart? I don't get it...

agners, 2023-10-30:
Essentially, we install two operating system in parallel (A/B slot). On each upgrade we install into the other slot, set a maker to try booting from that 3 times, and reboot. Once the reboot was successful, we mark that new slot as good, and from that point onwards the system will always use that new slot (-> upgrade successful).

Now, normally, when the system doesn't mark it as good, its really not good (as in, the system crashes during bootup, retries, crashes again, retries, at crashes a third time, thus causing the boot loader to switch back to the previous slot).

The question here is: Why does your system doesn't mark the new slot as good :thinking:  That is unclear to me, the Supervisor logs should have information about that.

It is safe to reinstall OS 11.1 (e.g. when you are in 11.0, just hit install again). This will essentially trigger the installation, reboot the system, and try to boot into 11.1 three times again.

BebeMischa, 2023-10-30:
Ok, now I get it a bit better. Well, I actually have seen the system rebooting more than 3x indeed on some of the attempts.

So it looks, like I'm stuck in some vicious circle, constantly switching between the two slots. But Supervisor does not complain about anything, all messages in the log are green...

I never had such problems with Debian'/Supervised combo, it did run seamless for 3 years...

BebeMischa, 2023-10-30:
So attempt 13 again succesfull, HA is running, sytem info says 11.1 We will see, what it does on next reboot...

agners, 2023-10-31:
@BebeMischa can you check the GRUB bootloader files?

```
ls -la /mnt/boot/EFI/BOOT/
grub-editenv /mnt/boot/EFI/BOOT/grubenv list
```

After an update, did you see the above mentioned line in the Supervisor log?

kdelios, 2023-10-31:
Just for the record I tried a few times to update my test setup Rpi4 8G booting from SSD from 11rc1 to 11.1 and nothing happened. System rebooted with the 11rc1 each time. So I did a host reboot and updated OS via CLI terminal and worked.

marceldeklerk, 2023-11-01:
I have the same problem, I click to update, it download the update then it goes offline, it tries to boot  three times (on the console i see errors about containerd not starting) after the 4th boot attempt it finally comes online again (took 10 min)  with the same version (10.5) and with the update (to 11.1) notification. I'm running HA on an Intel NUC.

[image]

Attached picture that I took from the console. I see these errors in every boot attempt, except the 4th attempt that is succesfull.

petervk, 2023-11-02:
I'm also having this error. I've tried to install 11.1 a few times and after a much longer than normal period of my Home Assistant not being available it finally boots up and is still running 11.0. Any idea of what can be causing this?

matit97, 2023-11-02:
Same problem, also from 10.1 to 11.1

paipaiv, 2023-11-02:
Same problem here, im in 10.5 try update to 11.1 but always return to 10.5

marcbernal, 2023-11-02:
Hey, same issue trying to update from 10.5 to 11 in the past weeks, and same with 11.1.

marceldeklerk, 2023-11-02:
I captured the bootlog by executing below:

`journalctl -b -1 > /mnt/data/supervisor/homeassistant/last-boot-full.log
`
In attached file, search for: `Failed to start containerd container runtime`

[last-boot-new2.log](https://github.com/home-assistant/operating-system/files/13244052/last-boot-new2.log)

```
Nov 02 19:51:38 homeassistant systemd[1]: containerd.service: Scheduled restart job, restart counter is at 1.
Nov 02 19:51:38 homeassistant systemd[1]: Stopped containerd container runtime.
Nov 02 19:51:38 homeassistant systemd[1]: Starting containerd container runtime...
Nov 02 19:51:38 homeassistant systemd[1]: containerd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Nov 02 19:51:38 homeassistant systemd[1]: containerd.service: Failed with result 'exit-code'.
Nov 02 19:51:38 homeassistant systemd[1]: Failed to start containerd container runtime.
```

agners, 2023-11-02:
@marceldeklerk your errors are very curious: containerd is on a read-only file system, as well as it's service files. I have no idea why it suddenly starts with INVALIDARGUMENT errors. The only thing I could think of is a corrupted root file system.

I guess you already tried, but even after a second installation attempt it doesn't start up?

Maybe the disk/SSD has really issues. Probably worth taking a full backup and download.

agners, 2023-11-02:
@marceldeklerk there are also this errors, which indicate disk issues:

```
Nov 01 20:28:59 homeassistant kernel: EXT4-fs (sda8): resizing filesystem from 29126769 to 29126769 blocks
Nov 01 20:28:59 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 32768
Nov 01 20:29:00 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 98304
Nov 01 20:29:00 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 163840
Nov 01 20:29:00 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 229376
Nov 01 20:29:00 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 294912
Nov 01 20:29:00 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 819200
Nov 01 20:29:00 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 884736
Nov 01 20:29:00 homeassistant kernel: EXT4-fs (sda8): Invalid checksum for backup superblock 1605632
Nov 01 20:29:00 homeassistant systemd-growfs[333]: Successfully resized "/mnt/data" to 111.1G bytes (3584 bytes lost due to blocksize).
```

I recommend replacing your disk.

marceldeklerk, 2023-11-03:
Thanks @agners. That doesn't look healthy indeed. 

Still strange that so many others here seem to have the same issue with the same upgrade to 11.1, and that the symptoms appear exactly like mine. 
@matit97 , @paipaiv , @marcbernal: Based on Agners observation, I am right to assume that your bootlog does not contain the following? ` containerd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT`

marcbernal, 2023-11-03:
In my case: 

1. Update starts
2. Pi restarts like two times and everything looks fine except for:

```
[FAILED] Failed to start Docker application Container Engine.
See 'systemctl status docker.service' for details.
[DEPEND] Dependency failed for HassOS supervisor.
```

3. After that, it restarts again and:
```

Waiting for the Home Assistant CLI to be ready...
[WARN] Home Assistant CLI not starting! Jump into emergency console...
# 
```

Trying to figure out if I can find more logs with errors.

Regards,

picard001, 2023-11-04:
Same here, updated a few times but nothing happened. Finally it appears to reboot but never finishes shutting down all services. I forced a reset and it failed to start containers and rebooted a few times. It finally switched the boot option to slot b (not familiar with how that works?) and it started back on 11.0.

marceldeklerk, 2023-11-05:
> @marceldeklerk there are also this errors, which indicate disk issues:
>
> ```
> Nov 01 20:28:59 homeassistant kernel: EXT4-fs (sda8): resizing filesystem from 29126769 to 29126769 blocks
> Nov 01 20:28:59 [...]

I've installed a new SSD in my NUC. Restored it from backup, and all is fine. Was able to upgrade 11.1 without issues. 
Thanks @agners

pedrofnm, 2023-11-08:
Just to give an update of my case, after I migrated my HA to an SSD drive and it got fixed. After restore, it loaded the 11.1 OS. So probably this gets fixed by just make a fresh install of HA and restore from a backup

asayler, 2023-11-19:
I'm seeing this issue as well, and have been going back through at least teh 10.X versions (it's not new to 11.x). Every so many reboots, the system will come up on an old version, presumably because it's booting the backup versions isneatd (in my case, I'm on 11.1, but the backup version is still 10.5 since 11.0 and 11.1 both got installed to the same partition). There's not really any rhyme or reason to what triggers the system to fail over to the older version. Each time it does, I re-run the upgrade and few times and eventually it kicks back over to the correct boot partition.

I'm running on a HA Yellow with a CM4 booting directly off an NVMe drive. I've attached boot logs from both several successful boots and several failed ones:

* [ha_boot_11.1_failed_1.txt](https://github.com/home-assistant/operating-system/files/13402523/ha_boot_11.1_failed_1.txt)
* [ha_boot_11.1_failed_2.txt](https://github.com/home-assistant/operating-system/files/13402524/ha_boot_11.1_failed_2.txt)
* [ha_boot_11.1_failed_3.txt](https://github.com/home-assistant/operating-system/files/13402525/ha_boot_11.1_failed_3.txt)
* [ha_boot_11.1_success_1.txt](https://github.com/home-assistant/operating-system/files/13402526/ha_boot_11.1_success_1.txt)
* [ha_boot_11.1_success_2.txt](https://github.com/home-assistant/operating-system/files/13402527/ha_boot_11.1_success_2.txt)
* [ha_boot_11.1_success_3.txt](https://github.com/home-assistant/operating-system/files/13402528/ha_boot_11.1_success_3.txt)

And the supervisor log from the most recent (successful) boot:

* [supervisor_2023-11-19T07-06-51.886Z.log](https://github.com/home-assistant/operating-system/files/13402531/supervisor_2023-11-19T07-06-51.886Z.log)

And the host log:

* [host_2023-11-19T07-07-54.880Z.log](https://github.com/home-assistant/operating-system/files/13402536/host_2023-11-19T07-07-54.880Z.log)

Let me know if I can provide anything else.

Is there a way to force the system to reboot to a specific slot? that might be nicer than just having to retry the update until it decides to use the A slot rather than fail over to the B slot.

JaCoLJcL, 2023-12-04:
Someone check the new 11.2 HAOS version? I'm stuck on 10.5 with this error :/

hugheaves, 2024-01-01:
FWIW, I had the same problem w/ upgrading to 11.1. So I did a clean install, and now can't upgrade to 11.2. I don't see anything useful in the logs.

asayler, 2024-01-15:
I'm still seeing this issue, and it seems to have gotten more persensit on the 11.3 -> 11.4 update attempt. On previous updates, I could generally get it to eventually take by trying it enough times, but I've attempted the 11.3 -> 11.4 update ~10 times thus far with no luck. As far as I can tell, the system isn't experiencing any failed boots. I've watched the entire process via the serial terminal, and it plays out as follows:

1. run `ha os update` from serial console (note: I'm booted into boot slot B when this occurs)
2. System runs the 11.4 update and says ti was successful
3. A minute or two later, the system reboots.
4. The reboot again boots into slot B, which is still running 11.3. AFAICT, it never even attempts to boot slot A.

Perhaps there's an error in the update script or `rauc` system that isn't properly setting which slot to attempt on the next boot. Is there a way I can manually override that?

I've attached the output from `journalctl -b -1 > /mnt/data/supervisor/homeassistant/last-boot-full.log`. For some reason, the log appears to go back more than the most recent boot (which was 1/14), and the initial timestamps are wrong until the RTC loads up. But you can see the most recent update attempt in the output at the line that starts with `Jan 15 05:34:36 ha rauc[483]`.

[last-boot-full.log](https://github.com/home-assistant/operating-system/files/13935469/last-boot-full.log)

I'm running this on an HA Yellow using a Raspberry Pi CM4. I installed HaOS directly to the NVMe drive about 10 months ago using the directions found in Step 9 of https://yellow.home-assistant.io/power-supply/#installing-home-assistant-software-on-kit.

Happy to test things out as needed.

asayler, 2024-01-15:
Adding to the above, I poked at my `/mnt/daat/rauc.db` file as well, it it shows both slots marked as okay. So I'm not sure what it keeps booting into Slot B, despite a successful 11.4 install in Slot A. File attached.

[rauc-20240114.db.txt](https://github.com/home-assistant/operating-system/files/13935525/rauc-20240114.db.txt)

asayler, 2024-01-15:
And here's the about from `rauc status`:

```
# rauc status
=== System Info ===
Compatible:  haos-yellow
Variant:
Booted from: kernel.1 (B)

=== Bootloader ===
Activated: kernel.0 (A)

=== Slot States ===
  [boot.0] (/dev/disk/by-partlabel/hassos-boot, vfat, inactive)

x [kernel.0] (/dev/disk/by-partlabel/hassos-kernel0, raw, inactive)
        bootname: A
        boot status: good
    [rootfs.0] (/dev/disk/by-partlabel/hassos-system0, raw, inactive)

o [kernel.1] (/dev/disk/by-partlabel/hassos-kernel1, raw, booted)
        bootname: B
        boot status: good
    [rootfs.1] (/dev/disk/by-partlabel/hassos-system1, raw, active)
```

So it even thinks Slot A is active, despite it booting into Slot B.

Any ideas?

asayler, 2024-01-15:
Okay, tried yet another 11.4 update and this time I actually did see it attempt to Boot Slot A and fail to boot. This seems to be the critical bit where it abandons the boot attempt:

```
[  OK  ] Finished HassOS AppArmor.
         Starting Docker Application Container Engine...
[FAILED] Failed to start Docker Application Container Engine.
See 'systemctl status docker.service' for details.
[DEPEND] Dependency failed for HassOS supervisor.
[  OK  ] Removed slice Slice /system/modprobe.
[  OK  ] Stopped target HassOS hardware targets.
[  OK  ] Stopped target HassOS zram targets.
```

Full log from one of the failed boot attempts: [ha_boot_11.4_failed.txt](https://github.com/home-assistant/operating-system/files/13935736/ha_boot_11.4_failed.txt)

Interestingly, the rauc status still marks Slot A as good/active, so perhaps that's just the way the u-boot rauc integration here works.

asayler, 2024-01-15:
I went ahead and opened a new issue for the 11.4 docker error ^^^ since it seems to be unique enough from this issue to warrant it: #3074.

petervk, 2024-01-30:
I'm still on haos 11.0 as I've never been able to successfully install anything newer. I've connected a monitor directly to my host (a dell 7040 micro) and watched it boot up and it seems to fail somewhere and go back to 11.0. I'm only able to access the last two boots using the commands provided. I tried to run "rauc" but I get a command not found error.

I've attached the output of the following two commands:
```
ha host logs -b -1 -n 10000 > /config/last-boot.log
ha host logs -b 0 -n 10000 > /config/current-boot.log

```
[current-boot.log](https://github.com/home-assistant/operating-system/files/14101707/current-boot.log)
[last-boot.log](https://github.com/home-assistant/operating-system/files/14101708/last-boot.log)

petervk, 2024-04-30:
No, I still can't update the OS and wish there was some help to update it
without reinstalling and restoring from a backup.

closed 2024-08-07 by github-actions[bot]

guidorobben, 2024-12-19:
I have this issue from 13.2 to 14.1.