home-assistant/operating-system #2875: Home Assistant Operating System Update

razzietheman, 2023-10-29:
### Describe the issue you are experiencing

try to update Home Assistant Operating System from 10.5 to 11.1 but only getting unknown error

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
23-10-29 08:04:29 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-10-29 08:04:50 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/11.1/haos_generic-x86-64-11.1.raucb
23-10-29 08:04:53 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-11.1.raucb
23-10-29 08:04:53 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.0 non-bootable: grub backend: Failed to run grub-editenv: Child process exited with code 1
```

### Anything in the Host logs that might be useful for us?

```txt
nothing
```

### System information

## System Information

version | core-2023.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1319
Downloaded Repositories | 17

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 228.5 GB
disk_used | 16.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.15.0), MariaDB (2.6.1), Samba share (10.0.2), FTP (4.7.3), Filebrowser (2.23.0_reverted), phpMyAdmin (0.8.9), Home Assistant Google Drive Backup (0.111.1), Advanced SSH & Web Terminal (15.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 5
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 19 oktober 2023 kl. 07:03
-- | --
current_recorder_run | 28 oktober 2023 kl. 20:01
estimated_db_size | 259.25 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Svensk Postutdelning</summary>

Version | 1.0.7
-- | --
Idle | false
Instances | 1

</details>

njbuch, 2023-10-29:
I am experiencing the same issue on my RPI4-64 box with HACS. Feel free to request more info. Skipped the 11.0 to wait for bugfix in 11.1 - but this upgrade wont complete either.

agners, 2023-10-29:
It seems that the bootloader has some issues. Can you share the host logs after an attempt (same place, but choose host logs on the top right)?

```
23-10-29 08:04:53 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.0 non-bootable: grub backend: Failed to run grub-editenv: Child process exited with code 1
```

@njbuch in this case this seems to be about an issue with the GRUB bootloader on x86-64. This is unrelated to Raspberry Pi. Can you open a new issue with all the logs (please check before if you see a simliar report for Raspberry Pi with the same errors)?

razzietheman, 2023-10-30:
> It seems that the bootloader has some issues. Can you share the host logs after an attempt (same place, but choose host logs on the top right)?
>
> ```
> 23-10-29 08:04:53 ERROR (MainThread) [...]

well the host logs show nothing at all.........

agners, 2023-10-30:
> well the host logs show nothing at all.........

Hm, there should always be some logs there. How does it look for you? Just a white page? Or some message about issues on loading the host logs?

Otherwise you can try the Terminal and then download the log file from the config directory:

```
ha host logs -b 0 -n 10000 > /config/host-logs.log
```

razzietheman, 2023-10-30:
> Hm, there should always be some logs there. How does it look for you? Just a white page? Or some message about issues on loading the host logs?
>
> Otherwise you can try the Terminal and then download [...]

my hostlog

Oct 30 09:12:10 homeassistant systemd[1]: Started Hostname Service.
Oct 30 09:12:10 homeassistant kernel: audit: type=1334 audit(1698657130.610:510): prog-id=194 op=LOAD
Oct 30 09:12:10 homeassistant kernel: audit: type=1334 audit(1698657130.610:511): prog-id=195 op=LOAD
Oct 30 09:12:10 homeassistant kernel: audit: type=1334 audit(1698657130.610:512): prog-id=196 op=LOAD
Oct 30 09:12:10 homeassistant systemd[1]: Starting Time & Date Service...
Oct 30 09:12:10 homeassistant systemd[1]: Started Time & Date Service.
Oct 30 09:12:40 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Oct 30 09:12:40 homeassistant kernel: audit: type=1334 audit(1698657160.675:513): prog-id=193 op=UNLOAD
Oct 30 09:12:40 homeassistant kernel: audit: type=1334 audit(1698657160.675:514): prog-id=192 op=UNLOAD
Oct 30 09:12:40 homeassistant kernel: audit: type=1334 audit(1698657160.675:515): prog-id=191 op=UNLOAD
Oct 30 09:12:40 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
Oct 30 09:12:40 homeassistant kernel: audit: type=1334 audit(1698657160.766:516): prog-id=196 op=UNLOAD
Oct 30 09:12:40 homeassistant kernel: audit: type=1334 audit(1698657160.766:517): prog-id=195 op=UNLOAD
Oct 30 09:12:40 homeassistant kernel: audit: type=1334 audit(1698657160.766:518): prog-id=194 op=UNLOAD
Oct 30 10:19:16 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7e68e1fb43bf00eb7436b1236c9023fe696b47928068ad7cf0147f9a8ab541ea-runc.jIYSEu.mount: Deactivated successfully.
Oct 30 11:18:50 homeassistant kernel: audit: type=1334 audit(1698664730.782:519): prog-id=197 op=LOAD
Oct 30 11:18:50 homeassistant kernel: audit: type=1334 audit(1698664730.783:520): prog-id=198 op=LOAD
Oct 30 11:18:50 homeassistant kernel: audit: type=1334 audit(1698664730.783:521): prog-id=199 op=LOAD
Oct 30 11:18:50 homeassistant systemd[1]: Starting Hostname Service...
Oct 30 11:18:50 homeassistant systemd[1]: Started Hostname Service.
Oct 30 11:18:50 homeassistant kernel: audit: type=1334 audit(1698664730.931:522): prog-id=200 op=LOAD
Oct 30 11:18:50 homeassistant kernel: audit: type=1334 audit(1698664730.931:523): prog-id=201 op=LOAD
Oct 30 11:18:50 homeassistant kernel: audit: type=1334 audit(1698664730.931:524): prog-id=202 op=LOAD
Oct 30 11:18:50 homeassistant systemd[1]: Starting Time & Date Service...
Oct 30 11:18:51 homeassistant systemd[1]: Started Time & Date Service.
Oct 30 11:19:20 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Oct 30 11:19:20 homeassistant kernel: audit: type=1334 audit(1698664760.971:525): prog-id=199 op=UNLOAD
Oct 30 11:19:20 homeassistant kernel: audit: type=1334 audit(1698664760.971:526): prog-id=198 op=UNLOAD
Oct 30 11:19:20 homeassistant kernel: audit: type=1334 audit(1698664760.971:527): prog-id=197 op=UNLOAD
Oct 30 11:19:21 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
Oct 30 11:19:21 homeassistant kernel: audit: type=1334 audit(1698664761.082:528): prog-id=202 op=UNLOAD
Oct 30 11:19:21 homeassistant kernel: audit: type=1334 audit(1698664761.082:529): prog-id=201 op=UNLOAD
Oct 30 11:19:21 homeassistant kernel: audit: type=1334 audit(1698664761.082:530): prog-id=200 op=UNLOAD
Oct 30 13:11:26 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7e68e1fb43bf00eb7436b1236c9023fe696b47928068ad7cf0147f9a8ab541ea-runc.rRIgTJ.mount: Deactivated successfully.
Oct 30 13:25:31 homeassistant kernel: audit: type=1334 audit(1698672331.099:531): prog-id=203 op=LOAD
Oct 30 13:25:31 homeassistant kernel: audit: type=1334 audit(1698672331.099:532): prog-id=204 op=LOAD
Oct 30 13:25:31 homeassistant kernel: audit: type=1334 audit(1698672331.100:533): prog-id=205 op=LOAD
Oct 30 13:25:31 homeassistant systemd[1]: Starting Hostname Service...
Oct 30 13:25:31 homeassistant systemd[1]: Started Hostname Service.
Oct 30 13:25:31 homeassistant kernel: audit: type=1334 audit(1698672331.254:534): prog-id=206 op=LOAD
Oct 30 13:25:31 homeassistant kernel: audit: type=1334 audit(1698672331.254:535): prog-id=207 op=LOAD
Oct 30 13:25:31 homeassistant kernel: audit: type=1334 audit(1698672331.254:536): prog-id=208 op=LOAD
Oct 30 13:25:31 homeassistant systemd[1]: Starting Time & Date Service...
Oct 30 13:25:31 homeassistant systemd[1]: Started Time & Date Service.
Oct 30 13:26:01 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Oct 30 13:26:01 homeassistant kernel: audit: type=1334 audit(1698672361.317:537): prog-id=205 op=UNLOAD
Oct 30 13:26:01 homeassistant kernel: audit: type=1334 audit(1698672361.317:538): prog-id=204 op=UNLOAD
Oct 30 13:26:01 homeassistant kernel: audit: type=1334 audit(1698672361.317:539): prog-id=203 op=UNLOAD
Oct 30 13:26:01 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
Oct 30 13:26:01 homeassistant kernel: audit: type=1334 audit(1698672361.407:540): prog-id=208 op=UNLOAD
Oct 30 13:26:01 homeassistant kernel: audit: type=1334 audit(1698672361.407:541): prog-id=207 op=UNLOAD
Oct 30 13:26:01 homeassistant kernel: audit: type=1334 audit(1698672361.407:542): prog-id=206 op=UNLOAD
Oct 30 14:03:02 homeassistant NetworkManager[400]: <info>  [1698674582.0924] dhcp4 (eno1): state changed new lease, address=192.168.86.58
[... cut ...]
Oct 30 18:14:50 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7e68e1fb43bf00eb7436b1236c9023fe696b47928068ad7cf0147f9a8ab541ea-runc.xa1zJh.mount: Deactivated successfully.
Oct 30 18:58:13 homeassistant rauc[404]: input bundle: /mnt/data/supervisor/tmp/hassos-11.1.raucb
Oct 30 18:58:13 homeassistant kernel: loop0: detected capacity change from 0 to 413818
Oct 30 18:58:13 homeassistant kernel: loop0: detected capacity change from 413818 to 413816
Oct 30 18:58:13 homeassistant rauc[688371]: grub-editenv: error: environment block too small.

agners, 2023-10-31:
It seems that GRUB failed to edit the environment:

```
 Oct 30 18:58:13 homeassistant rauc[688371]: grub-editenv: error: environment block too small.
```

It seems that the boot partition got corrupted on your system :cry:

Before doing further operations, it is probably a good idea to take a full backup and download it, just to be safe.

From the console (monitor/keyboard), you can check this by:

```
ls -la /mnt/boot/EFI/BOOT/grubenv
grub-editenv /mnt/boot/EFI/BOOT/grubenv list
```

You probably can fix it by copying a backup environoment as follows:
```
cp /mnt/boot/EFI/BOOT/grubenv-A /mnt/boot/EFI/BOOT/grubenv
```

Otherwise a new installation and restore from the full back is probably the best course of action.

razzietheman, 2023-11-02:
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:230
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 19:09:22 (1 occurrences)
Last logged: 19:09:22

[139826693366080] Error updating Home Assistant Operating System: Unknown error, see supervisor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 212, in async_install
    await async_update_os(self.hass, version)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 53, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 432, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 214, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Operating System: Unknown error, see supervisor

razzietheman, 2023-11-03:
> Yeah these logs doesn't tell us more. The problem is identified: The boot partition seems to be corrupted.
> 
> The best course of action is to take a full backup, download it and start over with a new installation, and restore the backup there.

how to reapair it inside HA? or can I flash it to the nuc from USB? won't to unplug the NUC and the harddrive and so on

closed 2024-02-10 by github-actions[bot]