home-assistant/operating-system #2876: Update to 11.1 on raspberry pi 3b with USB3 SSD

magin1, 2023-10-29:
### Describe the issue you are experiencing

After updating from 11.0 (i think) to 11.1  and now my HA is not booting anymore.
It hangs in u-boot unable to find storage device to boot from.
After a while it shows some lines of which i dont know what they mean.
[image]

I tried redoing to config needed to boot from ssd. no luck
I installed HA on a usb drive and booted from that. this worked.

So i thought it must be my SSD. so i installed virtualbox and Linux mint to check the SSD partitions.
And they are all there and i can see my config.
[image]

So now i'm out of ideas what i could be.
Would love some help.

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Not relevant

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

agners, 2023-10-29:
There was no change between HAOS 11.0 and 11.1. I can really say for sure what the issue is, but it seems the bootloader can't talk to the disk. I'd try to power off the whole system, maybe use a different USB hub or powered hub in between the Raspberry Pi and the USB SSD.

magin1, 2023-10-30:
> There was no change between HAOS 11.0 and 11.1. I can really say for sure what the issue is, but it seems the bootloader can't talk to the disk. I'd try to power off the whole system, maybe use a different USB hub or powered hub in between the Raspberry Pi and the USB SSD.

Hi thanks for you reply. toady i tested the following:
- left the raspberry off power whole night, in the morning still no differance.
- had another raspberry pi3b so i also made that albe to boot from ssd but sadly same result. Led from ssd flahses so it it talked to by the u-boot but no boot.
- also had some more powerfull chargers then the default raspberry one, connected those and same results:(

If its the bootloader not beeing able to talk to the disk for any reason, how can i check/repair this?

agners, 2023-10-31:
> If its the bootloader not beeing able to talk to the disk for any reason, how can i check/repair this?

Unfortunately, USB mass storage (especially disks) is sometimes really picky to get to work. U-Boot is known to have issues in some certain circumstances. It might also be related to the boot-up sequence.

Now since your setup worked previously, it would be intersting to know which release worked exactly. You can download old releases from GitHub.

What I would recommend is to use the data disk approach: So boot from SD card, but have all data stored on the SSD. This sidesteps the bootup problems, while allowing to use the speed and size of an USB attached SSD, see https://www.home-assistant.io/common-tasks/os/#using-external-data-disk.

joostlitjens, 2023-11-03:
I had same issues, i abandoned the raspi 3b ssd set up because it worked great for years but this bug is coming back every few releases. Hard too find out which is the latest good working version. Even 4.20 is not booting anymore with my set up so i gave up.

closed 2024-02-10 by github-actions[bot]