home-assistant/operating-system #2886: Systemd 252 breaks usb serial by-id paths

paradox460, 2023-10-30:
### Describe the issue you are experiencing

When attaching dongles, a reasonable practice is to use the `/dev/serial/by-id/<bus>-<serial>-<interface>` path, instead of `/dev/tty0` and others, that can "move around"

[Systemd introduced a bug in 252](https://github.com/systemd/systemd/issues/25238) that breaks these paths; they simply do not exist on systems running 252, because the udev serial ruleset for them is broken.

This was fixed in https://github.com/systemd/systemd/pull/25246, which added this feature back, and has been backported to systemd 252.1, as well as being included in systemd 253 and newer.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Attach a USB dongle to the host machine, such as a Z-Wave stick
2. Ensure dongle has proper passthrough configured in the VM
3. Ensure dongle appears as a `/dev/ttyUSB*` style device in the guest OS (HAOS)
4. You will not see the device at `/dev/serial/by-id/*` path, the `/dev/serial/by-id` directory is entirely missing

### Anything in the Supervisor logs that might be useful for us?

```txt
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
23-10-30 09:01:09 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-10-30 09:01:36 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

23-10-30 09:31:36 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

23-10-30 10:01:06 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
23-10-30 10:01:09 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-10-30 10:01:36 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

23-10-30 10:31:37 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

23-10-30 10:54:31 INFO (MainThread) [supervisor.host.info] Updating local host information
[... 125 lines cut ...]
Oct 30 18:00:09 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-cfe916de0bece10f2d86a06a565daebcc5d6fb3d0195908f023ce5a00b64375f-runc.kVtpoY.mount: Deactivated successfully.
Oct 30 18:00:20 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-1e2cfcbf015395dd6b7b2f6584c59c6c32a3fad4103b4e4ee52bfec5b347e738-runc.tCwKDn.mount: Deactivated successfully.
Oct 30 18:00:28 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ae30063642c2fdf7afea1c709c20eec527f0a943bf5360a4262e40ee0f5669fd-runc.CMeXlw.mount: Deactivated successfully.
Oct 30 18:00:30 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-717e6fdc4ff5bfe526ea8444fe41ce6307e6c21fdbc4dec30d6de914855ec5dc-runc.wYhgww.mount: Deactivated successfully.
Oct 30 18:00:31 homeassistant kernel: audit: type=1334 audit(1698688831.452:1696): prog-id=757 op=LOAD
Oct 30 18:00:31 homeassistant systemd[1]: Started Journal Gateway Service.
Oct 30 18:00:31 homeassistant systemd-timesyncd[398]: Network configuration changed, trying to establish connection.
Oct 30 18:00:31 homeassistant systemd-timesyncd[398]: Contacted time server 162.159.200.123:123 (time.cloudflare.com).
Oct 30 18:00:31 homeassistant systemd-journal-gatewayd[2783084]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
```

### Additional information

I opened this issue against HA-core, but was told it was better to put it here. The relevant HA-Core issue is: https://github.com/home-assistant/core/issues/101979

----

I also left [a comment with instructions on a bandaid fix](https://github.com/home-assistant/core/issues/101979#issuecomment-1762491892), quoted here:

> For those that find this issue, and are waiting for a fix, you can apply the [new rules file](https://raw.githubusercontent.com/yuwata/systemd/5286da064c97d2ac934cb301066aaa8605a3c8f9/rules.d/60-serial.rules) to your system.
> 
> 1. Create a new file at `/etc/udev/rules.d/60-serial.rule` with the contents of the [new rules](https://raw.githubusercontent.com/yuwata/systemd/5286da064c97d2ac934cb301066aaa8605a3c8f9/rules.d/60-serial.rules)
> 2. Run `sudo udevadm control --reload-rules && sudo udevadm trigger`
> 3. Unplug and replug (or the equivalent if you are running virtualized) the USB devices.
> 4. You should now see them mounted in `/dev/serial/by-id`

agners, 2023-10-31:
Home Assistant OS ships with systemd 252.9 (see https://github.com/home-assistant/buildroot/blob/82003cc280472999d0488a42e700ef8dacb58198/package/systemd/systemd.mk#L22), so this issue should be fixed.

In my test instances I also have `/dev/serial/by-id/` entries for USB devices.

> You will not see the device at /dev/serial/by-id/* path, the /dev/serial/by-id directory is entirely missing

How do you check this?

What virtualization environment are you using?

paradox460, 2023-11-01:
I'm in QEMU via virt-framework (virsh)

I've been checking it by accessing the console via `virsh console hass`, and then navigating to `/dev`. Prior to manually adding a serial rule to udev, I did not see them there at all. They worked prior to the update to 252.

agners, 2023-12-05:
There wasn't really an update to systemd with HAOS 11.2, but just in case, did that resolve it for you? (if so, then maybe the kernel stable updates helped :man_shrugging: )

I've checked virsh/libvirtd based installation on my end. Both a FTDI and a CP2102N based USB-to-serial adapater device appeared properly in `/dev/serial/by-id/`. Maybe this is a problem for certain devices only? What type of device do you use?

WinterPhoenix, 2023-12-06:
Jumping in on this; I have QEMU + Virsh too, and updated to HAOS 11.2, same problem.

```
Error setting up entry ConBee II for zha

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2289119-if00'
```

The bandaid fix works if you use [the actual new version of the 60-serial.rules](https://github.com/systemd/systemd/blob/main/rules.d/60-serial.rules). The one previously suggested is the old rules file.

agners, 2023-12-19:
This all really doesn't make a lot of sense currently: `/usr/lib/udev/rules.d/60-serial.rules` is there and it's content is

```
# do not edit this file, it will be overwritten on update

ACTION=="remove", GOTO="serial_end"
SUBSYSTEM!="tty", GOTO="serial_end"

SUBSYSTEMS=="usb", IMPORT{builtin}="usb_id", IMPORT{builtin}="hwdb --subsystem=usb"
SUBSYSTEMS=="pci", ENV{ID_BUS}=="", ENV{ID_BUS}="pci", \
  ENV{ID_VENDOR_ID}="$attr{vendor}", ENV{ID_MODEL_ID}="$attr{device}", \
  IMPORT{builtin}="hwdb --subsystem=pci"

# /dev/serial/by-path/, /dev/serial/by-id/ for USB devices
KERNEL!="ttyUSB[0-9]*|ttyACM[0-9]*", GOTO="serial_end"

SUBSYSTEMS=="usb-serial", ENV{.ID_PORT}="$attr{port_number}"

IMPORT{builtin}="path_id"
ENV{ID_PATH}=="?*", ENV{.ID_PORT}=="", SYMLINK+="serial/by-path/$env{ID_PATH}"
ENV{ID_PATH}=="?*", ENV{.ID_PORT}=="?*", SYMLINK+="serial/by-path/$env{ID_PATH}-port$env{.ID_PORT}"

ENV{ID_BUS}=="", GOTO="serial_end"
ENV{ID_SERIAL}=="", GOTO="serial_end"
ENV{ID_USB_INTERFACE_NUM}=="", GOTO="serial_end"
ENV{.ID_PORT}=="", SYMLINK+="serial/by-id/$env{ID_BUS}-$env{ID_SERIAL}-if$env{ID_USB_INTERFACE_NUM}"
ENV{.ID_PORT}=="?*", SYMLINK+="serial/by-id/$env{ID_BUS}-$env{ID_SERIAL}-if$env{ID_USB_INTERFACE_NUM}-port$env{.ID_PORT}"

LABEL="serial_end"
```

Comparing with the linked file, it only has the additional `ID_PATH_WITH_USB_REVISION` entries, which from what I understand is something which needs a newer systemd-udev version as well (see https://github.com/systemd/systemd/commit/873be895ed1fe65010fb84c6b71e2ec0a6b6fc91#diff-a712a6f4004cd0bd01dc6e4c9494e329c29cc52de1c4cbc24879c7f03c1aae29).

My best guess is that this is some race condition or random issue. The additional udev rules might change that race and/or it just randomly working at times.

closed 2024-04-11 by github-actions[bot]